PX4/PX4-Autopilot #20484: Parameter are missing from firmware shown by QGroundControl after checking out main tip

navunlimited, 2022-10-26:
## Describe the bug
Not sure it is hardware(board) related. I use beaglebone blue.
Checkout V1.13.1, everything works fine.
Checkout current main tip. QGroundControl display:
Parameters are missing from firmware. .....   Missing params:1:SYS_AUTOCONFIG, 1;COM_FLTMODE1, 1;COM_FLTMODE2, 1;COM_FLTMODE3, 1;COM_FLTMODE4, 1;COM_FLTMODE5, 1;COM_FLTMODE6
Due to these missing parameters, RC remote controller cannot setup flight mode, and manual remote control is not available. 

## To Reproduce
Steps to reproduce the behavior:
Tried checkout at two places in main branch, both have the same error message:
1) current main tip
2) #4fbfcf2375 Update sitl gazebo submodule

But V1.13.1 has no issue.

## Log Files and Screenshots
[image]

navunlimited, 2022-10-28:
I found the issue might be the checking function param_used() to check if parameter is used or not in mavline_parameters.cpp:
in MavlinkParametersManager::send_one(){
.
.
.
               /* look for the first parameter which is used */
		param_t p;

		do {
			/* walk through all parameters, including unused ones */
			p = param_for_index(_send_all_index);
			_send_all_index++; p_count++;

		} while (p != PARAM_INVALID && !param_used(p));

Those missing parameters are skip due to param_used() checked is false. Event those parameters are defined in commander_params.c and system_params.c
Currently my work-around is re-defined those paramters in px4.config script.

tayyabkhalil-313, 2022-11-08:
I faced the same issue and worked around it by formatting my SD card.