PX4/PX4-Autopilot #20503: Preflight errors during flight in offboard mode

Jaeyoung-Lim, 2022-10-28:
## Describe the bug
While flying in offboard mode, preflight errors keep appearing

## To Reproduce
Steps to reproduce the behavior:
1. Fly in offboard mode, for a while in SITL
2. Check px4 shell

## Expected behavior
Preflight errors should not prompt after takeoff

## Log Files and Screenshots
- Flight log: 

```
[ERROR] [1666958082.411005289, 101.460000000]: TM : Time jump detected. Resetting time synchroniser.
[ERROR] [1666958133.011041949, 152.040000000]: TM : Time jump detected. Resetting time synchroniser.
[ERROR] [1666958183.610969284, 202.620000000]: TM : Time jump detected. Resetting time synchroniser.
[ERROR] [1666958234.210952414, 253.200000000]: TM : Time jump detected. Resetting time synchroniser.
[ERROR] [1666958284.811088021, 303.776000000]: TM : Time jump detected. Resetting time synchroniser.
[ERROR] [1666958335.410965193, 354.356000000]: TM : Time jump detected. Resetting time synchroniser.
[ERROR] [1666958386.011019053, 404.936000000]: TM : Time jump detected. Resetting time synchroniser.
[ERROR] [1666958436.611025057, 455.516000000]: TM : Time jump detected. Resetting time synchroniser.
[ERROR] [1666958487.210986083, 506.092000000]: TM : Time jump detected. Resetting time synchroniser.
[ERROR] [1666958537.810985559, 556.672000000]: TM : Time jump detected. Resetting time synchroniser.
[ERROR] [1666958588.411034201, 607.252000000]: TM : Time jump detected. Resetting time synchroniser.
[ERROR] [1666958639.010983934, 657.832000000]: TM : Time jump detected. Resetting time synchroniser.
[ERROR] [1666958689.611000868, 708.408000000]: TM : Time jump detected. Resetting time synchroniser.
[ERROR] [1666958740.210998311, 758.988000000]: TM : Time jump detected. Resetting time synchroniser.
[ERROR] [1666958790.810989592, 809.564000000]: TM : Time jump detected. Resetting time synchroniser.
[ERROR] [1666958841.410952192, 860.144000000]: TM : Time jump detected. Resetting time synchroniser.
[ERROR] [1666958892.010979900, 910.724000000]: TM : Time jump detected. Resetting time synchroniser.
[ERROR] [1666958942.611082432, 961.304000000]: TM : Time jump detected. Resetting time synchroniser.
[ERROR] [1666958993.211071045, 1011.880000000]: TM : Time jump detected. Resetting time synchroniser.
[ERROR] [1666959043.810952042, 1062.460000000]: TM : Time jump detected. Resetting time synchroniser.
[ERROR] [1666959094.410951561, 1113.036000000]: TM : Time jump detected. Resetting time synchroniser.
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
[ERROR] [1666959145.011004882, 1163.612000000]: TM : Time jump detected. Resetting time synchroniser.
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
[ERROR] [1666959195.610966178, 1214.192000000]: TM : Time jump detected. Resetting time synchroniser.
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: velocity estimate error
WARN  [health_and_arming_checks] Preflight Fail: velocity estimate error
```

## Drone (please complete the following information):
- SITL Gazebo `plane` model

## Additional context
Add any other context about the problem here.

jstrebel, 2023-02-21:
Observe the same with 1.14 main(beta) (downloaded 21.2.23)

julianoes, 2023-10-20:
I'm gonna close this as outdated.

closed 2023-10-20 by julianoes