PX4/PX4-Autopilot #20507: Dismatch board id ( can not upload firmware to pixhawk cube orange plus )

JeongseokHyun, 2022-10-29:
I buy a new board ( cube orange plus ).
When I try to upload firmware in the board, It can not do.
As the information, The reason is dismatched board ID.
So, I check the source code.
I think even latest version PX4 don't support a new board.
So, I want to request it.

Futhermore, I have a question.
before update the PX4, I need to use temporary edit the source code for change the board id from 140 to 1063.
Then, I can upload the firmware in the board.
Can it be the problem???

ripdk12, 2022-11-04:
Hey man, take a look at this PR: #20304
I was able to compile and upload firmware for orange plus by following this.
I am still waiting for official support though, it seems long overdue honestly.

ripdk12, 2022-12-09:
@valenbase What exact command are you using?
You must be in the "PX4-Autopilot" folder, typing:
```make cubepilot_cubeorange_default upload```

EhrazImam, 2023-02-22:
Hi @valenbase Hope you are doing well can you help me with uploading of px4 Firmware on CUBE ORANGE PLUS as i am not able to upload, As i can see you have faced similar problem and you resolved it.
As i have seen to you suggestion but i am new to this, it will be great if you can help me with this.

ripdk12, 2023-02-22:
@EhrazImam What exactly is your problem, what have you tried?
As of the latest main, it should work out of the box by running:
`
git clone --recursive https://github.com/PX4/PX4-Autopilot.git
cd PX4-Autopilot
make cubepilot_cubeorangeplus_default upload
`

EhrazImam, 2023-02-22:
I followed your step and i am getting this error i simply connencted my ubuntu based system with cube orange plus with usb b cable
/PX4-Autopilot$  make cubepilot_cubeorangeplus_default upload
-- PX4 version: v1.14.0-beta1-429-g98263de17b (1.14.0)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- PX4 config file: /home/enord/PX4-Autopilot/boards/cubepilot/cubeorangeplus/default.px4board
-- PLATFORM nuttx
-- TOOLCHAIN arm-none-eabi
-- ARCHITECTURE cortex-m7
-- ROMFSROOT px4fmu_common
-- IO cubepilot_io-v2_default
-- SERIAL_GPS1 /dev/ttyS2
-- SERIAL_GPS2 /dev/ttyS5
-- SERIAL_TEL1 /dev/ttyS0
-- SERIAL_TEL2 /dev/ttyS1
-- UAVCAN_INTERFACES 2
-- PX4 config: cubepilot_cubeorangeplus_default
-- PX4 platform: nuttx
-- The CXX compiler identification is unknown
-- The C compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: arm-none-eabi-gcc
CMake Error at CMakeLists.txt:219 (project):
  The CMAKE_CXX_COMPILER:

    arm-none-eabi-g++

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:219 (project):
  The CMAKE_C_COMPILER:

    arm-none-eabi-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

-- Enabling double FP precision hardware instructions
CMake Error at CMakeLists.txt:219 (project):
  The CMAKE_ASM_COMPILER:

    arm-none-eabi-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Warning: Did not find file Compiler/-ASM
-- Enabling double FP precision hardware instructions
-- Configuring incomplete, errors occurred!
See also "/home/enord/PX4-Autopilot/build/cubepilot_cubeorangeplus_default/CMakeFiles/CMakeOutput.log".
See also "/home/enord/PX4-Autopilot/build/cubepilot_cubeorangeplus_default/CMakeFiles/CMakeError.log".
Error: /home/enord/PX4-Autopilot/build/cubepilot_cubeorangeplus_default is not a directory
make: *** [Makefile:227: cubepilot_cubeorangeplus_default] Error 1

EhrazImam, 2023-02-22:
Is this board supported with px4 1.11.3

ripdk12, 2023-02-22:
did you setup the development environment as by the guide in the [docs?](https://docs.px4.io/main/en/dev_setup/dev_env_linux_ubuntu.html)

EhrazImam, 2023-02-23:
@ripdk12 Thankyou it worked,
There is another issue i am facing

Issue details

Hi i am using Pixhawk 6C with px4(1.13.1) and CUBE orange plus px4(1.14.0)
When I’m changing the Pixhawk parameters in **QGC (SYS_MC_EST_GROUP = No position, SYS_HAS_BARO & SYS_HAS_MAG = 0)**, it only displays /mavros/imu/data_raw not /mavros/imu/data. With same parameter I have also tried on Pixhawk 4, Cube Orange and Cube Black with different firmware(1.11.0, 1.11.1, 1.12.0, 1.13.0, 1.13.1) and they all displaying /mavros/imu/data and /mavros/imu/data_raw as well.

MAVROS version and platform

Mavros: 1.14.0
ROS: Noetic
Ubuntu: 20.04
Autopilot type and version

[ ] ArduPilot
[ x ] PX4

It will be great if you can help me as i am getting it that the it is on all latest hardware but is there any way i can get that even i turned them off.

closed 2023-06-10 by JeongseokHyun