PX4/PX4-Autopilot #20517: Airspeed selector

astekachev, 2022-10-31:
Hi all!
I have 2 ms4525do speed sensors installed on my board V6X. When restarting the board by turning the power off and on, the board selects randomly either the first or the second sensor with the parameters specified:
ASPD_PRIMARY: First airspeed sensor
ASPD_FALLBACK_GW: Disable

There are also some questions about the displayed airspeed data in QGC - the sensor readings are strongly "jumping" at rest, jumps in readings up to +/- 15 km/h. At the same time the ASPD_FS_INTEG parameter is set to 50 m. How to make the speed sensor readings stop displaying jumping values? For example, as it is implemented for roll or pitch values - the accelerometer and gyroscope also output jumping values, but the final calculated pitch and roll values do not change at rest.

Best regards.

sfuhrer, 2022-11-02:
For the issue with 2 sensors not being consistently initialized/linked to the parameters: that's unfortunately not properly solved yet. What we would need is to have IDs for the airspeed sensors, and use that to define the primary, the scale, etc.
As a workaround I guess you could add a sleep in the startup script, to start one sensor clearly after the other. 

And then related to your second question (jumping values on the ground): the ms4525do is very sensitive to noise at a low airspeed. +/15km/h around 0km/h is quite common. You could propose to add a threshold on the display in QGC, e.g. only display airspeed data if |airspeed|>4m/s. On the flight controller side we rather don't want to do any filtering or value thresholding. 
ASPD_FS_INTEG is used for airspeed failure detection and not for filtering if you meant that.

astekachev, 2022-11-03:
Thank you very much for your comprehensive answer.
Later I will try to set the initialization delay for the airspeed sensors.