PX4/PX4-Autopilot #20519: No failsafe activated if offboard setpoint streaming is stopped

TedObrien, 2022-10-31:
## Describe the bug
No failsafe activated if offboard setpoint streaming is stopped, the drone will continue to use the last setpoint received and remain in offboard mode. Manual joystick input is also ignored.
## To Reproduce
Steps to reproduce the behavior:
1. on main branch using the micro-ros-agent
2. Begin an offboard script  that publishes trajectory_setpoint messages (such as https://github.com/Jaeyoung-Lim/px4-offboard)
3. stop the script (ctrl + c in terminal)
4. No failsafe message seen on QGroundcontrol, drone remains in offboard mode and using the on-screen joy stick in QGroundControl has no effect

## Log Files and Screenshots
log file: https://logs.px4.io/plot_app?log=ed7a2222-3de3-4eec-88ba-0c2ca5aa4225

Add screenshots to help explain your problem.
[image]

Velocity setpoints were sent for approximately 5 seconds and then stopped. However, the final velocity setpoint sent seems to be continued to be used, meaning the drone continued to descend in offboard mode despite not receiving new trajectory setpoint messages

## Drone (please complete the following information):
iris SITL gazebo simulation 

## Additional context
Add any other context about the problem here.

bkueng, 2022-11-02:
@TedObrien can you give https://github.com/PX4/PX4-Autopilot/pull/20535 a try?

TedObrien, 2022-11-02:
Hi @bkueng, that seems to have fixed it! although it's worth noting that once the failsafe is activated and the drone is in position mode, it seems to go straight back into offboard mode once setpoint streaming has resumed without commanding it to do so (it will not change into offboard mode from position mode if the drone has manually been put into position mode, only from the failsafe). Joystick input also does not trigger a failsafe which was the case in previous versions and is a good safety feature.

Many thanks for your help,

Ted

closed 2022-11-02 by dagar

bkueng, 2022-11-03:
> although it's worth noting that once the failsafe is activated and the drone is in position mode, it seems to go straight back into offboard mode once setpoint streaming has resumed without [...]

Yes the failsafe is cleared once the failsafe condition clears. This can be changed if you want.

> Joystick input also does not trigger a failsafe which was the case in previous versions and is a good safety feature.

This is handled here: https://github.com/PX4/PX4-Autopilot/blob/main/src/modules/commander/Commander.cpp#L2731.
Do you have that parameter set?