PX4/PX4-Autopilot #20524: Make target px4_sitl_rtps not found.

JetJie, 2022-10-31:
## Describe the bug
After git clone the latest version, I could not make the px4_sitl_rtps version, it shows px4_stil_rtps not found. but for the px4_sitl, it works well. I switch to the old stable version, like release/1.13 or release/1.12. I can make the px4_stil_rtps, but it is not successful.

## Log Files and Screenshots

xaveir@xaveir-desktop:~/PX4-Autopilot$ make px4_sitl_rtps
Makefile:523: *** "Make target px4_sitl_rtps not found. It either does not exist or px4_sitl_rtps cannot be the first argument. Use 'make help|list_config_targets' to get a list of all possible [configuration] targets.".  Stop.
[image]

beniaminopozzan, 2022-11-01:
The `micrortps` module, and thus all `rtps` targets have been removed (#20227). Now you have to use the `microdds` module and you can see here (https://github.com/Jaeyoung-Lim/px4-offboard) an example of how to interface it in ROS2.
Don't be scared by
> Currently need to be run from this https://github.com/PX4/PX4-Autopilot/pull/20227"

as that PR has already been merged.

The new documentation is WIP.

JetJie, 2022-11-03:
> The `micrortps` module, and thus all `rtps` targets have been removed (#20227). Now you have to use the `microdds` module and you can see here (https://github.com/Jaeyoung-Lim/px4-offboard) an [...]

I follow the example you mentioned
Should I install the micro-ros? It shows that micro-ros-agent: command not found.

beniaminopozzan, 2022-11-03:
Yes, you should. Depending on your system you can compile it from source or install it from snap or both.

JetJie, 2022-11-07:
> Yes, you should. Depending on your system you can compile it from source or install it from snap or both.

It works, thank you for your answer. And looking forward to a tutorial in the px4 website.

JetJie, 2022-11-08:
> The `micrortps` module, and thus all `rtps` targets have been removed (#20227). Now you have to use the `microdds` module and you can see here (https://github.com/Jaeyoung-Lim/px4-offboard) an [...]

is there any c++ example? or any tutorial?

junwoo091400, 2022-12-02:
Please check out the new documentation still pending: https://github.com/PX4/PX4-user_guide/pull/2094