PX4/PX4-Autopilot #20532: IO Unable to Open? [Pixhawk 6C]

8ball-ghost, 2022-11-01:
I've been trying to set up my Pixhawk 6C but whenever I run `px4io start` it aborts with an error stating “unable to communicate with IO.”  I have tried reflashing the firmware and updating px4io but nothing changes. Additionally, the IO's B/E orange light is constantly on when powered. I have looked at many of the similar issues in the past however none of the solutions have worked. 
Any help would be appreciated, thanks.

vincentpoont2, 2022-11-03:
@8ball-ghost Which firmware are you using? can you flash the latest stable firmware 1.13.1?

8ball-ghost, 2022-11-03:
@vincentpoont2 I have tried with both stable and master 1.13.1

vincentpoont2, 2022-11-04:
Do you have a GPS with safety switch? With it connected to FC, press and hold the safety switch then power on the FC. That will reset and update the IO. 

See if this fixes the issue.

julianoes, 2024-12-06:
No more follow up, closing.

closed 2024-12-06 by julianoes