PX4/PX4-Autopilot #20550: make px4_sitl jmavsim failed with fatal error: 'sprintf' is deprecated on Apple m1

iHrushikesh, 2022-11-05:
I am running the terminal as x86 (using rosetta). I am getting below error. Is there any workaround for 'sprintf' deprecation?

```
/Users/xxx/PX4-Autopilot/src/lib/sensor_calibration/Utilities.hpp:91:2: fatal error: 'sprintf' is deprecated: This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead. [-Wdeprecated-declarations]
        sprintf(str, "CAL_%s%u_%s", sensor_type, instance, cal_type);
        ^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk/usr/include/stdio.h:188:1: note: 'sprintf' has been explicitly marked deprecated here
__deprecated_msg("This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead.")
^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk/usr/include/sys/cdefs.h:215:48: note: expanded from macro '__deprecated_msg'
        #define __deprecated_msg(_msg) __attribute__((__deprecated__(_msg)))
                                                      ^
1 error generated.
[292/878] Generating uORB topic ucdr headers
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1
```

Pokemaster09, 2022-11-11:
I have exactly the same problem on my Macbook  M1 pro with Ventura 13.0, and I can nont  find workaround.

julesMindstorme, 2022-11-16:
Same issue here, but with terminal on Intel Mac running macOS Ventura

laborosh, 2022-11-16:
The 'sprintf' function is deprecated in the macOS 13 (Ventura) SDK.
You can resolve this issue in the following order.

1. Install the xcode command line tool (can be skipped if it already installed).
> xcode-select --install

2. Check the active Developer Directory Path
> select xcode -p

If the path is set to the Xcode app, change it to the command line tools path as follows:
> sudo xcode-select -s /Library/Developer/CommandLineTools

3. Change the macOS 13 SDK path to the macOS 12 SDK path.
The SDK folder in the downloaded command line tools (/Library/Developer/CommandLineTools/SDKs) contains the older SDK (macOS 12.3). The 'MacOSX12.sdk' file is the symbolic link of 'MacOSX12.3.sdk' folder.
You can change the SDK version by editing the 'CMakeLists.txt' file in the 'PX4-Autopilot' folder.
Add the following line to the first section of the 'CMakeLists.txt' file.

**set(CMAKE_OSX_SYSROOT "/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk")**

4. Delete the 'build' folder and rebuild again. Even thought you get some warning messages during compilation, maybe you can build successfully. : )

laborosh, 2022-11-16:
> I am running the terminal as x86 (using rosetta). I am getting below error. Is there any workaround for 'sprintf' deprecation?
>
> ``` [...]

The 'sprintf' function is deprecated in the macOS 13 (Ventura) SDK.
You can resolve this issue in the following order.

Install the xcode command line tool (can be skipped if it already installed).
xcode-select --install
Check the active Developer Directory Path
select xcode -p
If the path is set to the Xcode app, change it to the command line tools path as follows:

sudo xcode-select -s /Library/Developer/CommandLineTools
Change the macOS 13 SDK path to the macOS 12 SDK path.
The SDK folder in the downloaded command line tools (/Library/Developer/CommandLineTools/SDKs) contains the older SDK (macOS 12.3). The 'MacOSX12.sdk' file is the symbolic link of 'MacOSX12.3.sdk' folder.
You can change the SDK version by editing the 'CMakeLists.txt' file in the 'PX4-Autopilot' folder.
Add the following line to the first section of the 'CMakeLists.txt' file.
set(CMAKE_OSX_SYSROOT "/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk")

Delete the 'build' folder and rebuild again. Even thought you get some warning messages during compilation, maybe you can build successfully. : )

laborosh, 2022-11-16:
> I am running the terminal as x86 (using rosetta). I am getting below error. Is there any workaround for 'sprintf' deprecation?
>
> ``` [...]

The 'sprintf' function is deprecated in the macOS 13 (Ventura) SDK.
You can resolve this issue in the following order.

1. Install the xcode command line tool (can be skipped if it already installed).

> xcode-select --install

2. Check the active Developer Directory Path
> select xcode -p

3. If the path is set to the Xcode app, change it to the command line tools path as follows:
> sudo xcode-select -s /Library/Developer/CommandLineTools

4. Change the macOS 13 SDK path to the macOS 12 SDK path.
The SDK folder in the downloaded command line tools (/Library/Developer/CommandLineTools/SDKs) contains the older SDK (macOS 12.3). The 'MacOSX12.sdk' file is the symbolic link of 'MacOSX12.3.sdk' folder.
You can change the SDK version by editing the 'CMakeLists.txt' file in the 'PX4-Autopilot' folder.
Add the following line to the first section of the 'CMakeLists.txt' file.

set(CMAKE_OSX_SYSROOT "/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk")

5. Delete the 'build' folder and rebuild again. Even thought you get some warning messages during compilation, maybe you can build successfully. : )

wooi, 2022-11-28:
> The 'sprintf' function is deprecated in the macOS 13 (Ventura) SDK. You can resolve this issue in the following order.
>
> 1. Install the xcode command line tool (can be skipped if it already [...]

it's works for me

danylevskyi, 2023-01-27:
@laborosh 
Thank you! It helped.

kevinuav, 2023-02-15:
> The 'sprintf' function is deprecated in the macOS 13 (Ventura) SDK. You can resolve this issue in the following order.
>
> 1. Install the xcode command line tool (can be skipped if it already [...]

Thank you. The issue is fixed.

junwoo091400, 2023-03-05:
> it's works for me

Thanks for the solution! If this isn't in the main documentation / source code yet, could you make a PR for this? That would be very helpful!

closed 2023-03-05 by junwoo091400