PX4/PX4-Autopilot #20551: px4 [mavlink] Sarm_hardfault: Hard Fault:

lingbuweibu, 2022-11-06:
When debugging through the Qgroundcontrolstation console, this problem will occur, and the flight control will restart. This problem is very strange. Sometimes it's OK to add a line of code and compile it. It doesn't seem like a code problem, like a compiler problem.
INFO  [mavlink] Sarm_hardfault: Hard Fault:
arm_hardfault:   IRQ: 3 regs: 0x30007c04
arm_hardfault:   BASEPRI: 00000080 PRIMASK: 00000000 IPSR: 00000003 CONTROL: 00000000
arm_hardfault:   CFAULTS: 00008200 HFAULTS: 40000000 DFAULTS: 00000000 BFAULTADDR: 0000000b AFAULTS: 00000000
arm_hardfault: PANIC!!! Hard fault: 40000000
up_assert: Assertion failed at file:armv7-m/arm_hardfault.c line: 134 task: mavlink_rcv_if0
up_registerdump: R0: ffffffff 24000fe0 30000550 240051cc ffffffff 24000fe4 30000550 30007d50
up_registerdump: R8: 00000000 00000000 0000001b 00000046 0000000a 30007cd8 080231ff 080241f0
up_registerdump: xPSR: 01000000 BASEPRI: 00000080 CONTROL: 00000000
up_registerdump: EXC_RETURN: ffffffe9
up_dumpstate: sp:     24001290
up_dumpstate: IRQ stack:
up_dumpstate:   base: 24001320
up_dumpstate:   size: 00000320
up_dumpstate:   used: 00000188
up_stackdump: 24001280: 24001290 24004c90 30008068 08029ef9 00000080 00000000 00000046 0000000a
up_stackdump: 240012a0: 30007cd8 080231ff 080241f0 e000ed2c 0000000b e000ed2c 0000000b 00000000
up_stackdump: 240012c0: 00000000 00000000 00000000 0000001b 00000046 08021847 0802242d 0802154f
up_stackdump: 240012e0: 40000000 00000000 0000000b 00000000 24004c90 00000003 00000003 08022d27
up_stackdump: 24001300: 00000080 08020f59 00000080 30007c04 24000fe4 30000550 30007d50 080202d7
up_dumpstate: sp:     30007cd8
up_dumpstate: User stack:
up_dumpstate:   base: 30008068
up_dumpstate:   size: 00001134
up_dumpstate:   used: 000005ec
up_stackdump: 30007cc0: 00000000 3f000000 3ebef9dc 4031ef1f 20000010 000000f0 24000fe0 080231f1
up_stackdump: 30007ce0: 30000550 080231db 24000fe0 30000550 000000f0 08023011 38005750 38005750
up_stackdump: 30007d00: 00000000 0815e70b 00000000 00000000 00000000 30007de8 00000000 38005750
up_stackdump: 30007d20: 38005750 00000000 00000000 00000066 00000000 0000001b 00000000 0815d78f
up_stackdump: 30007d40: 38005750 080cc8ff 38005750 00000020 24004cb0 08023163 38005750 080229c7
up_stackdump: 30007d60: 24006a00 38005750 2000ea20 00000000 00000000 080cca55 2000ea38 30007ef4
up_stackdump: 30007d80: 0000004f 080ccadb 00000000 160a0000 00000a01 0a241000 3df2601d 4cac2288
up_stackdump: 30007da0: 73867008 27e5b81d 00000a1d e0d0c000 90501ff0 00001fde 0a244400 73cf6e1d
up_stackdump: 30007dc0: ff309edf 000064c0 00000080 00000100 00000000 08269000 30007ef4 2000ea38
up_stackdump: 30007de0: 2000f0b5 080d6f5f 00000000 2000ea20 2000ea38 080d78b5 0000001a 00000000
up_stackdump: 30007e00: deadbeef deadbeef 0000117f 00000000 0ac583f1 00000000 deadbeef deadbeef
up_stackdump: 30007e20: deadbeef deadbeef deadbeef deadbeef deadbeef 00000003 00000101 00000000
up_stackdump: 30007e40: 00000000 00000000 9edf73cf 64c0ff30 00800000 01000000 00000000 90000000
up_stackdump: 30007e60: 6066a026 22883df2 70084cac b81d7386 0a1d27e5 c0000000 1ff0e0d0 1fde9050
up_stackdump: 30007e80: 44000000 6e1d0a24 0ac59570 00000000 00000004 00000000 00000005 00000005
up_stackdump: 30007ea0: 00000000 00050000 00000000 00000000 deadbeef deadbeef deadbeef deadbeef
up_stackdump: 30007ec0: deadbeef deadbeef deadbeef deadbeef deadbeef deadbeef deadbeef deadbeef
up_stackdump: 30007ee0: deadbeef deadbeef deadbeef deadbeef deadbeef 4ffd66a0 ff5b0000 00007e00
up_stackdump: 30007f00: 00000000 160a0000 00000a01 0a241000 3df2601d 4cac2288 73867008 27e5b81d
up_stackdump: 30007f20: 00000a1d e0d0c000 90501ff0 00001fde 0a244400 73cf6e1d ff309edf 000064c0
up_stackdump: 30007f40: 00000080 00000100 00000000 00269000 00000000 00000000 00000000 00000000
up_stackdump: 30007f60: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
up_stackdump: 30007f80: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
up_stackdump: 30007fa0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
up_stackdump: 30007fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
up_stackdump: 30007fe0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
up_stackdump: 30008000: 00000000 00000000 000066a0 00000000 00000000 30000000 000000f0 00000000
up_stackdump: 30008020: 00000000 30000550 300006e0 2000d990 2000d9b4 00000000 00000000 00000000
up_stackdump: 30008040: 00000000 080d7a63 080d7a5d 08035793 00000000 00000000 00000000 00000000
up_stackdump: 30008060: 00000000 00000000 0003ff90 00001150 24004dd4 24004dc4 00000000 00010000
up_taskdump: Idle Task: PID=0 Stack Used=340 of 728
up_taskdump: hpwork: PID=1 Stack Used=332 of 1268
up_taskdump: lpwork: PID=2 Stack Used=332 of 1620
up_taskdump: init: PID=3 Stack Used=2276 of 2932
up_taskdump: wq:manager: PID=4 Stack Used=428 of 1260
up_taskdump: navigator: PID=396 Stack Used=1124 of 1772
up_taskdump: gps: PID=336 Stack Used=980 of 3420
up_taskdump: mavlink_if1: PID=350 Stack Used=1772 of 2740
up_taskdump: dataman: PID=31 Stack Used=796 of 1204
up_taskdump: mavlink_rcv_if1: PID=352 Stack Used=1300 of 4404
up_taskdump: wq:lp_default: PID=33 Stack Used=852 of 1924
up_taskdump: logger: PID=420 Stack Used=2436 of 3644
up_taskdump: wq:uavcan: PID=230 Stack Used=1876 of 3628
up_taskdump: wq:UART0: PID=360 Stack Used=740 of 1540
up_taskdump: uavcan_fw_srv: PID=233 Stack Used=1564 of 6004
up_taskdump: wq:nav_and_controllers: PID=298 Stack Used=1096 of 2244
up_taskdump: wq:rate_ctrl: PID=299 Stack Used=1056 of 1956
up_taskdump: log_writer_file: PID=428 Stack Used=388 of 1172
up_taskdump: wq:hp_default: PID=238 Stack Used=1252 of 1900
up_taskdump: mavlink_rcv_if0: PID=431 Stack Used=1516 of 4404
up_taskdump: commander: PID=305 Stack Used=1348 of 3220
up_taskdump: wq:UART1: PID=242 Stack Used=968 of 1540
up_taskdump: wq:UART6: PID=244 Stack Used=832 of 1540
up_taskdump: wq:UART4: PID=246 Stack Used=844 of 1540
up_taskdump: mavlink_if0: PID=313 Stack Used=1764 of 2828
[boot] Fault Log info File No 4 Length 3177 flags:0x01 state:0
[boot] Fault Logged on 2022-11-04-00:42:26 - Valid
[boot] There is a hard fault logged. Hold down the SPACE BAR, while booting to review!
sercon: Registering CDC/ACM s[hardfault_log] Fault Log info File No 4 Length 3177 flags:0x01 state:0
[hardfault_log] Fault Logged on 2022-11-04-00:42:26 - Valid
erial driver
Through disassembly, I found the following phenomena
(gdb) info line *0x080231ff
Line 329 of "semaphore/sem_holder.c"
   starts at address 0x80231f8 <nxsem_boostholderprio+8>
   and ends at 0x8023200 <nxsem_boostholderprio+16>.

  if (!nxsched_verify_tcb(htcb))

(gdb) info line *0x080241f0
Line 86 of "sched/sched_verifytcb.c"
   starts at address 0x80241f0 <nxsched_verify_tcb>
   and ends at 0x8024208 <nxsched_release_tcb>.
If you comment out the write function, there will be no restart,such as 
size_t MavlinkShell::write(uint8_t *buffer, size_t len)
{
//	return ::write(_to_shell_fd, buffer, len);
	return 0;
}
Who can help me see what's going on? This problem has been bothering me for a long time，
Code version v1.13.0-alpha1

lingbuweibu, 2022-11-06:
@bkueng any ideas?

bkueng, 2022-11-07:
Which board is this? Could it be this one https://github.com/PX4/PX4-Autopilot/pull/19724? (Not sure if that's included already)
What code changes do you have? In which modules?

lingbuweibu, 2022-11-07:
> Which board is this? Could it be this one #19724? (Not sure if that's included already) What code changes do you have? In which modules?

I use my own board, stm32h743vit6. This phenomenon often occurs. When I modify the code, I change the gps module, I change the radar driver, and I add the imu driver. Sometimes a line of code may not appear, although this line of code does not play an important role. Another phenomenon is that the download speed of my board is obviously slow after it has been downloaded to nearly 50%. The board is always slow until the download is completed. I download programs through USB.
The phenomenon that causes the board to restart using shell is the same as using mavlink_ info function matter

bkueng, 2022-11-08:
Can you update the bootloader to latest main? The slow flashing should be fixed there.

AlexKlimaj, 2022-11-08:
I've seen this many times on the Cube Orange. It will depend sometimes on whether I compile in native ubuntu or in WSL.

lingbuweibu, 2022-11-10:
@bkueng The bootloader was modified by myself following Cuav's example. I changed it according to the link you provided. Will it be able to restart the shell for a long-term test? However, the problem of slow download still exists when the download speed is close to 50%. I don't know what the problem is.
/* Pad out last section as the STM32H7 Flash write size is 256 bits. 32 bytes */
		. = ALIGN(16);
		FILL(0xffff)
		. += 16;
	} > AXI_SRAM AT > FLASH  = 0xffff

bkueng, 2022-11-11:
Please update the whole bootloader, I don't remember exactly when the flash speed was fixed.

lingbuweibu, 2022-11-11:
> Please update the whole bootloader, I don't remember exactly when the flash speed was fixed.

OK，I download the latest bootloader and try to change it。