PX4/PX4-Autopilot #20553: Microdds_client works for /fmu/out topics, but not for /fmu/in

max11gen, 2022-11-07:
## Describe the bug
I'm trying to get the new microdds_client to run with the updated offboard example from px4_ros_com. However, I can't get it running. The drone will never fly, nor arm. After some troubleshooting I realized that PX4 doesn't seem to be receiving messages from ROS2. When I do a `ros2 topic echo /fmu/in/offboard_control_mode` I can see that the offboard control node publishes the messages as expected. As you can see in the attached screenshot of the rqt_graph the topics are also wired correctly. But when I call `listener offboard_control_mode` in the px4-shell there are no messages and the listener reports that the topic was `never published`. When I run a `ros2 topic echo /fmu/out/sensor_combined`, I can see PX4's sensor messages normally. When I started `make px4_sitl gazebo` and then start the `micro_ros_agent` all the data writers are initialized as I see the following output in the PX4 shell:

> INFO  [microdds_client] successfully created rt/fmu/out/failsafe_flags data writer, topic id: 79
> INFO  [microdds_client] successfully created rt/fmu/out/sensor_combined data writer, topic id: 160
> INFO  [microdds_client] successfully created rt/fmu/out/timesync_status data writer, topic id: 180
> INFO  [microdds_client] successfully created rt/fmu/out/vehicle_attitude data writer, topic id: 198
> INFO  [microdds_client] successfully created rt/fmu/out/vehicle_control_mode data writer, topic id: 204
> INFO  [microdds_client] successfully created rt/fmu/out/vehicle_global_position data writer, topic id: 205
> INFO  [microdds_client] successfully created rt/fmu/out/vehicle_gps_position data writer, topic id: 207
> INFO  [microdds_client] successfully created rt/fmu/out/vehicle_local_position data writer, topic id: 211
> INFO  [microdds_client] successfully created rt/fmu/out/vehicle_odometry data writer, topic id: 216
> INFO  [microdds_client] successfully created rt/fmu/out/vehicle_status data writer, topic id: 221

I don't see anything similar for the data_readers. But looking at the code it seems that the data readers only report failing topics while the data writers report success for every topic that could be initialized. So that doesn't seem to be wrong, however, I find it to be a little confusing. Is there a reason for this?

## To Reproduce
1. run `make px4_sitl gazebo` in a shell
2. In another shell source a ROS2 workspace with the micro_ros_agent, px4_msgs and px4_ros_com packages installed.
3. In that shell run `ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888`
4. Start QGroundControl to be able to observe the drones behaviour better
5. In another shell source that workspace again and run `ros2 run px4_ros_com offboard_control`

## Expected behavior
I would expect to see the drone arming and taking off, but it does nothing.

## Log Files and Screenshots
https://logs.px4.io/plot_app?log=763d2029-174c-429a-920a-80a1116c9ba1

[image]

max11gen, 2022-11-07:
@Jaeyoung-Lim In the PR towards the new microdds_client architecture and px4_msgs integration I saw, that you got the offboard control running. Do you by any chance have any idea on what might be going wrong here?

Jaeyoung-Lim, 2022-11-07:
@max11gen Can you check if https://github.com/Jaeyoung-Lim/px4-offboard works for you?

max11gen, 2022-11-08:
@Jaeyoung-Lim That was actually the first thing I tried because I saw your comment in the PR. It was just afterwards that I realized that someone had already updated px4_ros_com, too. However, I just tried it again and it doesn't work either. Also, since you don't arm the vehicle in your code, I'm assuming you arm it via PX4 shell or QGC? But I tried that, too, and it didn't work. Any ideas?

JetJie, 2022-11-08:
> @Jaeyoung-Lim That was actually the first thing I tried because I saw your comment in the PR. It was just afterwards that I realized that someone had already updated px4_ros_com, too. However, I just tried it again and it doesn't work either. I attached a screenshot of rqt_graph. Also, since you don't arm the vehicle in your code, I'm assuming you arm it via PX4 shell or QGC? But I tried that, too, and it didn't work. Any ideas? [image]

in this https://github.com/Jaeyoung-Lim/px4-offboard . You first should takeoff and change the mode. I do it in the PX4 shell by command `commander takeoff` and then `commander mode offboard`. But this is a python node, I recently would like to create a c++ node.

max11gen, 2022-11-09:
> in this https://github.com/Jaeyoung-Lim/px4-offboard . You first should takeoff and change the mode. I do it in the PX4 shell by command `commander takeoff` and then `commander mode offboard`. But [...]

Thanks a lot, this actually works. My bad, I didn't look at it properly. But the problem with the px4_ros_com example persists.

JetJie, 2022-11-11:
> Thanks a lot, this actually works. My bad, I didn't look at it properly. But the problem with the px4_ros_com example persists.

You're right. I'm just waiting for someone to fix it.

Jaeyoung-Lim, 2022-11-11:
@JetJie This is an opensource project. Please consider adding a fix if you think something is not working

coen132, 2022-11-16:
@JetJie I experienced the same thing and looked into it. I logged the subscriber topics and noticed it gets published under a different topic id then what it should be: 
[image]
The internal topics are different so I guess that is were the error is