PX4/PX4-Autopilot #20554: SET_ACTUATOR and SET_SERVO in Mission

BrZ-Alex, 2022-11-07:
## set_servo
Waypoint with this command has ignored by drone. It doesnt work. I found the same issues in questions on forums, but there are no answers. 
In the source code I found the only place where this command is handle in mission_block.cpp: 
```
if (item.nav_cmd == NAV_CMD_DO_SET_SERVO) {
		PX4_INFO("DO_SET_SERVO command");

		// XXX: we should issue a vehicle command and handle this somewhere else
		actuator_controls_s actuators = {};
		actuators.timestamp = hrt_absolute_time();

		// params[0] actuator number to be set 0..5 (corresponds to AUX outputs 1..6)
		// params[1] new value for selected actuator in ms 900...2000
		actuators.control[(int)item.params[0]] = 1.0f / 2000 * -item.params[1];

		_actuator_pub.publish(actuators);
}
```
But as I see, its wrong implementation and #10320 wrote about this 4 years ago. So I dont know, how it can be working. 

## set_actuator
When I try to upload mission with this command in QGC, I receive the message "Ignoring mission item, invalid Item" and "Mission transfer failed. Error: Command is not supported. Item 1 Command: Set actuator"
The same issue I get when uploading mission in my own app with MAVSDK. It was expected because command MAV_CMD_DO_SET_ACTUATOR should use COMMAND_LONG frame instead of COMMAND_INT (I think the problem is this).

## conclusion
DO_SET_SERVO doesnt work at all, drone has no reaction to this command. DO_SET_ACTUATOR cant be used in Mission.
So, how can I use servo (AUX1, for example) in Mission mode? 
Thanks.

Drone firmware: 1.13.1, Quadrotor X
QGC: Daily master:9c77d15b2 2022-10-25

P.S. Servo work via RC and with setActuator command in MAVSDK. So the problem only in mission

bkueng, 2022-11-08:
I added the `SET_ACTUATOR` support for mission in https://github.com/PX4/PX4-Autopilot/pull/20325. Can you try with current PX4 `main`?

BrZ-Alex, 2022-11-08:
@bkueng Oh, it turns round that in 1.13.1 there is no `SET_ACTUATOR` mission support. Am I right? 

So, how can I build a firmware from the source code? Or what should I do to try the current PX4 `main`? 
Im sorry for this stupid question, Im really confused.

bkueng, 2022-11-08:
Yes but we can also backport it as it's a small change.
No worries, you can either build it yourself (https://docs.px4.io/main/en/dev_setup/building_px4.html), or flash the development version (https://docs.qgroundcontrol.com/master/en/SetupView/Firmware.html).

BrZ-Alex, 2022-11-11:
Thank you a lot! Now DO_SET_ACTUATOR is works correctly. 

But I have to use #20567 commit because QGC cant connect to drone with firmware based on any other commit. (QGC shows "vehicle did not respond to request for parameters" error). 

The question about MAVSDK is still open. In MissionRaw.MissionItem param5 and param6 must be Int, so we can use AUX5&6 only with values -1,0,1. Also you cant even set NaN value to this params. But Im not sure that this issue should be discussed there and not in MAVSDK repo

junwoo091400, 2022-11-11:
@BrZ-Alex I proposed removing the DO_SET_SERVO command entirely, as explained here: https://github.com/PX4/PX4-Autopilot/pull/20392

Would DO_SET_ACTUATOR fulfill your need? Or would DO_SET_SERVO still be necessary?

BrZ-Alex, 2022-11-12:
@junwoo091400 I described the problem with DO_SET_ACTUATOR above, you cant use AUX 5 and 6 except for -1, 0, 1 vaules. And you cant use another AUX without setting values to AUX 5/6 in MAVSDK mission. DO_SET_SERVO can solve this problem, but it doesnt work at the moment.