PX4/PX4-Autopilot #20555: Thrust Dropout when Switching from Manual to Altitude Mode

Finwood, 2022-11-07:
## Description
When transitioning from manual mode into altitude mode, the thrust drops out for a short moment. It recovers quickly, but the dropout still causes additional load and does not seem intentional.

## To Reproduce
Steps to reproduce the behavior:
1. power the drone, and switch into manual mode
2. arm and take off in manual mode
3. switch into altitude mode
4. observe error

## Expected behavior
The transition from manual to altitude mode should be smooth.

## Log Files and Screenshots
[Logfile](https://logs.px4.io/plot_app?log=55012ee5-f3a2-4a67-8419-052bc6378ba5)

[image]

## Drone
This dropout was observed on a haxacopter with a takeoff mass of over 10 kg, running on an Auterion Skynode.

## Additional context
During further analysis, I have found a related anomaly in the [vehicle_local_position_setpoint](https://docs.px4.io/main/en/msg_docs/vehicle_local_position_setpoint.html) message. After switching to altitude mode, the first (logged) acceleration setpoint commands ~10g downwards, which explains the low thrust value:

[image]

This acceleration value of 100 m/s² downward also appears in the position controller logic to inhibit thrust after ground contact:
https://github.com/PX4/PX4-Autopilot/blob/2e8918da66af37922ededee1cc2d2efffec4cfb2/src/modules/mc_pos_control/MulticopterPositionControl.cpp#L375

Finwood, 2022-11-11:
After going through other flights this year, I have found a few more examples showing the exact same behavior:

- [2022-03-15, Flight 9WY3KV](https://logs.px4.io/plot_app?log=2b73c1d1-2cc2-4c90-b1db-86345f4feb1c#Nav-Actuator-Controls)
- [2022-03-23, Flight NVMRJW](https://logs.px4.io/plot_app?log=d0afe7b7-85ef-4350-bbd6-ff85afa1f56b#Nav-Actuator-Controls)
- [2022-03-23, Flight KPKAZ1](https://logs.px4.io/plot_app?log=5b5f0488-73b3-45f8-b357-74bd9bb0698b#Nav-Actuator-Controls)
- [2022-03-23, Flight YZ5BEK](https://logs.px4.io/plot_app?log=e072bbd4-4be4-4964-b059-5515845afb7b#Nav-Actuator-Controls)
- [2022-06-15, Flight 5BQM8R](https://logs.px4.io/plot_app?log=93b1d41b-306d-4da5-8bd3-c10c9d978732#Nav-Actuator-Controls)
- [2022-06-15, Flight FF5B64](https://logs.px4.io/plot_app?log=a41e3755-d960-4ddf-9467-79b40f2672a0#Nav-Actuator-Controls)
- [2022-09-02, Flight C44JRW](https://logs.px4.io/plot_app?log=82296a5a-5768-4bed-8cda-21050e0a2e2e#Nav-Actuator-Controls)

<details>
<summary>expand to see images of the flights above</summary>

[image]

[image]

[image]

[image]

[image]

[image]

[image]
</details>

A few random remarks:
- the thrust dropout occurs at most once per flight, at the first switch to Altitude or Position mode
- the dropout may occur multiple times per boot, as can be seen with flights NVMRJW and KPKAZ1
- at all occasions, the field `vehicle_local_position_setpoint.acceleration[2]` had an initial (logged) value of 100 m/s²

MaEtUgR, 2022-11-14:
@Finwood Thanks a lot for your exemplary bug report! Note that you are running Auterion PX4 2.5. The problem you describe was solved by https://github.com/PX4/PX4-Autopilot/pull/19162 and does not persist anymore on more recent versions. If you are looking for a patched Auterion PX4 2.5 version please contact the company directly. The fix was not extensively backported because even though it's not nice it's also not safety critical. Please let me know if any PX4 stable version needs to be patched.

Finwood, 2022-11-17:
> The problem you describe was solved by #19162 and does not persist anymore on more recent versions.
That is good to hear, thank you for clarifying!

> If you are looking for a patched Auterion PX4 2.5 version please contact the company directly.
I am still waiting for a reply from the company, but that is unrelated to _this_ issue.

As far as I am concerned this issue is resolved. Thanks! 🎉

closed 2022-11-17 by Finwood