PX4/PX4-Autopilot #20580: Geocoordinate projection errors by assuming the earth is a sphere

Jaeyoung-Lim, 2022-11-09:
## Describe the bug
The current Map projection assumes that the earth is a sphere, 

https://github.com/PX4/PX4-Autopilot/blob/9246d38667311b8f53c3b38a13e35bf4a3969f70/src/lib/geo/geo.cpp#L71-L92

However, this can result in large errors when projecting WGS84 coordinates to a tangential local coordinate frame.
The projection should be done through a geodetic to cartesian conversion as in https://proj.org/operations/conversions/cart.html#cart

## To Reproduce
Steps to reproduce the behavior:
1. Run SITL (`make px4_sitl gazebo_plane`) with the following branch: https://github.com/PX4/PX4-SITL_gazebo/tree/pr-geographic-lib
2. Fly out 20km or more and compare the geoprojection
3. Compare the projection errors using Geographiclib using topocentric projection vs projection using spherical earth assumptions

## Expected behavior
A clear and concise description of what you expected to happen.

## Log Files and Screenshots
The figure below shows the difference between the altitude differences when flying 20km out between the spherical assumption, and a conversion using geographic lib
Logged topics show
[image]

*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/main/en/getting_started/flight_reporting.html)).
- Upload the log to the [PX4 Flight Review](http://logs.px4.io/)
- Share the link to the log (Copy and paste the URL of the log)

murundb, 2024-02-03:
Is there any follow up on this topic? This seems like an important issue to be resolved.

mcsauder, 2024-02-03:
Has the earth spheroid been updated since 1984?

mcsauder, 2024-02-03:
You know.... wgs 84?

mcsauder, 2024-02-03:
Once in a while you need an old guy to remind you that the earth is not burning down. Someone born before 1980.

mcsauder, 2024-02-04:
I don't agree, but that's fine. :)

murundb, 2024-02-04:
Which part you don't agree with?

mcsauder, 2024-02-04:
It's not new news that the earth is a spheoid, not a sphere. What do mean?

mcsauder, 2024-02-04:
It's a thing that happens because the earth has mass and spins on an axis. It's wider around the middle than it is from top to bottom.

mcsauder, 2024-02-04:
If it's not wgs84 and you need to fly more than 300miles, submit a pr. And yes,the earth is not a sphere. @Jaeyoung-Lim is quite capable of adding the math, he is as sharp as they get. More than 300miles is military and you can get yourself into trouble quick.

murundb, 2024-02-04:
I still don't understand what you don't agree with or what you are trying to tell me. `MapProjection::project()` is not doing standard curvilinear (with respect to WGS84 reference ellipsoid) to NED local tangent plane. Assuming you have established the reference origin of your local tangent plane, you would do transformations from curvilinear to ECEF to NED local tangent plane.

I am not saying that there is something wrong with 'MapProjection::project()'. For all I know, it might have been the original intent to not use WGS84 datum. Another observation is that for the standard conversion, you'd use the altitude above the reference ellipsoid, but in `MapProjection::project()`, altitude above mean sea level is being used.

mcsauder, 2024-02-04:
Can you do it with wgs84?