PX4/PX4-Autopilot #20582: Smart battery reports exactly half real current/voltage using bq40z80

Woody363, 2022-11-09:
When reading the values on PX4 the reported current and voltage are half of the real values. Temperature reports correctly, other bq40z80 values are, so far, uncalibrated.

We are using smart battery module with its default settings. BAT_V_DIV and BAT_A_PER_V can be changed but are not used in smart battery and do not effect the values.

We have a battery using bq40z80 management chip. The current and voltage on the chip have both been calibrated using TI's own bqstudio software. Latest firmware of 4800_0_04 is on the bq40z80.

AlexKlimaj, 2022-11-12:
The scale factor on the bq40z80 is probably set to 2.

Woody363, 2022-11-18:
Thanks for the suggestion, but I've checked and the scale factor isn't set (it's 0?) and there is only a scale factor for the current, not the voltage anyway.

For now I'm just calibrating to double the voltage and current.

I have also since found this post: https://e2e.ti.com/support/power-management-group/power-management/f/power-management-forum/946102/bq40z80-bq40z80-current-scaling

Which says there was a bug with TI's firmware showing double real voltage two years ago. Maybe this library was written with this firmware bug, and now they've fixed it it's now PX4 that's buggy?

AlexKlimaj, 2022-11-19:
It could be that your gauge is calibrated to the wrong voltage. Have you checked it in BQstudio?

Woody363, 2022-11-25:
Hi Alex, yes the gauge was originally calibrated using bqstudio, the issue I'm having is that the voltage and current reported in bqstudio is double the voltage reported in PX4. 

I've noticed that it's only the "Cell Voltage" from bqstudio that is important for the voltage calibration seen from PX4, and also looking at the battery status from Mavlink only shows three cell voltages rather than 6 (we are using a 6S battery) followed by a load of 0's and 65535's. [image] Note, the three voltages are double real values as temporary fix

 I think this explains why the voltage is halved. 6 cells are selected on the bq40z80 chip and in PX4 under Vehicle Setup -> Power so I'm unsure why this is the case. Will report back if I discover anything further.

Still not sure why Current is halved

Woody363, 2022-11-29:
Update: 
I have discovered that there should be a parameter of BAT1_C_MULT which I assume should be able to fix the current factor of two that we're experiencing.  https://github.com/PX4/PX4-Autopilot/blob/main/src/drivers/batt_smbus/parameters.c

However, the BAT1_C_MULT parameter isn't accessible on our build, not visible in the parameters list or in parameter list via mavlink ""param show ". We have tried px4 versions 12.3 13.0 and 13.2 and in all cases we're experiencing the same error as here: https://discuss.px4.io/t/wrong-power-tab-errors/25747 it seems like there is some deeper issue with the smart battery module which I don't know how to begin with.

Conclusion: 
Voltage halving is probably due to only first three cell voltages (out of six) getting read by PX4. Fix unknown, but doubling the cell voltage calibration on the bq40z80 is a temporary work around.

Current halving is probably due to an issue with the BAT1_C_MULT parameter not being set properly. Changing the multiplier on the BQ40Z80 to match is a temporary workaround.

closed 2022-11-29 by Woody363