PX4/PX4-Autopilot #20593: Problems with the full configuration parameters

bigworldsmall, 2022-11-11:
## Describe the bug
Hello everyone, we are trying to analyze PX4 by static analysis and found some problems with the full configuration parameters in the [PX4 User Guide](https://docs.px4.io/main/en).

e.g.
1. Defined in the configuration file(xxx\parameters.c), but not used in the source code
<html xmlns:v="urn:schemas-microsoft-com:vml" xmlns:o="urn:schemas-microsoft-com:office:office" xmlns:x="urn:schemas-microsoft-com:office:excel" xmlns="http://www.w3.org/TR/REC-html40">
<head>

<meta name=Generator content="Microsoft Excel">

</head>
<body>

COM_RCL_ACT_T
DSHOT_TEL_CFG
NAV_AH_ALT
NAV_AH_LAT
NAV_AH_LON
LPE_FGYRO_HP
LPE_X_LP
MBE_ENABLE
XRCE_DDS_UDP_PRT
MPC_VELD_LP
RBCLW_BAUD
SDLOG_MODE
IMU_GYRO_CAL_EN
SENS_EN_ADIS164X
SENS_EN_LL40LS
SENS_EN_PCF8583
SENS_EXT_I2C_PRB
SENS_INT_BARO_EN
SYS_CAL_ACCEL
SYS_CAL_BARO
SYS_CAL_GYRO
TEST_D
TEST_DEV
TEST_D_LP
TEST_HP
TEST_I
TEST_I_MAX
TEST_LP
TEST_MAX
TEST_MEAN
TEST_MIN
TEST_TRIM
UART_ESC_T_YAWR
SENS_OR_ADIS164X

</body>

</html>

2. Is defined in file(xxxxx\module.yaml), but we don't know where it is used in the source code
<html xmlns:v="urn:schemas-microsoft-com:vml" xmlns:o="urn:schemas-microsoft-com:office:office" xmlns:x="urn:schemas-microsoft-com:office:excel" xmlns="http://www.w3.org/TR/REC-html40">
<head>

<meta name=Generator content="Microsoft Excel">

</head>
<body>

GPS_1_CONFIG
GPS_2_CONFIG
ISBD_CONFIG
MAV_0_BROADCAST
MAV_0_CONFIG
MAV_0_FLOW_CTRL
MAV_0_FORWARD
MAV_0_MODE
MAV_0_RADIO_CTL
MAV_0_RATE
MAV_0_REMOTE_PRT
MAV_0_UDP_PRT
MAV_1_BROADCAST
MAV_1_CONFIG
MAV_1_FLOW_CTRL
MAV_1_MODE
MAV_1_RADIO_CTL
MAV_1_RATE
MAV_1_REMOTE_PRT
MAV_1_UDP_PRT
MAV_2_BROADCAST
MAV_2_CONFIG
MAV_2_FLOW_CTRL
MAV_2_MODE
MAV_2_RADIO_CTL
MAV_2_RATE
MAV_2_REMOTE_PRT
MAV_2_UDP_PRT
MAV_USEHILGPS
MSP_OSD_CONFIG
RBCLW_SER_CFG
SENS_CM8JL65_CFG
SENS_CM8JL65_R_0
SENS_LEDDAR1_CFG
SENS_SF0X_CFG
SENS_TFLOW_CFG
SENS_TFMINI_CFG
SENS_ULAND_CFG
RC_CRSF_PRT_CFG
RC_PORT_CONFIG
TEL_FRSKY_CONFIG
TEL_HOTT_CONFIG
MXS_SER_CFG
UWB_PORT_CFG

</body>

</html>

3. We could not find the information that it was defined and used
<html xmlns:v="urn:schemas-microsoft-com:vml" xmlns:o="urn:schemas-microsoft-com:office:office" xmlns:x="urn:schemas-microsoft-com:office:excel" xmlns="http://www.w3.org/TR/REC-html40">
<head>

<meta name=Generator content="Microsoft Excel">

</head>

<body>

ctl_bw
ctl_dir
ctl_gain
ctl_hz_idle
ctl_start_rate
esc_index
id_ext_status
int_ext_status
int_status
mot_i_max
mot_kv
mot_ls
mot_num_poles
mot_rs
mot_v_accel
mot_v_max
gnss.dyn_model
gnss.old_fix_msg
gnss.warn_dimens
gnss.warn_sats
uavcan.pubp-pres
SER_GPS1_BAUD
SER_GPS2_BAUD
SER_GPS3_BAUD
SER_PPB_BAUD
SER_RC_BAUD
SER_TEL1_BAUD
SER_TEL2_BAUD
SER_TEL3_BAUD
SER_TEL4_BAUD
SER_URT6_BAUD
SER_WIFI_BAUD

</body>

</html>

bkueng, 2022-11-11:
Most of those are used in startup scripts under https://github.com/PX4/PX4-Autopilot/tree/main/ROMFS. The first ones (small case) under 3. are for uavcan.
And some could possible be unused indeed.

junwoo091400, 2022-11-11:
Hi! How are you doing the static analysis? Would like to try that out & remove unnecessary parameters as well :hugs:

bigworldsmall, 2022-11-15:
> Most of those are used in startup scripts under https://github.com/PX4/PX4-Autopilot/tree/main/ROMFS. The first ones (small case) under 3. are for uavcan. And some could possible be unused indeed.

Thank you for your answer, I was wondering how it uses configuration options in file(xxxxx\module.yaml)，for example SENS_TFMINI_CFG in \src\drivers\distance_sensor\tfmini\module.yaml. And from the perspective of user configuration, is the configuration option in startup scripts not able to be changed by the user.

bigworldsmall, 2022-11-15:
> Hi! How are you doing the static analysis? Would like to try that out & remove unnecessary parameters as well 🤗

In our opinion, the configuration options in the official manual may be incomplete or incorrect, so we would like to be able to extract all of its configuration options directly from the source code. By comparing experimental data with official documents, the above problems were discovered.

MaEtUgR, 2022-11-15:
@bigworldsmall The same content was posted on Discord here: https://discord.com/channels/1022170275984457759/1040487659560239184/1040487670377349150

I posted about:
`COM_RCL_ACT_T`: good dicovery https://github.com/PX4/PX4-Autopilot/pull/20612
And the autogenerated serial configuration parameters: check https://github.com/PX4/PX4-Autopilot/pull/10337

bigworldsmall, 2022-11-16:
> @bigworldsmall The same content was posted on Discord here: https://discord.com/channels/1022170275984457759/1040487659560239184/1040487670377349150
> 
> I posted about: `COM_RCL_ACT_T`: good dicovery #20612 And the autogenerated serial configuration parameters: check #10337

What about the other configuration options in 1.
> `NAV_AH_ALT`
`NAV_AH_LAT`
`NAV_AH_LON`
...

MaEtUgR, 2022-11-16:
@bigworldsmall Some more might be deprecated. We can go through them. Sorry for closing.

dagar, 2022-11-16:
XRCE_DDS_UDP_PRT is in flux with more microdds updates coming, I'll take care of it.

bigworldsmall, 2022-11-17:
> The **???** ones e.g. `XRCE_DDS_UDP_PRT` I didn't really find exactly. Maybe someone can complement. #20616 got merged quicker than I thought so I opened #20618 in addition.
> 
> @bigworldsmall Could you double-check your list based on my references? I think there are generally a lot of serial configuration parameters that are a bit harder to spot but indeed in use. At least we found 5 unused parameter definitions thanks to your question.

Thanks for your good job, I double-check my list based on your references, and totally agree with you. Configuration options like `MPC_VELD_LP, LPE_X_LP`, are really hard to spot. `RBCLW_BAUD` in your references is empty, it seems not to be used anywhere.

PR #20612 (2022-11-16): commander_params: remove deprecated COM_RCL_ACT_T
## Describe problem solved by this pull request
@Curry reported on Discord some results of running static analysis and to my surprise `COM_RCL_ACT_T` was in the list of unused parameters: https://discord.com/channels/1022170275984457759/1040487659560239184/1040487670377349150

And duplicate here: fixes #20593

## Describe your solution
I remove the definition of `COM_RCL_ACT_T` because https://github.com/PX4/PX4-Autopilot/commit/455b885f86754985fb07726b7975f0f028537204 deprecated the parameter and failsafe delays got consolidated to `COM_FAIL_A [... cut]
PR #20616 (2022-11-16): Remove needless parameter definitions
## Problem
@bigworldsmall statically analyzed parameter use in PX4 and found quite some unused parameter definitions.

## Solution
I'm removing deprecated or otherwise needless parameters. See the table in:

closes #20593