PX4/PX4-Autopilot #20599: I2C port In/Out configuration issue

synerex2016, 2022-11-12:
Hi

I'm now talking about I2C port In/Out configuration.

In fact, we're faced on mass production of the LTE module integrated RTK GPS device. As you can imagine that the LTE module can be able to receive the RTCM data from the network RTK base stations automatically, so our new device deliver the RTCM data to PX4 thru I2C port.

However, we found out whenever we make a connection between PX4 and f9p, **PX4 make a configuration of I2C port In/Out is set to None(Disable) since stable version 1.13.1.** Therefore, our device cannot deliver the rtcm data coming from LTE modem to F9p.

Thus, Modification is required so that the setting for I2C does not change during F9P port setting in PX4. Or, we need an additional options.

Anyhow, we would like to know why PX4 make a configuration of I2C port in/out is set to None(Disable) since stable version 1.13.1?

Many thanks,
JD, Park

For your understanding, the following are our product features.

[image]
[image]

lfeng17, 2022-11-14:
Same problem.

bkueng, 2022-11-15:
I2C was disabled in https://github.com/PX4/PX4-GPSDrivers/pull/81. It might still be fine if we allow RTCM as input. @Igor-Misic can you have a look?

synerex2016, 2022-12-09:
Our product is working fine when tested by **changing** the F9P I2C setting in the GPS driver of PX4. F9P I2C port setting were performed by setting In : RTCM3, Out : NMEA.
**_Therefore, can you modify the PX4 GPS driver to set the F9P I2C port to In: RTCM3, Out: NMEA in default?_**

And while our testing, we found out that the GPS driver sets the measurement rate of **F9P to 10Hz**, but we think that part seems to be wrong.

According to the specification of F9P datasheet, The rate is as follows:
[image]
[image]

That is, when used as single GNSS, it is 7Hz and when used as a moving base, it is 5Hz.

**_Is this also being modified later?_**

Igor-Misic, 2022-12-13:
We disabled all communication at I2C input since F9P has an issue with the heading when it receives too much data to process. I'll try to enable RTCM only at I2C and check if the heading is still working. If not I'll add a parameter option to select a port for RTCM data.

synerex2016, 2022-12-14:
Many thanks for your great efforts. If possible, please enable IN:UBX+RTCM and OUT:UBX+NMEA at I2C.
We will waiting for your any feedbacks. 

> We disabled all communication at I2C input since F9P has an issue with the heading when it receives too much data to process. I'll try to enable RTCM only at I2C and check if the heading is still working. If not I'll add a parameter option to select a port for RTCM data.

bkueng, 2022-12-19:
@synerex2016 Igor added a configuration option in https://github.com/PX4/PX4-Autopilot/pull/20788.

closed 2022-12-19 by bkueng

synerex2016, 2022-12-21:
@bkueng @Igor-Misic Thanks for you guys great efforts! Keep in touch!