PX4/PX4-Autopilot #20600: Problem with respond telemetry from HereLink to Pixhawk 6x

rvjenya, 2022-11-13:
PX4 v1.13.1 - Stable Release
Hardware - Holybro Pixhawk 6x 
SmartController - Hex HereLink 

Hi guys, I'm trying to connect the HereLink system to Pixhawk 6x - but I don't respond to telemetry data from TELEM1 and TELEM2. I tried to change bartered for these ports, changing the pins to connectors. But I've got the same result - no Telemetry. What I'm doing wrong?

 I tried to use telem2 and changed baudrate but had result no telemetry. and I use XBP9X Radio in TELEM1, TELEM2, TELEM3 - but I had the same result.

Interesting fact... Ardupilot works normally, and I get telemetry from FC to HereLink.

I tried change FC to CubeOrange and it worked on PX4.
I have 4 FC Pixhawk 6x and this situation for all 6 fc when I connect HereLink. 

Maybe px4 1.13.1 version for Pixhawk 6x with problems?

![5B908DD3-43E6-42E7-A525-739A4F409210](https://user-images.githubusercontent.com/1352540/201522157-837ce4c6-bca2-48
[image]
[image]
79-a6fa-384ac7194df8.jpeg)
[image]

rvjenya, 2022-11-14:
I've found solution. I set “MAV_0_FLOW_CTRL” to “Force off”.

[image]
[image]

AlexKlimaj, 2022-11-14:
Try re-seating the v6x module. And or try a different telemetry port. Sometimes the module board to board connectors don't make full contact.

FriedrichTuttas, 2022-12-13:
I have a similar problem. TELEM3 works without any problems. However, the ports labeled TELEM1 and TELEM2 do not. When I plug in my [Drotek Telemetry Module](https://www.mybotshop.de/Drotek-Radio-Telemetry-433Mhz-3DR), QGC does not connect (and the little red LED of the telemetry module that is supposed to indicate data transfer does not light up). Only when pushing against the connector from different directions, every now and then data is sent and QGC connects to the Pixhawk 6X. I could confirm this with an oscilloscope. The TX pin of my TELEM3 port shows a continuous stream of data. The TX pins of the TELEM1 and TELEM2 ports only show data when I push against the connectors. But I have not yet figured out the right direction to push the connectors. Seeing data coming through the TX pins is very unreliable and undpredictable.

FriedrichTuttas, 2022-12-13:
Never mind. It is not a hardware issue in my case. A while back, I wrote a module that communicates via TELEM3 port. Today, I altered it to use TELEM1 port instead to test the TELEM1 port and I get the following weird behavior:
After booting, there is no data coming through the TX pin of TELEM1. Then I start my module via the MAVLink console in QGC by typing "my_module_name start". Within the start function of my module, it opens the corresponding uart port of TELEM1, but, at this point, it does not write any data! However, as soon as I start my module, I can see data coming through the TX pin of TELEM1 in my oscilloscope with a baudrate of 57600. Now, I can stop my module again by typing "my_module_name stop" and the data will keep coming.

So, opening the uart port in a different module must somehow enable MAVLink to use this port. I have no idea where to look to fix this issue.

james-avetics, 2023-03-10:
I have to set MAV_0_FLOW_CTRL to 0 (Force off) and it works with herelink or any mavlink serial TTL device plugged in to TELEM1.

closed 2025-10-26 by github-actions[bot]