PX4/PX4-Autopilot #20608: Parameter Category and Group

theojaasp, 2022-11-15:
How can i create a new costuç paramter category and group? 
So when I open paramter editor I can find easier needed parameters. 
I changed in the params.c files in PX4-Autopilot/src/ but when I compile there is no changes. 
But in PX4-Autopilot/build/parameters.xml I can see that the parameter I have changed the group are having the group and category assinged. 
But in QGC it is not.

junwoo091400, 2022-11-22:
That `parameters.xml` needs to be recognized by QGC. For that, there's 2 different ways:

1. Using `COMPONENT_METADATA` (which fetches the Parameters metadata from the vehicle)
2. QGC fetching the parameters xml while uploading the Firmware file (.px4) to the target

Are you using real hardware (Did you flash .px4 via QGC)? Or a SITL (it should utilize the first method by default)

You can read about this article to understand the second option:
https://beginnerjunwoo.wordpress.com/2020/08/23/px4-deepdive-2-all-about-px4s-airframes-xml-parsing-and-where-its-used-qgc-px4/

theojaasp, 2022-12-02:
@junwoo091400 Thanks for your answer.. I am Uploading the Firmware file (.px4) to the target board and not using simulation. 
In parameters.xml file which is generated the created group exists like I have attached here.
The group I am trying to add is named Flying

[image]

The file where I changed the group name for the used parameters is PX4-Autopilot/src/modules/fw_att_control.c and looks as follows

`/**
 * Max manual roll
 *
 * Max roll for manual control in attitude stabilized mode
 *
 * @unit deg
 * @min 0.0
 * @max 90.0
 * @decimal 1
 * @increment 0.5
 * @group Flying
 */
PARAM_DEFINE_FLOAT(FW_MAN_R_MAX, 45.0f);

/**
 * Max manual pitch
 *
 * Max pitch for manual control in attitude stabilized mode
 *
 * @unit deg
 * @min 0.0
 * @max 90.0
 * @decimal 1
 * @increment 0.5
 * @group Flying
 */
PARAM_DEFINE_FLOAT(FW_MAN_P_MAX, 45.0f);
`

How can I get QGC to display the group? 
I although did add a new Airframe, but that is visible and useable. There are no problems so far. 
 Only the Parameter Group is not showing up. Maybe there is a step I made wrong.

theojaasp, 2022-12-07:
I found the way to do it inside QGC Source code. 
The link you send me helped a lot
Thanks for you help.

andrewaKEF, 2022-12-07:
@theojaasp what was your solution? I am also running into this problem

theojaasp, 2022-12-10:
> @theojaasp what was your solution? I am also running into this problem

@andrewaKEF  I am using QGroundControl from souorce code. I did not find a way to make it done in PX4-Firmware...
I just played around with the Categorization, Groups and Classes in the /qgroundcontrol/src/FirmwarePlugin/PX4/PX4ParameterFactMetaData.xml file.
I don't know if that will work in usual but for me it works.

@dagar  Is there an ther way to do it?
Maybe something where we can favoritize our parameters so we have it easier when flying to change them?

theojaasp, 2022-12-11:
> @theojaasp what was your solution? I am also running into this problem

@andrewaKEF  I am using QGroundControl from souorce code. I did not find a way to make it done in PX4-Firmware...
I just played around with the Categorization, Groups and Classes in the qgroundcontrol/src/FirmwarePlugin/PX4/PX4ParameterFactMetaData.xml file.
I don't know if that will work in usual but for me it works.

@dagar  Is there an ther way to do it?
Maybe something where we can favoritize our parameters so we have it easier when flying to change them?