PX4/PX4-Autopilot #20610: CubePilot CubeOrange Sensor Start Sequence Issue

wsolichin-sees, 2022-11-15:
## Describe the bug
I'm currently using PX4 Firmware v1.13.1 with CubePilot CubeOrange. I have set the SENS_EN_LL40LS to I2C and SENS_EN_IRLOCK to Enabled. Both are connected to an I2C splitter to the I2C pin out on CubePilot CubeOrange. However, I cannot see any MAVLink message of DISTANCE_SENSOR or uORB message irlock_report. Further debugging, both the drivers `ll40ls` and `irlock` did not start. On PX4 v1.12.1 Firmware, both drivers worked perfectly. I tried to start `ll40ls` (Distance sensors driver) first, followed by `irlock` (IR-Lock drivers), but the `ll40ls` drivers did not load. To load the `ll40ls`, I have to start `irlock` drivers, followed by `ll40ls`.

## To Reproduce
1. On PX4 v1.13.1 with CubePilot CubeOrange, set SENS_EN_LL40LS to I2C and SENS_EN_IRLOCK to Enabled.
2. Check if distance sensors / IR-Lock drivers has been loaded, typing either:
```
ll40ls status
```
or
```
irlock status
```
3. They are both not started, and starting `ll40ls start -X` will return `no instance started`.

## Expected behavior
When setting SENS_EN_LL40LS to I2C and SENS_EN_IRLOCK to Enabled, I will be able to see both messages by default, and the drivers start on boot. For reference, PX4 firmware v1.12.1 loads both drivers without issue (Following the old instructions of starting the `irlock` and `landing_target_estimator` module via SD Card: https://docs.px4.io/v1.12/en/advanced_features/precland.html#firmware-configuration, and setting SENS_EN_LL40LS to I2C).

## Log Files and Screenshots
No flight log is needed to troubleshoot the following setup, here is a screenshot of the modules debugging via MAVLink Console:

[image]

## Drone (please complete the following information):
- Airframe type: Octorotor Coaxial
- PX4 Autopilot version: v1.13.1.

## Additional context
None

ealdaz, 2022-11-17:
I have got the exact same issue in my CubePilot Orange, have you had any luck ?

wsolichin-sees, 2022-11-21:
Update with the I2C sensor startup issue. For the distance sensors, I have added a `_retries = 1;` in line 126 of the `ll40ls.cpp` file here: https://github.com/PX4/PX4-Autopilot/blob/release/1.13/src/drivers/distance_sensor/ll40ls/LidarLiteI2C.cpp#L126, and that seems to fix it.

For the IR-Lock, I noticed that providing external power to the IR-Lock, rather than using the Pixhawk flight controller as a power source seems to fix it. I'm guessing the boot sequence has changed between v1.12.1 and v1.13.1. Suspect that in v1.13.1 I2C communications begin prior to power been actually supplied to the sensor. Here is the screenshot of the IR-Lock and LL40LS sensors starting on boot.

[image]

DavidPaddy97, 2023-05-22:
Would be awesome if someone took a look at this, quite an annoying bug at the moment for us too!