PX4/PX4-Autopilot #20615: PX4 SITL unable to takeoff typhoon_h480 - primary EKF changed 0 (filter fault)

guiaugustoga987, 2022-11-15:
### Describe the bug

I'm trying to use the typhoon_h480 with SITL in gazebo. At first, I created a world with a 3DR Iris model and run `no_sim=1 make px4_sitl_default gazebo` and then `roslaunch gazebo_ros empty_world.launch world_name:=$(pwd)/Tools/sitl_gazebo/worlds/typhoon.world`. Everything works fine, I can control the UAV correctly. Now I made the same steps, but I changed the drone to typhoon_h480. When I launch the world I receive several warnings regarding 'primary EKF changed' and ' baro hgt timeout - reset to baro'.

### To Reproduce

Steps to reproduce the behavior:

1.     Create a simple world in gazebo with typhoon_h480.
2.     Run `no_sim=1 make px4_sitl_default gazebo` and `roslaunch gazebo_ros empty_world.launch world_name:=$(pwd)/Tools/sitl_gazebo/worlds/typhoon.world`

### Expected behavior

Should be able to takeoff and control the UAV.

### Log Files and Screenshots

```
sim=1 make px4_sitl_default gazebo
[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[3/4] cd /home/guilherme/src/Firmware/...me/src/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/guilherme/src/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: none
world: none
src_path: /home/guilherme/src/Firmware
build_path: /home/guilherme/src/Firmware/build/px4_sitl_default
empty model, setting iris as default
SITL COMMAND: "/home/guilherme/src/Firmware/build/px4_sitl_default/bin/px4" "/home/guilherme/src/Firmware/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/home/guilherme/src/Firmware"/test_data
INFO  [px4] Creating symlink /home/guilherme/src/Firmware/build/px4_sitl_default/etc -> /home/guilherme/src/Firmware/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
Info: found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file eeprom/parameters_10016
INFO  [parameters] BSON document size 486 bytes, decoded 486 bytes
[param] Loaded: eeprom/parameters_10016
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
PX4 SIM HOST: localhost
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
INFO  [simulator] Simulator connected on TCP port 4560.
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
INFO  [ekf2] starting instance 0, IMU:0 (1310988), MAG:0 (197388)
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [ekf2] starting instance 1, IMU:1 (1310996), MAG:0 (197388)
INFO  [ekf2] starting instance 2, IMU:2 (1311004), MAG:0 (197388)
INFO  [ekf2] starting instance 3, IMU:0 (1310988), MAG:1 (197644)
INFO  [ekf2] starting instance 4, IMU:1 (1310996), MAG:1 (197644)
INFO  [ekf2] starting instance 5, IMU:2 (1311004), MAG:1 (197644)
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2022-11-15/21_47_19.ulg
INFO  [logger] Opened full log file: ./log/2022-11-15/21_47_19.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
[... 155 lines cut ...]
WARN  [ecl/EKF] baro hgt timeout - reset to baro
WARN  [ecl/EKF] baro hgt timeout - reset to baro
WARN  [ecl/EKF] baro hgt timeout - reset to baro
WARN  [ecl/EKF] baro hgt timeout - reset to baro
WARN  [ecl/EKF] baro hgt timeout - reset to baro
WARN  [ecl/EKF] baro hgt timeout - reset to baro
WARN  [ecl/EKF] baro hgt timeout - reset to baro
WARN  [ecl/EKF] baro hgt timeout - reset to baro
WARN  [ecl/EKF] baro hgt timeout - reset to baro
WARN  [ecl/EKF] baro hgt timeout - reset to baro
WARN  [ecl/EKF] baro hgt timeout - reset to baro
WARN  [ecl/EKF] baro hgt timeout - reset to baro
WARN  [ekf2] primary EKF changed 1 (filter fault) -> 4
WARN  [ekf2] primary EKF changed 4 (filter fault) -> 0
WARN  [ekf2] primary EKF changed 0 (filter fault) -> 1
WARN  [ecl/EKF] baro hgt timeout - reset to baro
WARN  [ecl/EKF] baro hgt timeout - reset to baro
WARN  [ecl/EKF] baro hgt timeout - reset to baro
WARN  [ecl/EKF] baro hgt timeout - reset to baro
WARN  [ecl/EKF] baro hgt timeout - reset to baro
WARN  [ecl/EKF] WARN  [ecl/EKF] baro hgt timeout - reset to baro
baro hgt timeout - reset to baro
```

I'm not sure what is causing this behavior. Any help is appreciated.

guiaugustoga987, 2022-11-16:
I'm not sure if this has something to do with the error I'm facing, but this is what is happening when I add the Typhoon H480 to gazebo : 

[image]

Jaeyoung-Lim, 2022-11-16:
@guiaugustoga987 Have you tried

```
make px4_sitl gazebo_typhoon_h480
```

guiaugustoga987, 2022-11-17:
I just solved it. My .world file was getting the models from .gazebo folder. I changed to the sitl_gazebo/models folder and now It is working as intended.

Thank for your inside.

ANTHONYBRIDGELI, 2023-03-21:
> ### Describe the bug
>
> I'm trying to use the typhoon_h480 with SITL in gazebo. At first, I created a world with a 3DR Iris model and run `no_sim=1 make px4_sitl_default gazebo` and then `roslaunch [...]

There is the same problem within my project,but I seem to find a way to solve it probably.Just try to load the initial position higher than the normal,when I set the initial z axis 0.2,the error disappeared.

closed 2025-09-26 by Jaeyoung-Lim