PX4/PX4-Autopilot #20631: Add parameter to enable mavlink stream over USB

ThomasRigi, 2022-11-17:
From discord thread https://discord.com/channels/1022170275984457759/1032553608417394709

## Describe problem solved by the proposed feature
PX4 no longer sends mavlink via USB unless a heartbeat is received via this channel. This causes trouble to some existing setups, e.g. MAVROS paired with mavlink-router.

## Describe your preferred solution
Have a parameter that lets you choose to always stream mavlink over USB, as was the case up to v1.12.

## Describe possible alternatives
Hardcode it to always send, but that's less clean of a solution

A minimal version of the hardcoded solution can be obtained by moving the four variables https://github.com/PX4/PX4-Autopilot/blob/8b7c0746800b3597d9418f6442696a1f6fbf3e71/platforms/nuttx/src/px4/common/cdc_acm_check.cpp#L205-L208 out of two if-statements to line https://github.com/PX4/PX4-Autopilot/blob/8b7c0746800b3597d9418f6442696a1f6fbf3e71/platforms/nuttx/src/px4/common/cdc_acm_check.cpp#L138 and setting `bool launch_mavlink = true;` (replacing `bool launch_mavlink = true;` with `bool launch_mavlink = MAVLINK_USB_EN;` or whatever you want to call the parameter for it to not be hard-coded)

Katawann, 2023-01-30:
It looks like this behavior of v1.13 waiting for hearbeat is also giving some issue on MAVSDK [side](https://github.com/mavlink/MAVSDK/issues/1967) (fixed by now)

I also saw that there were some [proposition](https://github.com/mavlink-router/mavlink-router/pull/171) to add heartbeat in mavlink router but it has never been merged

ThomasRigi, 2023-01-30:
Yes, your changes are correct. I wouldn't know what doesn't work then. I didn't have to change any parameters on our configuration.

I only tested on a Pixhawk 4, maybe different hardware reacts differently ?

dagar, 2023-01-30:
If there really is nothing else we can use then I'll add a parameter.

julianoes, 2023-01-30:
Adding a param would certainly be helpful for some configurations. That being said, it's now fixed for MAVSDK.

Katawann, 2023-01-30:
Proposition of @ThomasRigi works for me as well at the end !

Katawann, 2023-02-13:
A temporary fix proposed by @dagar and that works for me is to add in the SD card a file `/etc/extras.txt` with the following command:

```
sercon
mavlink start -d /dev/ttyACM0
```

Katawann, 2023-06-23:
Hi @dagar, @julianoes I saw a [new fix](https://github.com/PX4/PX4-Autopilot/pull/20745) for this issue on v1.14, but I just did a test and I still have the same problem. By default I don't see USB with mavlink status. I still need to add the hotfix script from previous comment to enable USB and be able to stream log with mavlink-router... Below is the output when the script is there, without it I don't see the instance USB CDC

[image]

Let me know if I missed an extra step to make it works !

julianoes, 2023-06-24:
To me it sounds like mavlink-router should also send a heartbeat if it wants to talk mavlink and do log streaming.

Katawann, 2023-06-26:
There was a proposition at some point [here](https://github.com/mavlink-router/mavlink-router/pull/171) but I read somewhere that mavlink-router purpose is to be a router and therefore sending heartbeat is out of its scope.

I'm also wondering why it was working on v1.10.1 of PX4 and not anymore on v1.13 or higher

I'll open an issue on mavlink-router side and see what they say

julianoes, 2023-06-26:
If mavlink-router does log streaming then it's not just a router but needs to work by itself, and should have an id in that case.

The change was because PX4 now auto-detects mavlink vs NuttX nsh. Not sure who asked for that feature but it's there.

dakejahl, 2023-10-16:
> Not sure who asked for that feature but it's there.

@julianoes Do you agree it would be more useful for it to just be mavlink? Since you can always open a shell via mavlink. A param to override this and force starting mavlink would still be useful

dakejahl, 2023-10-17:
> A temporary fix proposed by @dagar and that works for me is to add in the SD card a file `/etc/extras.txt` with the following command:
> 
> ```
> sercon
> mavlink start -d /dev/ttyACM0
> ```

This doesn't appear to work anymore

Katawann, 2023-10-17:
> This doesn't appear to work anymore

if it doesn't work add a sleep between sercon and trying to start mavlink on /dev/ttyACM0
```
sercon
sleep 5s
mavlink start -d /dev/ttyACM0
```

ThomasRigi, 2023-10-17:
[cdc_acm_check.zip](https://github.com/PX4/PX4-Autopilot/files/12926719/cdc_acm_check.zip)

@dakejahl you can try with this version of cdc_acm_check.cpp if you want.

dakejahl, 2023-10-17:
I made a PR which adds a parameter for this
https://github.com/PX4/PX4-Autopilot/pull/22234

PR #22234 (2024-05-20): [mavlink] Parameter to always start on USB
Added parameter and logic to always start mavlink on USB instead of auto-detecting the protocol. The mavlink mode can now be selected as well. I also refactored the code into a module and moved it into the LP px4 work queue. I refactored everything into functions and a state machine.

Added parameters
SYS_USB_AUTO
USB_MAV_MODE

**The problem being solved**
When running mavlink-router on a companion computer, no data will be routed until a GCS connects. This is because in typical mavlink-router configuration there is only one `Server` endpoint which co [... cut]
closed 2024-05-20 by AlexKlimaj