PX4/PX4-Autopilot #20632: Build Error when running make px4_sitl gazebo

cmm6758, 2022-11-18:
Hello,

Whenever I attempt to build the simulator for PX4 I receive this error below. I am new to this so I am not sure what I am doing wrong. What does this error mean and what should I do to resolve it?

$make px4_sitl gazebo

[  1%] Built target events_header
[  1%] Built target ver_gen
[  1%] Built target parameters_xml
[  1%] Built target uorb_headers
[  1%] Built target actuators_json
[  1%] Built target flighttasks_generated
[  1%] Built target output_functions_header
[  2%] Built target drivers_board
[  3%] Built target git_gps_devices
[  3%] Built target topic_bridge_files
[  3%] Built target git_mavlink_v2
[  3%] Built target uorb_ucdr_headers
[  3%] Built target git_micro_xrce_dds_client
[  3%] Built target git_gazebo
[  3%] Built target events_json
[  3%] Built target perf
[  3%] Built target tinybson
[  5%] Built target work_queue
[  5%] Built target version
[  5%] Built target geo
[  5%] Built target wind_estimator
[  6%] Built target world_magnetic_model
[  6%] Built target mode_util
[  6%] Generating Mavlink development: src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
[  8%] Performing configure step for 'sitl_gazebo'
[  9%] Built target libmicroxrceddsclient_project
[ 27%] Built target uorb_msgs
-- install-prefix: /usr/local
-- cmake build type: RelWithDebInfo
-- Using C++17 standard
[ 27%] Built target component_general_json
[ 27%] Built target parameters
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.58") found components: system thread filesystem
[ 28%] Built target romfs_gen_files_target
Scanning dependencies of target px4_work_queue
[ 28%] Building CXX object platforms/common/px4_work_queue/CMakeFiles/px4_work_queue.dir/ScheduledWorkItem.cpp.o
[ 28%] Building CXX object platforms/common/px4_work_queue/CMakeFiles/px4_work_queue.dir/WorkItem.cpp.o
Traceback (most recent call last):
  File "/home/mellarijocm/Firmware/src/modules/mavlink/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "/home/mellarijocm/Firmware/src/modules/mavlink/mavlink/pymavlink/generator/mavgen.py", line 26, in <module>
    from future import standard_library
ModuleNotFoundError: No module named 'future'
make[4]: *** [src/modules/mavlink/CMakeFiles/mavlink_c_generate.dir/build.make:62: mavlink/development/development.h] Error 1
make[3]: *** [CMakeFiles/Makefile2:10299: src/modules/mavlink/CMakeFiles/mavlink_c_generate.dir/all] Error 2
make[3]: *** Waiting for unfinished jobs....
[ 28%] Building CXX object platforms/common/px4_work_queue/CMakeFiles/px4_work_queue.dir/WorkItemSingleShot.cpp.o
[ 28%] Building CXX object platforms/common/px4_work_queue/CMakeFiles/px4_work_queue.dir/WorkQueue.cpp.o
[ 28%] Building CXX object platforms/common/px4_work_queue/CMakeFiles/px4_work_queue.dir/WorkQueueManager.cpp.o
[ 28%] Linking CXX static library libpx4_work_queue.a
[ 28%] Built target px4_work_queue
-- Found DART: /usr/include (Required is at least version "6.6") found components: dart
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.40.0") found components: thread system filesystem program_options regex iostreams date_time
-- Looking for ignition-math6 -- found version 6.13.0
-- Searching for dependencies of ignition-math6
-- Looking for OGRE...
-- Found Ogre Ghadamon (1.9.0)
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
[... 167 lines cut ...]
   used as include directory in directory /home/mellarijocm/Firmware/Tools/simulation/gazebo/sitl_gazebo
   used as include directory in directory /home/mellarijocm/Firmware/Tools/simulation/gazebo/sitl_gazebo
   used as include directory in directory /home/mellarijocm/Firmware/Tools/simulation/gazebo/sitl_gazebo
   used as include directory in directory /home/mellarijocm/Firmware/Tools/simulation/gazebo/sitl_gazebo
   used as include directory in directory /home/mellarijocm/Firmware/Tools/simulation/gazebo/sitl_gazebo
   used as include directory in directory /home/mellarijocm/Firmware/Tools/simulation/gazebo/sitl_gazebo
   used as include directory in directory /home/mellarijocm/Firmware/Tools/simulation/gazebo/sitl_gazebo
   used as include directory in directory /home/mellarijocm/Firmware/Tools/simulation/gazebo/sitl_gazebo
   used as include directory in directory /home/mellarijocm/Firmware/Tools/simulation/gazebo/sitl_gazebo/unit_tests
   used as include directory in directory /home/mellarijocm/Firmware/Tools/simulation/gazebo/sitl_gazebo/unit_tests
   used as include directory in directory /home/mellarijocm/Firmware/Tools/simulation/gazebo/sitl_gazebo/unit_tests
   used as include directory in directory /home/mellarijocm/Firmware/Tools/simulation/gazebo/sitl_gazebo/unit_tests
   used as include directory in directory /home/mellarijocm/Firmware/Tools/simulation/gazebo/sitl_gazebo/unit_tests
   used as include directory in directory /home/mellarijocm/Firmware/Tools/simulation/gazebo/sitl_gazebo/unit_tests
   used as include directory in directory /home/mellarijocm/Firmware/Tools/simulation/gazebo/sitl_gazebo/unit_tests
   used as include directory in directory /home/mellarijocm/Firmware/Tools/simulation/gazebo/sitl_gazebo/unit_tests

-- Configuring incomplete, errors occurred!
See also "/home/mellarijocm/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/home/mellarijocm/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
make[4]: *** [src/modules/simulation/simulator_mavlink/CMakeFiles/sitl_gazebo.dir/build.make:107: external/Stamp/sitl_gazebo/sitl_gazebo-configure] Error 1
make[3]: *** [CMakeFiles/Makefile2:34202: src/modules/simulation/simulator_mavlink/CMakeFiles/sitl_gazebo.dir/all] Error 2
make[2]: *** [CMakeFiles/Makefile2:37289: src/modules/simulation/simulator_mavlink/CMakeFiles/gazebo_rover.dir/rule] Error 2
make[1]: *** [Makefile:14331: gazebo_rover] Error 2
make: *** [Makefile:232: px4_sitl] Error 2

tayyabkhalil-313, 2022-11-21:
> from future import standard_library
ModuleNotFoundError: No module named 'future'

Seems like python library 'future' is missing. Install it by running '`pip install future`' (or  '`pip3 install future`' if you have pip3).

junwoo091400, 2023-03-25:
I am also troubleshooting this, I think it's related to the MAVLink dependency not being found:

```
-- Could NOT find MAVLink (missing: MAVLINK_INCLUDE_DIRS) (found version "2.0")
```

junwoo091400, 2023-03-25:
Seems to be related: https://github.com/gnuradio/gnuradio/issues/3581

junwoo091400, 2023-03-25:
Found the solution:

I had to first build the `px4_sitl_default`, and then after that built the `px4_sitl_default sitl_gazebo-classic`. In other words:

```bash
make px4_sitl_default
make px4_sitl_default sitl_gazebo-classic
```

Note that I am doing this on current `main` branch (v1.14.0-beta), so the `-classic` part should be omitted from your side (if testing with v1.13.0 or lower versions)!

closed 2023-03-25 by junwoo091400