PX4/PX4-Autopilot #20633: Issue with the Pixhawk 6c's IMU

EhrazImam, 2022-11-18:
Issue details

Hi i am using Pixhawk 6C with px4(1.13.1)
When I’m changing the Pixhawk parameters in QGC (SYS_MC_EST_GROUP = No position, SYS_HAS_BARO & SYS_HAS_MAG = 0), it only displays /mavros/imu/data_raw not /mavros/imu/data_raw. With same parameter I have also tried on Pixhawk 4, Cube Orange and Cube Black with different firmware(1.11.0, 1.11.1, 1.12.0, 1.13.0, 1.13.1) and they all displaying /mavros/imu/data and /mavros/imu/data_raw as well.

MAVROS version and platform

Mavros: 1.14.0
ROS: Noetic
Ubuntu: 20.04
Autopilot type and version

[ ] ArduPilot
[ x ] PX4

Version: 1.13.1

EhrazImam, 2022-12-20:
@afwilkin Thankyou typing error

EhrazImam, 2023-02-23:
@bresch @julianoes @LorenzMeier  Please help me with this

afwilkin, 2023-02-23:
@EhrazImam the best way to get help is not through spamming the creators of PX4. 

The first question is why do you need that specific topic? What are you trying to accomplish

EhrazImam, 2023-02-23:
@afwilkin Sorry will keep it in mind.
1. In my onboard control i am using  **nh.subscribe<sensor_msgs::Imu>("/mavros/imu/data",** why mavros/imu/data because i am using the orientation of IMU **u.q = imu.q * odom.q.inverse() * q;** when i am changing parameters i am no longer to give IMU's Orinetation because it is just publishing mavros/imu/data_raw.
2. Why i am changing the parameter because i need to test it in high magnetic environment where i need to turn it off and while changing this parameter and PID it was working smooth as compared to SYS_MC_EST_GROUP=ekf2 (recommended),  SYS_HAS_BARO & SYS_HAS_MAG = 1). 
3. And for localization i am using VIO.

afwilkin, 2023-02-23:
I am still unsure on exactly what you are trying to accomplish, but it appears you are trying to get magnetometer data in highly disruptive environments is that correct?

Also, if you are flying over vio why do you need your magnetometer?

afwilkin, 2023-02-23:
I guess my question is, if you are flying over VIO, why do you need the magnetometer?

EhrazImam, 2023-03-16:
The VIO pose is computed relative to the world frame where the x-axis points to the forward direction when we start VIO.
However, the PX4 FCU update orientation is based on its own world frame.
These two world frames are not aligned, especially in the yaw direction.
So we have to subscribe the mavros/imu/data to acquire the yaw difference between the world frames of VIO and the FCU, so that we can convert the quaternion command computed in the VIO world frame to the FCU world frame.

bzw-uestc, 2023-12-06:
I encountered the same problem, has this problem been solved?