PX4/PX4-Autopilot #20634: microdds_client not working with real hardware

Ecuashungo, 2022-11-18:
## Describe the bug
I use a raspberry pi 4 companion computer with ros2 humble and micro_ros_agent installed. When I try to connect it to a pixhawk 4 mini setup for HITL through serial (TELEM 1 port on pixhawk, /dev/ttyAMA1 on the raspberry pi) they start the communication, but very quickly abort.
On the companion computer the following message is shown:
```
pi@cm17:~/ros_ws $ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyAMA1 -b 921600 -v
[1668776671.744447] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1668776671.745096] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1668776672.420570] info     | Root.cpp           | create_client            | create                 | client_key: 0xAAAABBBB, session_id: 0x81
[1668776672.420935] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0xAAAABBBB, address: 1
[ros2run]: Segmentation fault

```
On the px4-side I get the following error after executing `microdds_client start -t serial -d /dev/ttyS1 -b 921600`:

```
nsh> microdds_client start -t serial -d /dev/ttyS1 -b 921600
nsh> ERROR [microdds_client] create entities failed: participant: 255
```

## To Reproduce
0. Prepare the flight controller for HITL mode as explained [here](https://docs.px4.io/main/en/simulation/hitl.html) and install micro-ros-agent on the companion computer.
1. Setup the cabling between the companion computer and the pixhawk 4 mini
2. Start HITL environment (pixhawk 4 mini connected to the development computer through USB) as explained [here](https://docs.px4.io/main/en/simulation/hitl.html#gazebo).
3. On the companion computer run `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyAMA1 -b 921600 -v`
4. In QGC open the mavlink shell of the pixhawk 4 mini
5. in the mavlink shell run `microdds_client start -t serial -d /dev/ttyS1 -b 921600`

## Expected behavior
When I run this on my development computer in a SITL environment the agent and the client communicate through udp without a problem. In the SITL example I used a [modified c++ node (see my pull request)](https://github.com/Jaeyoung-Lim/px4-offboard/pull/7) inspired by [px4-offboard](https://github.com/Jaeyoung-Lim/px4-offboard).

I expect the agent and the client to communicate as they do in SITL.

## Log Files and Screenshots
None

## Drone (please complete the following information):
I use a Pixhawk 4 mini in HITL mode

## Additional context
I am sure that the serial connection between the companion computer works, because I have tried to access the mavlink shell with the [mavlink_shell.py script](https://github.com/PX4/PX4-Autopilot/blob/main/Tools/mavlink_shell.py) from the companion computer. After that I have disabled Mavlink on TELEM 1 (MAV_0_CONFIG = 0), set the serial baudrate to 921600 (SER_TEL1_BAUD = 921600) and enabled (XRCE_DDS_0_CFG = TELEM1).

dagar, 2022-11-18:
I'm not sure if there's any useful information, but you could try launching the agent with more verbosity to see if it has anything helpful (-v4 I think). 

Have you tried pixhawk -> PC through serial for comparison? I'll give it a try myself later, but I did verify it during development earlier this year.

Ecuashungo, 2022-11-21:
I tried to launch it with `-v 6` which is the maximum verbosity and I still get the same error. There are some extra messages, that I don't understand right now:

```
ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyAMA1 -b 921600 -v 6

[1669023041.369509] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1669023041.370166] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1669023112.161773] info     | Root.cpp           | create_client            | create                 | client_key: 0xAAAABBBB, session_id: 0x81
[1669023112.161988] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0xAAAABBBB, address: 1
[1669023112.162331] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAAAABBBB, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1669023112.165235] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAAAABBBB, len: 108, data: 
0000: 81 80 00 00 01 05 62 00 00 0A 00 11 01 02 00 00 53 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 70 78 34 5F 6D 69 63 72 6F 5F 78 72 63 65
0040: 5F 64 64 73 3C 2F 6E 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63 69 70 61 6E 74 3E
0060: 3C 2F 64 64 73 3E 00 00 00 00 00 00
[ros2run]: Segmentation fault
```
Does the data mean anything to you?

In the meantime I have tried the same setup with a pixhawk 4 on the TELEM 2 port. Is it correct to set MAV_1_CONFIG to Disabled and the XRCE_DDS_0_CFG to TELEM2? 

I will try to do the same with my development computer and post updates here.

Ecuashungo, 2022-11-21:
@dagar I have tried to do the same setup with my development computer running ubuntu 20.04 and ROS2 Galactic and I could make it work just fine. Therefore, the error must be somewhere on my companion computer setup.

I have tried to reinstall the `micro_ros_agent` and during the re-installation there was a warning:

```
/usr/bin/ld: warning: libfastrtps.so.2.6, needed by /home/pi/ros2_humble/install/rmw_fastrtps_shared_cpp/lib/librmw_fastrtps_shared_cpp.so, 
may conflict with libfastrtps.so.2.8.1
---
Finished <<< micro_ros_agent [5min 50s]
```

Could this be part of my problem?

Ecuashungo, 2022-11-28:
After raising the [issue on the agent side](https://github.com/micro-ROS/micro_ros_setup/issues/597), I have managed to make the microdds_client to agent bridge communicate just fine using the Micro XRCE-DDS Agent instead of the micro-ros-agent.

I was able to run an offboard example using HITL and commanding trajectory setpoints from the raspberry pi companion computer. One problem that remains is that I get a warning in the nuttShell saying:
`WARN [timesync] RTT too high for timesync: 106 ms`

Does this have to do anything with the microdds_client?

dirksavage88, 2022-11-30:
@Ecuashungo The microxrcedds agent you installed uses a superbuild cmake to fetch dependencies, it probably would have resolved the libfastrtps dynamic library issue. It looks like you had a newer version of libfastrtps installed.

zcelil, 2022-11-30:
I am also in HITL mode but I can't find microdds_client in mavlink shell. I'm planning to do the same ros2 offboard example via serial configuration. So, I need to run microdds_client serial on the host computer (px4 side) but there is no microdds_client command and XRCE_DDS_0_CFG parameter at all. My board is fmu_v5 and im in HITL mode. Do you have any ideas about what might be the problem?

ViniciusAbrao, 2023-02-13:
@Ecuashungo please, how do you start the microdds_client ? is it via mavlink? I am trying to do it but if I work with mavlink I can not simultaneously work with the xrce-dds birdge !! How do you manage it?

ozitoune, 2023-02-15:
Hello , i 'm on the same situation. Impossible to see any topics. I'm on ubuntu 22.4 with humble.
I have the xrce client started on pixhawk (i see it on top) and i can run Xrce agent or microros agent on my raspberry pi but i can't see any topics.
Is there any solution to make this work ? 
Why they replace the micro rtps in the main branch if the new micro dds is not working with harware ???
I can downgrade  px4 firmware because i use a pixhawk 6X only compatible with the 1.13.1 firmware.
pixhawk is connected to my raspberry by usb, can this work or should i connect by telemetrie 2?
Thank for any help

ViniciusAbrao, 2023-02-15:
@ozitoune I managed to fix it as following (reference -  https://github.com/PX4/PX4-Autopilot/issues/21120#issuecomment-1428347091):

if you want to use TELEM1, you have to stop mavlink and then activate the microdds_client. You can do that changing the parameters:
MAV_0_CONFIG = 0
to disable mavlink on TELEM1,
XRCE_DDS_0_CFG = TELEM1
to start microdds_client on TELEM1 and SER_TEL1_BAUD to set the baudrate.

to use TELEM2 (ttyS3 of pixhawk):
XRCE_DDS_0_CFG = TELEM2
SER_TEL2_BAUD = 921600
To check if it is already running, in the QGroundControl's MavlinkConsole: microdds_client status
if not running yet, start the client in the MavlinkConsole: microdds_client start -t serial -d /dev/ttyS3 -b 921600
In raspberrypi: sudo MicroXRCEAgent serial --dev /dev/serial0 -b 921600

ozitoune, 2023-02-15:
Thank you,  you're an angel !! 
There is one mouth i try to make this work ... But impossible via usb , i have read, long time ago , usb is allow to development only, so during the developpement phase it would be possible to use ACM0, but apperently no ! 

I'm a little confused , AMA0 is when we use a fdti/usb  adaptator? so if connect telem2 on a fdti adaptator then on a raspberry usb port it could be work ?  in a raspbery is /dev/serial0  or /dev/ttyAMA0 ???

The best is to connect directly telem2 to raspbery tx and rx ?

I have a pixhawk 6x so i would like to connect via ethernet but , i dont see where to tell the ip adress on Xrce client / agent, there is -p for the port but how to specify the host ? 
Thanks a lot for your work and your help !  For now i can't see the topics, but i think tomorow i will connect telem2 to raspberry and it would better !!

Another question Mavros is still usable ? There is a MAVROS / ROS2 page so perhaps  it could be usable if i can't make it work with xrce.

Thanks again !!

ViniciusAbrao, 2023-02-15:
> Thank you, you're an angel !! There is one mouth i try to make this work ... But impossible via usb , i have read, long time ago , usb is allow to development only, so during the developpement phase [...]

@ozitoune you do not need a fdti/usb adapter in order to connect the pixhawk with raspberrypi. You can connect it in the raspberrypi IO pins directly. When you do it, you are working with the AMA0 in the raspberrypi (same as serial0).

I think that mavlink is still working correctly, since you need it for communication in QGroundControl, so should not have problems with mavros.

ozitoune, 2023-02-15:
ok so if I understand I can run `sudo MicroXRCEAgent serial --dev /dev/serial0 -b 921600`  or ` sudo MicroXRCEAgent serial --dev /dev/AMA0 -b 921600` this 2 commands do the same think ? 
what did you mean by " you also need to configure the raspberry's serial port /dev/ttyAMA0."
And do you have an idea if it's possible to connect directly my pixhawk 6x via ethernet on the raspbery and run the client / agent in udp4 with the ip adress of the raspberry /piwhaxk ?
Thank you

ViniciusAbrao, 2023-02-15:
> ok so if I understand I can run `sudo MicroXRCEAgent serial --dev /dev/serial0 -b 921600` or ` sudo MicroXRCEAgent serial --dev /dev/AMA0 -b 921600` this 2 commands do the same think ? 

No, the correct is /dev/ttyAMA0.

what did you mean by " you also need to configure the raspberry's serial port /dev/ttyAMA0." 

There are some tutorials better explaining the steps, like here:

https://youtu.be/kB9YyG2V-nA

Double check if your pixhawk model have the same TELEM pins sequence as in the video.

And do you have an idea if it's possible to connect directly my pixhawk 6x via ethernet on the raspbery and run the client / agent in udp4 with the ip adress of the raspberry /piwhaxk ? Thank you

The communication is via pixhawk's TELEM, so the connection must be via tx/rx/ground pins .

ozitoune, 2023-02-15:
Thank you for the link !! 
Now i can see the topic list by running : `sudo MicroXRCEAgent udp4 --port 8888 ROS_DOMAIN_ID=0` on the raspbery
and microdds_client start -t udp -p 8888 -h 192.168.0.100 (the rpi adress)
the pixhawk and the raspbery arre connected on the same router via ethernet
now `ros2 topic list` give me all the topics but they are empty or not published by the pixhawk ??
`
ros2 topic echo /fmu/out/vehicule_attitude`
"WARNING: topic [/fmu/out/vehicule_attitude] does not appear to be published yet
Could not determine the type for the passed topic"
Do you know where i'm wrong ?

It's not working but i'm happy , it's the first time i see the list of topics !!

ViniciusAbrao, 2023-02-15:
> 

Great, my pixhawk does not have ethernet, so I only tested with TELEM.
Probably you are not with the drone completely configured, so without GPS maybe the topics will no be publishing any message.

ozitoune, 2023-02-16:
All topics works now !! only topics with the need off GPS don't publish data because i'm indoor.

Do you know how to make the start off the xrce client automatic at px4 startup?

ViniciusAbrao, 2023-02-16:
> All topics works now !! only topics with the need off GPS don't publish data because i'm indoor.
 
Did you just plug the GPS and all the topics start to work or did you do anything else?

> Do you know how to make the start off the xrce client automatic at px4 startup?

Looks like is not necessary to start client, as you can see here:

https://github.com/PX4/PX4-Autopilot/issues/21120#issuecomment-1430193467

beniaminopozzan, 2023-02-17:
@ozitoune congratulations for having microdds on UDP working on the Pixhawk!

Let me start by answering to this

> Do you know how to make the start off the xrce client automatic at px4 startup?

As you need to specify the agent address, you cannot do that from the PX4 parameters. However you can still start the client automatically using a custom startup file on the SD: https://docs.px4.io/main/en/concept/system_startup.html#starting-additional-applications-extras-txt just put `microdds_client start -t udp -p 8888 -h 192.168.0.100` there and it's done.

As we are fixing up the new documentation for the microdds bridge I have a fer questions for you:

1. Did you follow these instructions https://docs.px4.io/main/en/advanced_config/ethernet_setup.html for configuring the Ethernet communication on the pixhawk before starting the microdds client?
3. What did it change between 

   > now ros2 topic list give me all the topics but they are empty or not published by the pixhawk ??
   ros2 topic echo /fmu/out/vehicule_attitude
   "WARNING: topic [/fmu/out/vehicule_attitude] does not appear to be published yet
   Could not determine the type for the passed topic"

   and

   > All topics works now !! only topics with the need off GPS don't publish data because i'm indoor.

   Are you saying that all topics that don't require a position estimate were correctly been sent from the beginning?

1. Could you describe step by step the procedure you followed to configure client+agent? That would be a great contribution for the new documentation for microdds via UDP on real drones!

---

Just a few details for the serial connection:

> There is one mouth i try to make this work ... But impossible via usb , i have read, long time ago , usb is allow to development only, so during the developpement phase it would be possible to use ACM0, but apperently no !

You can have mavlink on `/dev/ttyACM0` (PX4 side I mean), after all QGC connects to it when using the USB cable, but I don't know about microdds...

> Another question Mavros is still usable ? There is a MAVROS / ROS2 page so perhaps it could be usable if i can't make it work with xrce.

Yes, of course.

Rearding using the serial protocol, you can perfectly use a fdti/usb adaptator, but the device name will be different, normally it is something like `/dev/ttyUSB*`. Configuring the hardware serial on the Raspberry depends on the Raspberry version: version 3 and 4 differs in terms of number of serial ports. And also the OS on the Raspberry has its importance! The commands to enable the serial port are different If you are using raspberry OS or ubuntu server, for example. But there are tons of tutorial online specific for a particular board and OS.

>  so if connect telem2 on a fdti adaptator then on a raspberry usb port it could be work

It would definitively works but you may need to set the right permission to access the device on your raspberry.

ozitoune, 2023-02-17:
@ViniciusAbrao  I have the gps connected and all topics are published out of the box, only the topic /fmu/out/failsafe_flags do nothing, no error , no message, i don't know why.

@beniaminopozzan the client start automaticly now but when i start the agent, there is nothing, i have to stop the client then rtestart it to see the topics. 
1. No , my raspberry and my pixhawk are connected on the same router with dhcp, rpi is 192.168.0.100, pixhawk 192.168.0.101 and the router is 192.168.1.1 , the next step is to connect them without gateway but i don't know if i have to put the router addres inn the pixhawk parameter if i don't have router. If i put static address for the raspberry 192.168.0.1 and put 192.168.0.1 for the router parameter in pixhawk perhaps it will work, i will try this and i will tell you .

2. I don't know what's happen, i think the first time i make a mistake and forget to source a think. 

3.  -I have first installed all the px4 toolchain on ubuntu 22.04 (some difulty for some libs like fast-rtp-gen and arm-none-eabi-gcc ....) i have also problems with gazebo, i think not the correct version of jdk or other problem of dependencies. I have to investigate it more
    -After i install  ros 2 with the humble binary using the humble documentation method for ubuntu 22.04.
    -Next step i install px4_msg and px4_ros com using the documention , but impossible to compile with a desktop session on a raspberry pi 4 !! i have the 4G model but every time a freeze occurs to the raspberry due to full RAM . I disconnect the desktop seession and open a terminal session , the compilation was done corectly.
    - Then i have installed microros using `https://github.com/micro-ROS/micro_ros_setup`
    - i have also installed micro xrce dds agent  via snap because i was not sure it was correctly installed with microros.
    - to make it work i do :
```
source /opt/ros/humble/setup.bash 
source ~/px4_ros_com_ros2/install/setup.bash
source ~/microros_ws/install/setup.bash 
```
i start the agent via `ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888
`
or `sudo MicroXRCEAgent serial --dev /dev/serial0 -b 921600`

the i connect to the mavlink console via 
./Tools/mavlink_shell.py `/dev/ttyACM0
`
i stop the client and then restart it with 
microdds_client start -t udp -p 8888 -h `192.168.1.100`

And now it work !

beniaminopozzan, 2023-02-18:
@ozitoune thanks for your feedback!

> No , my raspberry and my pixhawk are connected on the same router with dhcp, rpi is 192.168.0.100, pixhawk 192.168.0.101 and the router is 192.168.1.1 , the next step is to connect them without [...]

This is really interesting! I did not know PX4 has DHCP (and it is actually well documented! https://docs.px4.io/main/en/advanced_config/ethernet_setup.html#setting-up-the-ethernet-network, my mystake!)
Anyway, good to know!

> some difulty for some libs like fast-rtp-gen and arm-none-eabi-gcc

If you are using your Raspberry just as companion computer, then you don't need PX4-Autopilot and all the build toolchain (namely, no arm-none-eabi-gcc is needed) and as you are using micro XRCE-DDS , even fast-rtp-gen is not needed (I mean this one https://docs.px4.io/main/en/dev_setup/fast-dds-installation.html#fast-rtps-gen is not needed)

> Next step i install px4_msg and px4_ros com using the documention , but impossible to compile with a desktop session on a raspberry pi 4 !! i have the 4G model but every time a freeze occurs to the [...]

you can limit the number of threads during compilation: see https://answers.ros.org/question/368249/colcon-build-number-of-threads/, the build command should be
```sh
MAKEFLAGS="-j1 " colcon build --executor-sequential
```
And eventually other `colcon` arguments.

> source /opt/ros/humble/setup.bash 
source ~/px4_ros_com_ros2/install/setup.bash
source ~/microros_ws/install/setup.bash

If you source the main ros packages and commands (`source /opt/ros/humble/setup.bash`) then the next workspaces that you want to source can be sourced using `install/local_setup` as `install/setup.bash` sources again the main ros packages.
To be more precise, `install/setup.bash`  sources all workspaces that where sourced when you called `colcon build` for the workspace you are considering.

> Edit : i forgot , as you said i have to set the right permission, i add my user to dialout group but the command sudo adduser 'user' 'dialout'  or sudo usermod -a -G dialout 'user' don't work for me and i have the same permission error, the solution who work for me is to create a file etc/udev/rules.d/50-myusb.rules with this lines :

You should check whats the group of `/dev/ttyACM0`. That is the group you want your user to be part of. And after adding it you need to logout and login.

>> As you need to specify the agent address, you cannot do that from the PX4 parameters.
Thanks for this , i understand why now, the adress of the agent !

>I think it's what i have made to make it work, i will try to make a fresh install of this to see if i forgot a think.

Please, let me know if the sequence:

1. Raspberry Pi switched off
2. Pixhawk switching on with client automatically starting using custom startup (no parameters) 
3. Switch on Raspberry then start agent

works. I just want to make sure you don't need the agent running before starting the client.

> But what is the difference off using fdti/usb and then plug into usb port of raspberry and directly plug the pixhawk on raspberry usb port ?

The Pixhawk USB port is _tricky_ I'm still trying to figure out how it is configured, but to give you some reasons why it is more complicated that a TELEM port

- the USB is used for firmware update
- the USB has a running mavlink instance. How do you switch it off?

> I would like to move servo connected to my pixhawk from ros2 , i mean i have to publish on /fmu/in/vehicle_command but what i have to publish ?

You mean you want to do in ROS2 what you can do with MAVLINK? https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_ACTUATOR
configuring your output as `Offboard Acutator Set X` https://docs.px4.io/main/en/payloads/#generic-actuator-control ?

If so, then the command message must be something like
```
timestamp: the timestamp
param1: 0.0
param2: 0.0
param3: 0.0
param4: 0.0
param5: 0.0
param6: 0.0
param7: 0.0
command: 187
target_system: 1
target_component: 1
source_system: 1
source_component: 1
confirmation: 0
from_external: true
```
set `paramX` , with `X` the number you put in `Offboard Acutator Set X` as the servo value between [-1, 1]. `command: 187` is `VEHICLE_CMD_DO_SET_ACTUATOR` so PX4 can understand the command. `target_X` and `source_X` identify your vehicle and the sender (if you are working with multiple vehicles you need to change them).

ozitoune, 2023-02-27:
Hello , sory for time to awnser ! 

@beniaminopozzan  thank you for all this precisions ! 

> Please, let me know if the sequence:

    Raspberry Pi switched off
    Pixhawk switching on with client automatically starting using custom startup (no parameters)
    Switch on Raspberry then start agent
Without -h option and the adress of the companion it's not working.
With starting the client with the companion adress is working, no mater if i start the agent in firts or the client in first, is always working !

Firts when i echo the topic /fmu/out/vehicle_control_mode; here is the result :
`---
timestamp: 1677492999523714
flag_armed: false
flag_multicopter_position_control_enabled: false
flag_control_manual_enabled: false
flag_control_auto_enabled: true
flag_control_offboard_enabled: false
flag_control_rates_enabled: true
flag_control_attitude_enabled: true
flag_control_acceleration_enabled: false
flag_control_velocity_enabled: true
flag_control_position_enabled: true
flag_control_altitude_enabled: true
flag_control_climb_rate_enabled: true
flag_control_termination_enabled: false
---
`
So i need to put the flag : " flag_control_offboard_enabled: false " to true ? how to do this ?
i see there is a topic : /fmu/in/offboard_control_mode, should i publish "true" here to modify the flag ?

If i understand i need to specifies in the actuator section that my servo 1 and 2 are controlled offboard ? but if i do this, can i control also with rc and the px4 autopiliot can control this 2 servo to control my model or are they "ONLY" offboard cotroled ?

Here is my problem, i want to use px4 with paramotor, the problem the is no frame paramotor. 

So there is 2 servo to control left brake, right brake, i need to control them with my rc to pilot manually, but i also need px4 control them to make return to home or waypoint navigation, but i also need ros2 control them for more high level algorithm like obstacle avoidance, trajectory, collaborative work , autopilot with artificial intelligence, ect....
Is it impossible to do this with px4 ?

beniaminopozzan, 2023-02-28:
> Without -h option and the adress of the companion it's not working.
With starting the client with the companion adress is working, no mater if i start the agent in firts or the client in first, is always working !

Great!

> i see there is a topic : /fmu/in/offboard_control_mode, should i publish "true" here to modify the flag ?

What do you want to achieve? Is it just to control the servos? If so then no, you don't need to change anything, the "`VEHICLE_CMD_DO_SET_ACTUATOR` way" is completely decoupled from the the drone control architecture.

But as you said:

> Here is my problem, i want to use px4 with paramotor, the problem the is no frame paramotor.

This changes everything!

> So there is 2 servo to control left brake, right brake, i need to control them with my rc to pilot manually, but i also need px4 control them to make return to home or waypoint navigation, but i [...]
Is it impossible to do this with px4 ?

Short answer, yes. Long answer... requires more work! You need to define a frame paramotor, this way you can treat the paramotor as any other supported vehicle and use manual control, automatic control (return to home, etc), offboard control. As we are going quite out of topic (way out of topic :) do you mind opening a new topic at https://discuss.px4.io/ , or a new issue here? Consider also this topic! https://discuss.px4.io/t/px4-paramotor-project-1-first-flight-with-px4/30881

closed 2023-04-04 by Jaeyoung-Lim