PX4/PX4-Autopilot #20642: Fatal error board/src/can.c

astekachev, 2022-11-19:
If configured CONFIG_CAN=y, CONFIG_STM32_CAN1=y in defconfig file and try make firmware

```
/src/can.c:26:10: fatal error: stm32_can.h: No such file or directory
   26 | #include "stm32_can.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
```

astekachev, 2022-11-20:
In 
PX4-Autopilot/platforms/nuttx/NuttX/nuttx/arch/arm/src/stm32h7 
and 
PX4-Autopilot/platforms/nuttx/NuttX/nuttx/arch/arm/src/stm32h7/hardware
does not exist files stm32_can.h stm32_can.c

PetervdPerk-NXP, 2022-11-28:
The UAVCAN can drivers run in userspace i.e. they dont use the NuttX HAL layer.
https://github.com/PX4/PX4-Autopilot/tree/main/src/drivers/uavcan/uavcan_drivers

I've been working on the SocketCAN implementation layer but that's only tested on the UCANS32K146 in theory we could move all the drivers to the SocketCAN HAL layer so we can easily use CAN with multiple apps on multiple platforms using the same interface.

astekachev, 2022-11-28:
> The UAVCAN can drivers run in userspace i.e. they dont use the NuttX HAL layer. https://github.com/PX4/PX4-Autopilot/tree/main/src/drivers/uavcan/uavcan_drivers
>
> I've been working on the SocketCAN [...]

Hi Peter!
I have FMU-V6X board and some time ago I wrote a driver to get data from MMW radar via CAN bus. But it turned out to be not working for the above reasons, then I tried to understand how FMU-V6X boards interacts with the CAN bus, because there is no driver at NuttX level. It turned out that the Socketcan UAVCAN with Cyphal is initialization of the CAN bus port and work with it, but only with the extended format of the frames. In my case I need to get standard frames on a can bus which also uses UAVCAN at different speeds - UAVCAN uses CAN 2.0B at 1Mbps, and Radar uses CAN 2.0A at 500kbps.
I'm not a very good programmer - I'm just learning, something simple I can still write by example. But how to connect to uavcan process and pull CAN 2.0A frames sent by radar yet - I'm confused with namespaces, references, pointers, CMakeList is the secret black magic for me :). I also had an idea to put some kind of global variable or structure into uavcan code to transfer frames, but it didn't work either when compiling files the linker gives an error.
Of course it would be great if there was a single NuttX HAL layer for CAN through which would work all the applications like uavcan/cyphal and others, to be honest I am extremely surprised that the CAN bus drivers are in UAVCAN application level and not in NuttX for STM32H7 boards. And the can.c file for initializing the board has not changed for several years and does not work for STM32H7 platforms. This is even more confusing and I would temporarily remove this file or at least indicate in the notes that it does not work for STM32H7 yet.

Best Regards, Alex

PetervdPerk-NXP, 2022-11-28:
Hi Alex,

So the situation is as follows

1. A dedicated CAN driver used by UAVCAN that claims the full CAN pheripheral and runs the driver in userspace
2. NuttX Character device CAN driver in NuttX /dev/canX but only works with 1 application at a time.
3. NuttX SocketCAN driver (Same as the Linux SocketCAN) that allows you communicate with CAN using Sockets and supports multiple applications using the same CAN pheripheral.

Inside PX4 only option 1 and 3 have been used, but for different purposes. Long term UAVCAN has to move over to option 3 so we can run multiple CAN applications on the same bus.

> In my case I need to get standard frames on a can bus which also uses UAVCAN at different speeds - UAVCAN uses CAN 2.0B at 1Mbps, and Radar uses CAN 2.0A at 500kbps.

This is simply not possible on CAN bus all peripherals have run at the same speed. What you could do is that you use CAN0 for UAVCAN and CAN1 for your radar.

astekachev, 2022-11-28:
> Hi Alex,
>
> So the situation is as follows
>
> 1. A dedicated CAN driver used by UAVCAN that claims the full CAN pheripheral and runs the driver in userspace
> 2. NuttX Character device CAN driver in NuttX [...]

Peter, thank you for your comment!
Then I will go the way of the bridge node UAVCAN. Then I will need to buy a radar with UART interface connect it to Teensy 4.0 and send data by the CAN bus with UAVCAN protocol but at a speed of 1Mbit