PX4/PX4-Autopilot #20644: Minimum RTL altitude not abided by

ryanjAA, 2022-11-21:
On Main (see log - SITL).

When minimum Return altitude is set higher to current alt and the vehicle is put in return, it doesnt climb.

https://review.px4.io/plot_app?log=740ea31d-34e3-4904-888f-1607481cabb2

junwoo091400, 2022-11-22:
Sort of related to https://github.com/PX4/PX4-Autopilot/pull/19971, @MaEtUgR didn't you say that the RTL altitude setting is practically ignored under certain conditions due to the RTL cone logic?

sfuhrer, 2023-02-02:
No, not related to https://github.com/PX4/PX4-Autopilot/pull/19971, as the cone is only active in MC mode and the flight here is in FW.

sfuhrer, 2023-02-02:
@ryanjAA can you reproduce with latest `main`? I quickly tried but couldn't.