PX4/PX4-Autopilot #20646: Loiter counterclockwise NAV_LOITER_RADIUS in firmware 1.11.3

theojaasp, 2022-11-21:
I want to loiter counter clockwise. 
When I do the mission plan I can do that with a - in front of the raduis I want to fly the Loiter WP. 
But when not reaching the altitude of Loiter WP it loiters clockwise with the value we gave at NAV_LOITER_RAD. This leads to problems because the plane has to change to counter clockwise then. How can I loiter counterclockwise when using NAV_LOITER_RAD?? Will a - in front of the value do it?

sfuhrer, 2023-02-15:
No, outside of a mission loiter it will always loiter CW. We would need a small code change to change the loiter default to CCW with a negative parameter NAV_LOITER_RAD - would you be able to make that change?

sfuhrer, 2023-02-28:
Let's make an explicit parameter for the direction. Let's also have a look at the mission interface for loiter direction (also separate instead of sign of loiter_radius field).

ryanjAA, 2024-03-28:
Ran into this problem today. Pinging @sfuhrer re this.

Seungwon-Yeo, 2024-03-28:
Dooray! 메일 발송 실패 안내

  
  
    
    
  
  
    
메일 발송
실패 안내
    
  
  
    
        
          
***@***.***) 
          님께
보낸
메일이
전송되지
못하였습니다.
        
        
          실패 사유를 확인해보세요.
        
    
  
  
    
      
        
          
            * 받는 사람 : 
***@***.***) 
          
          
            * 발송 시간 : 
2024-03-29T06:01:32
          
          
            * 메일 제목 : 
Re: [PX4/PX4-Autopilot] Loiter counterclockwise NAV_LOITER_RADIUS in firmware 1.11.3 (Issue #20646)
          
          
            
              
                * 실패 사유 : 
              
              
                
대외비/기밀 메일은 외부 계정으로 발송이 제한되어 일부 메일 발송 실패(수신자가 전달 설정한 경우 포함)
                
              
            
                      
          
        
      
    
  
  
    
        
          이 메일은 발신전용으로 회신되지 않습니다.
          더 궁금하신 사항은
          ***@***.***
          으로 문의해 주시기 바랍니다.
        
    
  
  
    
        © Dooray!.

theojaasp, 2024-08-01:
No I am not able to make the code change..

Sent with [Proton Mail](https://proton.me/) secure email.

sfuhrer, 2024-08-14:
To make sure we talk all of the same thing: You want to be able to change the default loiter direction from CW to CCW through a parameter? When would you for example set that?

ryanjAA, 2024-08-14:
Ya. Actually I think it should be qgc side but we need firmware support for it. + cw and - ccw is prob all we need.

sfuhrer, 2024-08-14:
> Ya. Actually I think it should be qgc side but we need firmware support for it. + cw and - ccw is prob all we need.

Can you elaborate what you mean with "should be qgc side"? Are you talking about the option to set the loiter direction for a GoTo setpoint? Or how else do you command loiters?

ryanjAA, 2024-08-14:
Oh I guess you're right. disregard last comment. You need the ability both to be able to change the direction (cw/ccw) in the params so when you hit it via RC switch or QGC mode to loiter it goes the direction you want but what I meant (also) was that when clicking goto, you could also set the direction (which is done param side), it's only in mission plans you can choose the direction

Recap:

Mission modes are fine, the ability to choose direction is already there
Non-mission modes: direction (cw/ccw) is chosen based on a param. Logic should listen to this in loiter mode, loiter switch via RC and Goto setpoint via QGC. What I would think would be good QGC side is to be able to also set a goto setpoint and tell it which direction I want.

theojaasp, 2024-09-16:
I think the best should be to have a parameter or 2 with a Loiter/Hold mode categeroy and then you can choose left for ccw and right for cw to have. So you then can put the hold mode/goto and mission mode loiter seperately

Sent with [Proton Mail](https://proton.me/) secure email.