PX4/PX4-Autopilot #20650: Make target px4_sitl_rtps not found

MineCold20, 2022-11-22:
Hello everybody!

I'm trying to setup ROS 2 - PX4 bridge as shown in https://docs.px4.io/main/en/ros/ros2_comm.html. Troubles start with "Sanity Check the Installation" section where I'm expected to make px4_sitl_rtps. When I run the command in the guide the output is: 

> Makefile:523: *** "Make target px4_sitl_rtps not found. It either does not exist or px4_sitl_rtps cannot be the first argument. Use 'make help|list_config_targets' to get a list of all possible [configuration] targets.".  Stop.

I've already build the PX4 Software by sending `make px4_sitl jmavsim`. After this, I've got only one folder in my build folder named "px4_sitl_default". 

Anyone knows how to build to have also px4_sitl_rtps target?

ripdk12, 2022-11-24:
Use the stable releases if you want to follow the docs.
Micro-RTPS has been removed in the main branch, and the documentation isn't quite ready yet.

MineCold20, 2022-11-29:
> Use the stable releases if you want to follow the docs. Micro-RTPS has been removed in the main branch, and the documentation isn't quite ready yet.

I used the last stable release (1.13.2) and I followed the docs but the error occurred. Is it possible that the repository has been updated without Micro-RTPS and the documentation refers to previous version?

junwoo091400, 2022-12-02:
Please refer to the pending documentation update PR on up to date info on ROS2 setup & target builds: https://github.com/PX4/PX4-user_guide/pull/2094

MineCold20, 2022-12-02:
> Please refer to the pending documentation update PR on up to date info on ROS2 setup & target builds: [PX4/PX4-user_guide#2094](https://github.com/PX4/PX4-user_guide/pull/2094)

@junwoo091400 thank you for your reply. Do you know which are my alternatives until documentation will be ready. I've tried to install previous PX4-Autopilot version but micrortps_agent command cannot yet be found. I believe that it is due to the new commit of px4_ros_com and px4_msgs packages. How can I download previous versions of these packages?

MineCold20, 2022-12-02:
Thank you all for your replies. I'm going to build previous version of PX4-Autopilot to make it work with rtps until the official documentation will be ready!

closed 2022-12-02 by MineCold20