PX4/PX4-Autopilot #20658: air speed - pitot erroneous behavior - sporadic readings

igal-colugo, 2022-11-23:
hi,
we are using the px4 version 1.13.0 on a cube black with MS4525DO pitot,
in VTOL mode.
we always experiencing a very glittery air speed measuring , even after repeated pitot calibration,
sometimes we get negative air speed readings - even when in fixed wing mode - while flying forward..

Steps to reproduce the behavior:
1. use a VTOL generic air frame
2. takeoff in MC state - stabilize mode
3. move to FW mode, - make sure throttle is in steady position - for example 75% (preferably in none windy condition - to eliminate disturbances) 
4. airspeed readings - are very sporadic... 

## Expected behavior
a steady (positive) stream of airspeed readings

## Log Files and Screenshots
https://logs.px4.io/plot_app?log=2ba55de7-8df5-4cc1-97c0-53922607ccbe

dagar, 2022-11-23:
It starts to look okay at speed, but otherwise it's pretty bad. For that particular sensor you'll need to calibrate often (it's sensitive to temperature changes). That might explain the offset, but I'm not sure about those spikes (more prominent in the second half).

[image]

What does your pitot and sensor installation look like? If you hold your finger against the tip of the pitot does it hold pressure showing a large positive airspeed?

igal-colugo, 2022-11-24:
Hi Daniel,

Yes- when I press my finger against the dynamic pitot hole – I get a steady – hi speed reading

Do you need any more information, I have numerous flights with the same issue,

Btw, can you recommend a more suitable pitot?  

 

Igal Gosis

Software

E ***@***.*** ***@***.***> 
36 Hayarkon st. I Yavne Israel I 8122739 

colugo-sys.com <http://colugo-sys.com>  | 

 

 <https://user-images.githubusercontent.com/84712/203587552-2676b139-4b99-42e0-85d6-08f457bb934b.png> 

This email has been checked for viruses by Avast antivirus software.
www.avast.com

ttropcic, 2022-11-28:
I was experiencing the same. 
We didn't fly with it yet, only tested on the bench, but the Sensirion SDP3x Airspeed Sensor looks a lot more temperature stable and is having less of those sporadic readings.

igal-colugo, 2022-11-28:
thank you for the input ttropcic!
we will try it to!