PX4/PX4-Autopilot #20667: Drone do not take off in offboard control example

biagio-trimarchi, 2022-11-24:
## Describe the bug
I am following the "ROS 2 Offboard Control Example" in the documentation: https://docs.px4.io/main/en/ros/ros2_offboard_control.html
The simulated drone arms the motor but does not take off as expected. 

## To Reproduce
Steps to reproduce the behaviour:
1. Run gazebo simulation: make px4_sitl gazebo
2. Run QGroundControl
3. Run micro ROS agent for communication: micro-ros-agent udp4 --port 8888
4. Run offboard control example from the package provided in the documentation: ros2 run px4_ros_com offboard_control 
5. Now the drone arms the motor and stays still

## Expected behaviour
I would expect the drone to take off.

## Screenshot
[image]

fabrizioschiano, 2022-11-24:
I had the same issue regarding the `No Manual Control input` and solved it by doing the following: 

- Download [QGroundControl.AppImage](https://d176tv9ibo4jno.cloudfront.net/latest/QGroundControl.AppImage).
```
wget https://d176tv9ibo4jno.cloudfront.net/latest/QGroundControl.AppImage
```
- Install QGroundControl
```
chmod +x ./QGroundControl.AppImage
```
- Run QGroundControl (Check if the software works on your computer).
```
./QGroundControl.AppImage
``` 

Then, enable `virtual joystick` to deactivate `No Manual Control Input`. As reported in this [forum](https://discuss.px4.io/t/failsafe-mode-keeps-activating-and-deactivating/25778), a joystick is required for safety reasons in `OFFBOARD mode`. The `Failsafe mode` can be avoided in simulation. However, **you should not** do this while using a real drone. It is dangerous as you do not have a manual backup to recover the drone!
To enable `virtual joystick` go to: *Select Tool* (big Q in the top-left corner) -> *Application Settings*. Select the option `virtual joystick`. **NOTE:** This is needed only once, QGroundControl will save this configuration.

Jaeyoung-Lim, 2022-11-24:
@biagio-trimarchi  You need to switch the vehicle into offboard mode

biagio-trimarchi, 2022-11-25:
I was able to run the example only after executing the command:
- commander mode offboard
on the px4 terminal as suggested by @Jaeyoung-Lim.
Shouldn't the offboard mode be activated by the ROS2 node provided in the example?

Jaeyoung-Lim, 2022-11-25:
@biagio-trimarchi Not really, since it is not safe to automatically enable offboard if you are testing on a real platform.

Especially given that ros2 you are publishing internal uorb messages directly.

biagio-trimarchi, 2022-11-25:
Thanks for the explanation. :)

closed 2022-11-25 by biagio-trimarchi

ElliotHYLee, 2022-11-26:
> To enable `virtual joystick` go to: _Select Tool_ (big Q in the top-left corner) -> _Application Settings_. Select the option `virtual joystick`. **NOTE:** This is needed only once, QGroundControl will save this configuration.

I had the same issue and was able to resolve thanks to this thread. 

Could I know what the QGC virtual joystick option does for the PX4 internally? Does it configure any parameters of any module in PX4? If not, is the gazebo somehow directly talking to QGC? I checked that if I don't turn on QGC, it does not hover.

For debugging purposes (no battery and PWM wires are connected), if I force to turn off this safety feature, which part should I look into?

Thanks in advance.