PX4/PX4-Autopilot #20668: Issues when using the SITL GZ launch configuration through VSCode Launch and Debug panel

lvanasse, 2022-11-24:
## Describe the bug
At the moment, when using the Launch and Debug window in VSCode and using the launch configuration `SITL (GZ)`, I get the error that ign is not present in the script on Ubuntu 22.04. And when testing it on Ubuntu 20.04.5 LTS, I also get errors.

## To Reproduce
Steps to reproduce the behavior:
1. On an Ubuntu 20.04 LTS or Ubuntu 22.04 LTS, download PX4
2. Install the dependencies with the `ubuntu.sh` script
3. Took off '....'
4. See error

## Expected behavior
A clear and concise description of what you expected to happen.

## Log Files and Screenshots
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/main/en/getting_started/flight_reporting.html)).
- Upload the log to the [PX4 Flight Review](http://logs.px4.io/)
- Share the link to the log (Copy and paste the URL of the log)

Add screenshots to help explain your problem.

## Drone (please complete the following information):
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

## Additional context
Add any other context about the problem here.

closed 2022-11-24 by lvanasse

ivanpadavan, 2023-02-28:
I had this issue not related to vscode stuff so I was not able to run gazebo simulation on macos.
To overcome this i needed to change GZBridge.cpp lines 276 and 119 (as for b5a6d6db0da2964d2fe4e40fd613fc579358dc68)
```cpp
if (_node.Request(create_service, req, 10000, rep, result)) {
...
int sleep_count_limit = 10000;
```
1000 to 10000

master0v, 2023-04-24:
Got the following error trying to run `make px4_sitl gz_x500`:

```
INFO  [gz_bridge] world: default, model name: x500_0, simulation model: x500
ERROR [gz_bridge] timed out waiting for clock message
ERROR [gz_bridge] Task start failed (-1)
ERROR [px4] Startup script returned with return value: 2
```

Tried the fixes above, that didn't help.

vacabun, 2023-04-25:
> Got the following error trying to run `make px4_sitl gz_x500`:
>
> ```
> INFO  [gz_bridge] world: default, model name: x500_0, simulation model: x500
> ERROR [gz_bridge] timed out waiting for clock message [...]

I also had the same problem.

lvanasse, 2023-04-25:
I think my fix was that I was missing some dependencies. It was fixed in the installation script a while back.

vacabun, 2023-04-26:
Thank you for your answer, it was fixed after i turn off the firewall.

bastianhjaeger, 2023-08-17:
In case anyone else stumbles over it: 
In my case there was a instance of the gazebo ignition server running in the background causing the issue. Killing it resolved things.

RodrigofrBastos, 2023-08-31:
> 

how do you kill the ig server running in the background?

bastianhjaeger, 2024-01-16:
Sorry, oversaw this. I killed it be searching for in in htop and killing it there.

usmanNoor5, 2024-10-06:
i have same error did anyone solves it

Meowchine1, 2025-07-23:
Faced a problem with Gazebo and px4 interaction: gz_bridge failed to start and spawn model. Can anyone help? 
```
 MicroXRCEAgent udp4 -p 8888
 gz sim -v 4 aruco_field.sdf
 export PX4_GZ_MODEL_NAME=x500_vision_0 && 
 export PX4_GZ_MODEL_POSE='3,3,0.5,0,0,0,0,0' && 
 cd ~/PX4-Autopilot && make px4_sitl gz_x500_vision 
```
Error:
```
INFO [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
INFO [init] found model autostart file as SYS_AUTOSTART=4005
INFO [param] selected parameter default file parameters.bson
INFO [param] importing from 'parameters.bson'
INFO [parameters] BSON document size 316 bytes, decoded 316 bytes (INT32:13, FLOAT:3)
INFO [param] selected parameter backup file parameters_backup. bson
INFO [dataman] data manager file './dataman' size is 7872608 bytes
INFO [init] Gazebo simulator
INFO [init] gazebo already running world: aruco_world
WARN [init] PX4_GZ_MODEL_POSE not set, spawning at origin.
INFO [gz_bridge] world: aruco_world, model name: x500_vision_0, simulation model: x500_vision
INFO [gz_bridge] Requested Model Position: 0,0,0,0,0,0,0,0,0,0
INFO [gz_bridge] Model position z is less than or equal to 0. 0, moving upwards
ERROR [gz_bridge] timed out waiting for clock message
ERROR [gz_bridge] Task start failed (-1)
ERROR [init] gz_bridge failed to start and spawn model
ERROR [px4] Startup script returned with return value: 256
FAILED: src/modules/simulation/gz_bridge/CMakeFiles/gz_x500_vision /home/vor/PX4-Autopilot/build/px4_sitl_default/src/modules/simulation/gz_bridge/CMakeFiles/gz_x500_vision 
cd /home/vor/PX4- Autopilot/build/px4_sitl_default/src/modules/simulation/gz_bridge && /usr/bin/cmake -E env PX4_SIM_MODEL=gz_x500_vision /home/vor/PX4-Autopilot/build/px4_sitl_default/bin/px4
ninja: build stopped: subcommand failed.
make: *** [Makefile:232: px4_sitl] Error 1
```

Translated with [DeepL.com](https://www.deepl.com/?utm_campaign=product&utm_source=web_translator&utm_medium=web&utm_content=copy_free_translation) (free version)

usmanNoor5, 2025-07-23:
I guess using webots simulator for this is a better choice