PX4/PX4-Autopilot #20681: Auto-detect INAXXX Digital Power Sensors

vincentpoont2, 2022-11-29:
Currently, when users use a INA228 or INA238 Power sensors with FMUv6X, they will have to go enable & disable the parameters manually. Is there a way to Auto-detect or enable them all so user do not have to find out what sensor they have in the power module and then manually configure this?

[image]

dagar, 2022-11-29:
What's the full list of possible power modules?

Can we assume only the power ports and 1 INA per bus?

vincentpoont2, 2022-12-01:
@dagar Currently there are INA226/INA228/INA238, each bus only have one power ports and will only have one INA.

For our Pixhawk 6X, besides the INA chip, there are other devices on the bus. there are MAG (IST8310) and RGB lamp driver chip (NCP5623C) on the GPS.

julianoes, 2022-12-06:
I had a look at the datasheets: [INA226](https://www.ti.com/lit/ds/symlink/ina226.pdf), [INA228](https://www.ti.com/lit/ds/symlink/ina228.pdf), [INA238](https://www.ti.com/lit/ds/symlink/ina238.pdf).

The I2C address can vary depending on the wiring of the A0 and A1 pins, so we will have to auto-detect the configurations used.

The chip can be determined using bits 15-4 of the Die ID Register (`0xFF`):

- INA226: `0x2260`
- INA228: `0x2280`
- INA238: `0x2380`

So that tells me that it should be possible - at least in theory - to auto-detect which one is connected on which I2C bus, right?

julianoes, 2023-04-10:
@dagar should we combine the 3 drivers into one that can then do the auto-detection?

julianoes, 2023-12-10:
@MaEtUgR I agree. I wasn't aware that @niklaut  was creating something Auterion specific while I was trying to address the problem a bit more general, as instructed by @dagar.

niklaut, 2023-12-12:
The auterion PM was upstreamed so that we have the same functionality as before. It additionally to detection and auto starting sets different shunt resistor values depending on address, so it does a little more. I guess this is where a device tree config would really shine.

vincentpoont2, 2023-12-12:
@niklaut @ThomasDebrunner @julianoes 
Now that there is a Auterion PM selector that can set different shunt resistor value here https://github.com/PX4/PX4-Autopilot/pull/22073. Can we make changes to the INA238_SHUNT Default Value for consistency? 

Currently [INA226 ](https://github.com/PX4/PX4-Autopilot/blob/35c1d5d9b1f80bb964f33a6686774d8a85415f77/src/drivers/power_monitor/ina226/ina226_params.c#L76)& [INA228](https://github.com/PX4/PX4-Autopilot/blob/35c1d5d9b1f80bb964f33a6686774d8a85415f77/src/drivers/power_monitor/ina228/ina228_params.c#L76) both has value 0.0005 (500μOhm Shunt), but INA238 had value of 0.0003 (300μOhm Shunt).

We really want to keep all the shunt resistor the same across the board, so the user can just plug and play, instead of keeping track of which shunt resistor is used and which parameter to modify. We uses all INA226, INA228, and INA 238. 

This is the PR I suggested: https://github.com/PX4/PX4-Autopilot/pull/22051

In Ardupilot, it is also set to be 500μOhm across the board.