PX4/PX4-Autopilot #20687: Arkflow flashes solid RED

hichamhendy, 2022-11-30:
## Describe the bug
A clear and concise description of the bug.

I need to use the arkflow to calculate an estimate of the position. The sensor is working just fine sice I tried with another flight controller (cube) and all good. Ad I said, it worked one before but at some point, I used othe parameter configuration to fuse SLAM and after that tried to have both together; yet, I noticed the sensor isn't working with my flight controller anymore

Firmware Version 1.12.3 and last 1.13 (the same behavior)
Flight Controller: FMUv5 of Holybro

The error might be related to UAVCAN but I can't trace. I get the following when I check the status 

`
NuttShell (NSH) NuttX-10.2.0
nsh> uavcan status
Pool allocator status:
    Capacity hard/soft: 500/250 blocks
    Reserved:  91 blocks
    Allocated: 44 blocks

UAVCAN node status:
    Internal failures: 0
    Transfer errors:   0
    RX transfers:      0
    TX transfers:      20419

CAN1 status:
    HW errors: 304103
    IO errors: 323734
    RX frames: 0
    TX frames: 3157
CAN2 status:
    HW errors: 2
    IO errors: 2
    RX frames: 795
    TX frames: 22827

control latency: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
INFO  [mixer_module] Switched to rate_ctrl work queue: 0
INFO  [mixer_module] Mixer loaded: no
INFO  [mixer_module] Driver instance: 0
INFO  [mixer_module] Channel Configuration:
INFO  [mixer_module] Channel 0: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 1: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 2: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 3: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 4: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 5: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 6: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 7: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 8: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 9: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 10: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 11: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 12: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 13: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 14: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 15: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191

Sensor 'baro':
name: uavcan_baro
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty

Sensor 'mag':
name: uavcan_mag
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty

Sensor 'flow':
name: uavcan_flow
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty

Sensor 'battery':
name: uavcan_battery
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty

Sensor 'airspeed':
name: uavcan_airspeed
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty

Sensor 'differential_pressure':
name: uavcan_differential_pressure
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty

Sensor 'rangefinder':
name: uavcan_rangefinder
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty

Sensor 'accel':
name: uavcan_accel
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty

Sensor 'gyro':
name: uavcan_gyro
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty

Online nodes (Node ID, Health, Mode):

uavcan: cycle time: 298872 events, 16834506us elapsed, 56.33us avg, min 15us max 3736us 132.985us rms
uavcan: cycle interval: 298872 events, 2738.96us avg, min 26us max 8260us 1007.105us rms
nsh> `

## Expected behavior
I expect to get blue led as before!

## Additional context
As I said the whole contest to support SLAM with arkflow velocities and hence got more stable!

AlexKlimaj, 2022-11-30:
https://github.com/PX4/PX4-Autopilot/issues/18049#issuecomment-1332445085

AlexKlimaj, 2022-12-08:
https://github.com/PX4/PX4-Autopilot/issues/18049#issuecomment-1343361469

closed 2022-12-08 by AlexKlimaj