PX4/PX4-Autopilot #20690: Remote ID - Dynamic Operator Position

ryanjAA, 2022-12-01:
Remote ID will require the dynamic operator position sent/transmitted <1 second from when timestamped (sent not received).

This will likely require some small changes to send up GCS position and 2hz is probably more than sufficient given above. 

Assuming that this will need a GPS connected to the GCS (USB or internal).

Want to make sure we aren't over-burdening the link.

As discussed on todays dev-call @dagar

ryanjAA, 2022-12-02:
As noted in Discord, the FAA is requiring that that GCS GPS needs to be using SBAS (or if internal, to be using a GPS integrated into a phone or tablet). 

Thanks Josh Henderson (Git @ doesnt work on his hendjoshsr71)

BluemarkInnovations, 2022-12-09:
Let me know if I can test something.