PX4/PX4-Autopilot #20692: Import custom model for sitl px4

airsimdevdd, 2022-12-01:
when I'm importing custom model into px4  i can't find file platforms/posix/cmake/sitl_target.cmake i
to add my model

dagar, 2022-12-01:
Things have moved around a bit, as we're working towards making it much simpler to integrate with simulators.

Can you expand on what you're trying to do exactly?

The list of models for sitl_gazebo is now in `src/modules/simulation/simulator_mavlink/sitl_targets_gazebo.cmake`.

https://github.com/PX4/PX4-Autopilot/blob/c5dc1221b68352725630cbda5c04bc0f5c3c1617/src/modules/simulation/simulator_mavlink/sitl_targets_gazebo.cmake#L64

airsimdevdd, 2022-12-02:
I want to add a quadrotor frame of dji f450 @dagar thank for ur help

closed 2023-03-15 by airsimdevdd