PX4/PX4-Autopilot #20693: DMA issue on fmuv6x with neopixel driver

baschuuuu, 2022-12-01:
## Describe the bug
Running the neopixel driver on a holybro pixhawk 6x does not produce the expected signal.
I use the dma-version of this driver: `srgbled_dma.cpp` and the latest release (v1.13.2).

## To Reproduce
Added to `fmu-v6x/src/board_config.h`:
```
#  define BOARD_HAS_N_S_RGB_LED          10
#  define S_RGB_LED_DMA                             DMAMAP_TIM5_CH4
#  define S_RGB_LED_TIMER                          5   /* timer 5   */
#  define S_RGB_LED_CHANNEL                    4   /* channel 4  */
// #  define S_RGB_LED_CHANNELN              1   /* channel 4N */
#  define S_RGB_LED_TIM_GPIO                    GPIO_TIM5_CH4OUT_2 /* PI0 aka FMU_CH1*/
```
Build and flash, then in nsh:
`neopixel start`

## Expected behavior
I would expect to see a pwm-like signal on pin FMU_CH1 of two different duty-cycles.
Total bit time is 1200 ns, 0b1 is ----___, (600ns|600ns) 0b0 is ---____ (300ns|900ns).

## Log Files and Screenshots
The signal on FMU_CH1 is either constantly zero, or
if I enable DMA-debug-prints and disable DMA on the serial console (because it would freeze otherwise) I get a weird non-const signal with some pulses that have nothing to do with the expected behaviour.

log:
https://logs.px4.io/plot_app?log=b063b920-a603-4820-9d73-71acd0f30f78

## Drone (please complete the following information):
holybro pixhawk 6x on baseboard mini, powered with USBC.

## Additional context
With print statements in `stm32h7/stm32_dma.c` (for debugging), the serial console freezes.
Configuring the serial console without DMA would solve the freezing issue.

baschuuuu, 2022-12-01:
@davids5 do you have any hints on how to debug this?

closed 2023-01-13 by baschuuuu

FrankvVeelen, 2023-03-16:
Did you get any further with this? running into similar issues

dagar, 2023-03-16:
> Did you get any further with this? running into similar issues

@FrankvVeelen What hardware do you have exactly and what are you seeing? Specific lights, flight controller, how it's connected, etc.

davids5, 2023-03-16:
@FrankvVeelen  Did you  validate that the `dma_handle` from `stm32_dmachannel(S_RGB_LED_DMA); ` is not null?

davids5, 2023-03-16:
Where did `DMAMUX1_TIM12_CH1` come from? I do not see a DMA source from TIM12 in arch/arm/src/stm32h7/hardware/stm32h7x3xx_dmamux.h

Also is It needs to be a in a DMAMAP_DMA12_XXXX like entry to provide a DMA{1|2} and a mux

FrankvVeelen, 2023-03-16:
Hey @davids5 , thanks for taking a look at the problem. ```DMAMUX1_TIM12_CH1``` was my shot in the dark at trying to get it to work, since DMA TIM12 does not seem to exist. Thanks for the pointer to ```DMAMAP_DMA12_XXXX```!

Here is where I'm at now:
I have for now changed the part in my board_config.h to:
```
#  define S_RGB_LED_DMA              	 DMAMAP_DMA12_TIM1CH2_1
#  define S_RGB_LED_TIMER                1   /* timer 12    */
#  define S_RGB_LED_CHANNEL              2   /* channel 1  */
// #  define S_RGB_LED_CHANNELN             1   /* channel 2N */
#  define S_RGB_LED_TIM_GPIO             GPIO_TIM1_CH2OUT_2 // GPIO_TIM12_CH2OUT_2
```
In my understanding this should result in a LED control on ```PWM_FMU3```. The code now compiles, and the FCU doesn't crash upon ```neopixel start```. However the LEDs are also not driven.
I have verified that the ```dma_handle``` is not null. At the moment I do not have access to a scope, but i will at a later moment. Is there any other way to verify that "something" is actually happening?

Another question I have:
Since I have already produced a few PCBs for the flight controllers, I'm kind of stuck with using FMU7 & FMU8 for two different sets of LEDs. Both of those use TIM12. Is it not possible to use DMA with TIM12? Would it be possible with some changes in software, or is there a hardware limitation in the stm?

davids5, 2023-03-17:
> Hey @davids5 , thanks for taking a look at the problem. `DMAMUX1_TIM12_CH1` was my shot in the dark at trying to get it to work, since DMA TIM12 does not seem to exist. Thanks for the pointer to [...]

There are only 8 channels per DMA [see](https://github.com/PX4/PX4-Autopilot/blob/main/boards/px4/fmu-v6x/nuttx-config/include/board_dma_map.h) for the utilization for the DMA . Yow may need trade off one of the DMA2 selections 
Maybe the [debug console](https://github.com/PX4/PX4-Autopilot/blob/main/boards/px4/fmu-v6x/nuttx-config/include/board_dma_map.h#L72) 

> Another question I have: Since I have already produced a few PCBs for the flight controllers, I'm kind of stuck with using FMU7 & FMU8 for two different sets of LEDs. Both of those use TIM12. Is it [...]

Refer to DMAMUX1, DMA1 and DMA2 connections in the STM32H7 reference manual.  I think only `tim12_trgo` is supported. IIRC the code needs to do synchronous update on the DMA request from the compare request and that is not wired to the DMA mux. 

I am not sure what your application is, to advise you on how to solve the issues.

FrankvVeelen, 2023-03-20:
Hey there again, 
I have got it "working" with ```PWM_FMU5```. The leds turn on and respond to changes. The problem I'm still having is that the reset part of the protocol does not seem to work. I have got access to my oscilloscope today, and i can see all bits being transmitted as they should be, but the reset voltage seems to be randomly high or low (the time in between bits being sent). 

Looking at the code I dont see any obvious way in which this bit would be set either, do you have an idea how it should be set?
If needed I can provide some oscilloscope shots.

davids5, 2023-03-20:
@FrankvVeelen It has been a while since worked on this. I though it is was the lack of a transition that constituted a reset. Do you have the datasheet?

FrankvVeelen, 2023-03-20:
I understand @davids5 . Thanks for your time providing this support. These are the "official" [Adafruit](https://cdn-shop.adafruit.com/datasheets/WS2812B.pdf) ones. The datasheet specifies a "low voltage time" of >50us. The specific ones i use are a little bit different [WS2812B-V5](http://www.peace-corp.co.jp/data/WS2812B-V5_V1.0_EN.pdf) and specify a "low voltage time" of >280us. All different versions I looked at do specify a low voltage time. It could of-course be the case that some modules do allow a no-transition time, but it's not what the spec says.

Maybe we could add driving the voltage low on the dma_callback, or somehow make sure that the last bit drives the output low? If you have any advice on what would be the best way to drive the output low in between packets, I can test it on my setup. I will also test with some official adafruit neopixels to see if they behave any different.

davids5, 2023-03-21:
Could it be Line 387. Preloading the first bit? , try writing it to a 1 (SLED_rCCR = 1); 

To see what the code is doing you can use the PROBE() macros. see board.h but avoid the FMU channel you are using! Adding PROBE(1,0) before the `stm32_dmastart` and PROBE(1,1); in the `dma_callback` will help to see the timing relative to the bit stream.

davids5, 2023-03-23:
@FrankvVeelen, it could be an issues with the pwm output driver.  To test it try adding a `return 0;` to the main.  `extern "C" __EXPORT int pwm_out_main(int argc, char *argv[]); If not, then it is time to debug the hardfault.

davids5, 2023-03-24:
@FrankvVeelen Also what are all the DMA assignments?

farhangnaderi, 2024-08-03:
> Hey there again, I have got it "working" with `PWM_FMU5`. The leds turn on and respond to changes. The problem I'm still having is that the reset part of the protocol does not seem to work. I have [...]

Hey @FrankvVeelen How did you choose your GPIOs?
I am trying to do this but no luck so far! 
Would be great if you could give a hint. 
@davids5 Is there anyway I can find GPIO Map for 6x board? Like how I can find the way ```GPIO_TIM1_CH2OUT_2``` is defined?

omapex, 2024-10-23:
@FrankvVeelen, can you summarize all the steps and final changes you made to successfully get neopixel driver to perform led_control?