PX4/PX4-Autopilot #20694: Control Allocator - Can't tie ailerons up (crow) to flap deployment

ryanjAA, 2022-12-01:
On main. Pixhawk 4 for testing.

With the mixer setup we have ailerons go up when flaps are deployed to slow down.

On CA, there isn't a way (that I can figure out) to get this to do the same.

When in mission mode during  flight, flaps should be up, ailerons should be ailerons. During Landing, flaps go down, ailerons go up. 

During takeoff only flaps should go down if set (need to be decoupled from the ailerons or takeoff would be hard).

Aileron deflection should be a settable amount (not hardcoded).

During non mission mode you should be able to tie flaps and the crow/speedbrake setup to a channel and use it to slow down for a manual landing.

@dagar as discussed just now re 1.14 blocker

junwoo091400, 2023-06-29:
@ryanjAA did https://github.com/PX4/PX4-Autopilot/pull/20815 fix this issue?

ryanjAA, 2023-06-29:
> @ryanjAA did #20815 fix this issue?

Yes. Fixed and closed.

closed 2023-06-29 by ryanjAA