PX4/PX4-Autopilot #20698: Add more parameters to Multi-Vehicle Simulation scripts

akontsevich, 2022-12-02:
## Describe problem solved by the proposed feature
`HEADLESS=1` parameter has no impact on Multi-Vehicle Simulation scripts:
- [sitl_multiple_run.sh](https://github.com/PX4/PX4-Autopilot/blob/main/Tools/simulation/sitl_multiple_run.sh)
- [gazebo sitl_multiple_run.sh](https://github.com/PX4/PX4-Autopilot/blob/main/Tools/simulation/gazebo/sitl_multiple_run.sh)
- [jmavsim_run.sh](https://github.com/PX4/PX4-Autopilot/blob/main/Tools/simulation/jmavsim/jmavsim_run.sh)

How to run HEADLESS in this case?

Also [mavlink module](https://docs.px4.io/main/en/modules/modules_communication.html#mavlink_usage) allows to  [Enable UDP Broadcasting](https://docs.px4.io/main/en/simulation/#enable-udp-broadcasting) or  [Enable Streaming to Specific Address](https://docs.px4.io/main/en/simulation/#enable-streaming-to-specific-address), however Multi-Vehicle Simulation scripts do no allow to specify any parameter for streaming to a specific IP-address or broadcasting.

Single vehicle simulation also does not allow to enable streaming or broadcasting:
```
HEADLESS=1 make px4_sitl gazebo_typhoon_h480
```

Need to such parameters or scripts appear to be very incomplete, need to write hacks to enable these features in them. Any thoughts how can we do this?

dagar, 2022-12-02:
Can we make a list of specific use cases and work through the missing pieces?

Personally I think new Gazebo would be the best baseline for a proper multi-vehicle simulation going forward.

dagar, 2022-12-02:
If possible I'd like to avoid writing more custom helper scripts that sit around untested/unmaintained. I'd like to find one good multi-vehicle simulation work flow and get some testing in place.

@Jaeyoung-Lim @bperseghetti

bperseghetti, 2022-12-02:
@dagar @Jaeyoung-Lim I think with new gazebo you can just let people create their own multi-vehicle worlds, save them and run them.

I personally see very little value in the previous multi-vehicle scripts from my commercial and R&D perspectives.

For instance, when I have a super unique test world I typically have super unique vehicles in specific locations for my multi-vehicle tests, current scripts are really only useful (in my opinion) for testing QGC multi-vehicle connections/Basic CI and novel "look how many vehicles that I launched". 

I think we should inquire what scripts people are using for real meaningful simulation and why. More importantly what are they testing with it and how does it have value (or not) in the current scripts.

I also think that running a simulation from a build should be a "novel way" and NOT a suggested way of running the simulation.

I would prefer to use the simulators with their full features and let PX4 work WITH the simulator instead of having it break the simulators full functionality by incomplete feature-sets that are very hard/brittle (with few select options) from scripts.

Note this is just my opinion though.

dagar, 2022-12-02:
> I also think that running a simulation from a build should be a "novel way" and NOT a suggested way of running the simulation.

:+1:

dagar, 2022-12-02:
I largely agree, but I still think we could maintain some minimal documented example to point people towards when they inevitably ask.

akontsevich, 2022-12-02:
I do not like to create Gazebo worlds - could be big task for us. We use your scripts for single or multiple drone simulation with video broadcasting utilizing our QGroundControl fork for various scenarios: MV automatic mission path generation and flying, flight conformance monitoring (including speed, bearing, altitude, etc), manual control, drone sensors control, etc etc. And we stream this to different devices: PC, Android tablet, phone, GCS, etc. Good to have simple script to satisfy our needs. We crated docker image for this which runs something like this:
```
export PX4_HOME_LAT=${LATITUDE}
export PX4_HOME_LON=${LONGITUDE}
export PX4_HOME_ALT=${ELEVATION}
export HEADLESS=1

    /PX4-Autopilot/Tools/simulation/gazebo/sitl_multiple_run.sh \
    -w "${WORLD}" -s "${VEHICLE_MODEL}:${NUM_VEHICLES}"
```
But HEADLESS does not work and no IP streaming or broadcasting as well. Same for jmavsim.

dagar, 2022-12-02:
> But HEADLESS does not work and no IP streaming or broadcasting as well. Same for jmavsim.

HEADLESS we can fix quickly here, but what do you want for per instance broadcast and streaming ip? 

> Same for jmavsim.

I would strongly recommend any jmavsim interest or development effort be redirected to new Gazebo at this point.

dagar, 2022-12-02:
> We want `mavlink start -p -t ${IP_ADDRESS}`. For now we use weird script that patches all run commands in `/build/px4_sitl_default/etc/init.d/*`

Is that the same ip address for all instances?

lars-nagel, 2023-11-21:
Hey there! Are there any updates on the HEADLESS option? Got it fixed already?