PX4/PX4-Autopilot #20708: Unable to use UART3 on Kakute h7

TedObrien, 2022-12-06:
## Describe the bug
Using the PX4 v13.2 on the holybro kakute h7 I am unable to use ports `dev/tty/S1` or `/dev/tty/S2` as TELEM2. 
`
## To Reproduce
Steps to reproduce the behavior:
- Flash v1.13.2 
- Plug component such as telemetry unit into TELEM1 and configure baud rate 
- works perfectly
- now assign to telemetry 2 and set baud rate
- when `CONFIG_BOARD_SERIAL_TEL2="/dev/ttyS1"`  in `default.pxboard`,  telemetry does not work at all, https://docs.px4.io/main/en/flight_controller/kakuteh7.html states that "/dev/ttyS1" is not broken out so i guess this is unsurprising 
- If you change `CONFIG_BOARD_SERIAL_TEL2="/dev/ttyS2" ` then you can connect to QGC via telemetry but cannot send data to the drone, only receive. Perhapse there is a port conflict as `/dev/ttyS2 is` asigned as the "Dubug console" however i am unsure of how to confirm this

## Expected behavior
I would expect TELEM2 to behave like TELEM1, if `/dev/ttyS2` is asigned as the debug console, is there a way to disable it so i can use this port as an additional uart?
## Log Files and Screenshots
Please let me know if there is any additional information that would be helpful.
Many thanks

j-chen-opteran, 2023-01-09:
TX3/RX3 on the Kakute H7 is indeed the debug console according to the docs [here](https://docs.px4.io/main/en/flight_controller/kakuteh7.html). I've confirmed this - by sticking a FTDI tool and using a tool like `screen` you can access the NuttX console via terminal.

TedObrien, 2023-01-09:
Hi @j-chen-opteran thanks for getting back to me. I was actually aiming to use the TX3/RX3 pins for an external sensor, so was asking if it is possible to **disable** the NuttX debug console from using this port, as I do not currently need it, and it appears to be interfering. TELEM2 was not enabled by default but was added by me manually so i could try and use it for said sensor. 
Many thanks

j-chen-opteran, 2023-01-09:
AFAIK no way to do this unfortunately - I had the same issue on that board as I ran out of UARTs. The bluetooth chip consuming another UART doesn't help :(

TedObrien, 2023-01-11:
That’s a shame, thanks for the help!

julianoes, 2023-01-16:
So this can be closed?

closed 2023-01-17 by TedObrien