PX4/PX4-Autopilot #20731: Log Download on Cube Orange

danielhonies, 2022-12-08:
## Describe the bug
On current main building on Ubuntu, log download through QGC doesn't work on the Cube Orange. When trying to download, QGC shows 'error'

## To Reproduce
Build Main, Flash on Cube Orange, Fly, try to download log

ryanjAA, 2022-12-09:
This seems to be a common issue. Can you save your Params and upload that here. 

Is anyone not having this issue and we can compare?

TedObrien, 2022-12-14:
Also having issues with this on 13.2, did you find a solution?

ripdk12, 2022-12-20:
I can only download the log when connected via telemetry radio, when I try to download log while plugged in USB I get the same error.

danielhonies, 2023-01-09:
fixed by https://github.com/PX4/PX4-Autopilot/pull/20824

closed 2023-01-09 by danielhonies