PX4/PX4-Autopilot #20747: v1.13: missing logged topics

ThomasRigi, 2022-12-12:
## Describe the bug
After https://github.com/PX4/PX4-Autopilot/pull/18645, VTOLs don't have actuator_outputs.00 and .01 logged anymore. 

Instead, actuator_outputs.02 appears. From `listener actuator_outputs` in mavlink shell, the old topics are no longer published.
[image]

## Expected behavior
Log them as before.

## Log Files and Screenshots
https://logs.px4.io/plot_app?log=23b4df2c-d1fd-4df4-bfdc-8e774b1bda36 : Bench test with Generic Quadplane VTOL airframe.

## Drone (please complete the following information):
- Pixhawk 4

## Additional context
Discord conversation : https://discord.com/channels/1022170275984457759/1045633346266468422

bkueng, 2022-12-13:
As per discord, this is because of uavcan (esc's + servo's) is enabled. We can add the 4th instance to the log as well in 1.13 (for control allocation it's different again).
An alternative is to advertise when/if a mixer is loaded.

ThomasRigi, 2022-12-13:
Just to make sure I understood correctly:
The Uavcan module is faster in advertising its actuator_outputs, and takes the spot 00 and 01. And once the mixer's initialisation happens, it gets assigned instances 02 and 03, and these are the PWM values that I'm actually interested in getting. 

So the quick-and-dirty option would be to just put `add_optional_topic_multi("actuator_outputs", 100, 5);`, but then we also need to adjust flight review for them to be visible there I suppose.

And your alternative of advertising when a mixer is loaded would happen before uavcan initialisation, meaning that the topics should once again be on instances 00 and 01 as they were in v1.12.

I'd prefer the alternative I think, as it's more logical and consistent to me if actuator_controls.00/01 maps to actuator_outputs.00/01 respectively, but I have to admit that I don't understand the bootup sequence well enough to implement it myself.

bkueng, 2022-12-14:
Yes it's certainly cleaner but I also want make sure we don't introduce too many changes on 1.13. It's a small one though: https://github.com/PX4/PX4-Autopilot/pull/20767

sfuhrer, 2022-12-23:
Can this be closed with the fix for 1.13 being in? Or do we still need a fix for main?

ThomasRigi, 2022-12-23:
I think it can be closed as to my understanding it only affected setups
that are still using the old mixer files, which no longer exist on main.

closed 2023-01-17 by sfuhrer