PX4/PX4-Autopilot #20762: Pixhawk 6x: MAVLink not working on TELEM1 ports

FriedrichTuttas, 2022-12-13:
## Describe the bug
On the new Pixhawk 6X, there is no data coming through the TX pin of TELEM1. TELEM3 works, though.

## To Reproduce
1. Hold oscilloscope probe to TX pin of TELEM1 port.
2.  See no data coming through.
3. Hold Probe to TX pin of TELEM3 port.
4. See data.

## Expected behavior
Data coming through TELEM1.

FriedrichTuttas, 2022-12-13:
It gets weirder:

A while back, I wrote a module that communicates via TELEM3 port. Today, I altered it to use TELEM1 port instead to test the TELEM1 port and I get the following weird behavior:
After booting, there is no data coming through the TX pin of TELEM1. Then I start my module via the MAVLink console in QGC by typing "my_module_name start". Within the start function of my module, it opens the corresponding uart port of TELEM1, but, at this point, it does not write any data! However, as soon as I start my module, I can see data coming through the TX pin of TELEM1 in my oscilloscope with a baudrate of 57600. Now, I can stop my module again by typing "my_module_name stop" and the data will keep coming.

So, opening the uart port in a different module must somehow enable MAVLink to use this port. I have no idea where to look to fix this issue.

FriedrichTuttas, 2022-12-14:
For testing, I installed Ardupilot on the Pixhawk 6X and here the TELEM1 port works without a problem.

julianoes, 2022-12-22:
Does it work with stock PX4 v1.13.2 flashed without your changes, with the appropriate default params set?
```
MAV_0_CONFIG 101
MAV_0_FLOW_CTRL 2
MAV_0_MODE 0
MAV_0_RADIO_CTL 1
MAV_0_RATE 1200
SER_TEL1_BAUD 57600
```

If the stock image works, then I would say this is an issue with your code additions/changes. Without seeing them it would then be hard for anyone to diagnose the problem

fritut08, 2022-12-22:
I flashed version 1.13.2 directly from within QGC and it still does not work.

julianoes, 2022-12-22:
That's with the parameters reset?

fritut08, 2022-12-22:
I cannot confirm right now but I wanna say I am pretty sure that I resetted them when I tried version v1.13.2.

ashu41, 2023-01-04:
I am facing the same issue. I have tested the stock PX4 v1.13.2 on Pixhawk 6x and Telem 1 does not work. To debug I did the following:

1. To make sure the board isn't damaged I flashed PX4 v1.12.3 on Pixhawk 6x and the telem 1 works fine. 
2. To verify if the stock PX4 v1.13.2 firmware, I flashed it on Pixhawk 5x and again telem 1 works fine. 

In conclusion, this bug is only with Pixhawk 6x in combination with the stock PX4 v1.13.2 firmware version. 

@FriedrichTuttas let me know if I can help in any way. If needed I can do some more tests to debug this issue.

julianoes, 2023-01-06:
@ashu41 can you try with params reset? Or can you post the params that you have set?

ashu41, 2023-01-06:
Yes, I have tried params reset. Here is the param file:
[06_01_23_v6x_default.params.txt](https://github.com/PX4/PX4-Autopilot/files/10359958/06_01_23_v6x_default.params.txt)

FriedrichTuttas, 2023-01-09:
@ashu41 Thank you for your input! This gives me the peace of mind that it was not just me. :D I did some more digging in the firmware source code searching for modules that use the TELEM1 port or use a baud rate of 57600 but I could not find anything suspicious. I thought that maybe the error stems from the fact that the uart port numbering changed. In the Pixhawk 4, the TELEM1 port was `/dev/ttyS1` ([source](https://docs.px4.io/main/en/flight_controller/pixhawk4.html#serial-port-mapping)) while it is `/dev/ttyS6` on the Pixhawk 6x ([source](https://docs.px4.io/main/en/flight_controller/pixhawk6x.html#serial-port-mapping)). I also had a look at the startup procedure for MAVLink but that all seemed alright to me as well. Therefore, I do not know where else to look.

ashu41, 2023-01-12:
I am also stuck with this bug and unable to figure out the source problem. I have 10 Pixhawk 6X boards but can't use it. I am hoping @julianoes and other PX4 devs will resolve this soon.

FriedrichTuttas, 2023-01-12:
> In conclusion, this bug is only with Pixhawk 6x in combination with the stock PX4 v1.13.2 firmware version.

I also tested v1.13.0 and v1.13.1 and have the same error there. @ashu41, you said that it worked in v1.12.3. So, the error must have been introduced with the changes between v1.12 and v.1.13.

julianoes, 2023-01-16:
@ashu41 have you tried more than one board? Just to rule out a hardware problem.

And can you confirm this is with default params?

julianoes, 2023-01-16:
Although, you already said that it would work at v1.12, so that speaks against hardware indeed.

julianoes, 2023-01-16:
I just verified this with v1.13.2 and it works fine for my Holybro 6X with default (bigger) baseboard. I have a telemetry radio connected to Telem1 to test this, and it works out of the box after a `param reset_all`.

FriedrichTuttas, 2023-01-16:
Interesting. I just tested it again. Here is my extensive step-by-step procedure (in case I messed something up):

1. Start QGC.
2. Go to Vehicle Setup -> Firmware
3. Connect Pixhawk 6X to computer through USB and wait until it is recognized by QGC.
4. "PX4 Pro Stable Release v1.13.2" is pre-selected. I do not change that and click OK to start flashing the new firmware.
5. Wait until the flashing process is completed and the board has reconnected to QGC.
6. Go to Analyze Tools -> MAVLink Console.
7. Issue command `param reset_all`. I exported the parameters after the reset to [this file](https://github.com/PX4/PX4-Autopilot/files/10423527/params.txt).
8. Unplug the USB cable of the Pixhawk 6X.
9. Connect telemetry radio to TELEM1 port, connect the other telemetry radio to a USB port of my computer.
10. Connect the USB port of Pixhawk 6X to power using an adapter for an AC power socket (so as not to connect it to the computer via USB).
11. QGC will not connect to the Pixhawk 6X. Also, the LED on the telemetry radio connected to the Pixhawk 6X that is supposed to indicate transmission of data does not flash at all. Also, I cannot see any data on the TELEM1 port's TX pin in my oscilloscope.

I tested this on the Pixhawk 6X with the default and the mini baseboard. Same result. With the default baseboard, I continued to test the TELEM3 port with the following steps:

12. Unplug telemetry radio from the TELEM1 port.
13. Connect Pixhawk 6X to computer through USB and wait until it is recognized by QGC.
14. Go to Vehicle Setup -> Parameters.
15. Search for "mav" and change the parameter `MAV_0_CONFIG` from TELEM1 to TELEM3.
16. Unplug the USB cable of the Pixhawk 6X.
17. Connect telemetry radio to TELEM3 port.
18. Connect the USB port of Pixhawk 6X to power using an adapter for an AC power socket.
19. After a few seconds, QGC will connect to the Pixhawk 6X through the telemetry radio. Its LED flashes indicating data transmission and the oscilloscope confirms data on the TELEM3 port's TX pin.

So, with my Pixhawk 6X on PX4 v1.13.2, TELEM3 works and TELEM1 does not.

To confirm again, I repeated the first 10 steps, but instead of flashing PX4 on step 4, I flashed Ardupilot (leaving all the dropdowns as their default value), and I omitted steps 6 and 7. Running Ardupilot, the QGC will connect to the Pixhawk 6X through its TELEM1 port without a problem.

@ashu41 can you confirm that you get the same results following these steps?

julianoes, 2023-01-16:
Thanks @FriedrichTuttas. I followed your exact steps. I also exported my params and they were identical apart from the GYRO calibration bias. For me MAVLink on Telem1 is flowing as expected.

Can you share the output of `mavlink status`? You can just get that using MAVLink shell via USB.

For me, it's as follows when connected via radio:

```
nsh> mavlink status

instance #0:
    GCS heartbeat valid
    mavlink chan: #0
    type:        RADIO Link
      rssi:        210
      remote rssi:    209
      txbuf:    100
      noise:    46
      remote noise:    48
      rx errors:    0
      fixed:    0
    flow control: ON
    rates:
      tx: 759.9 B/s
      txerr: 0.0 B/s
      tx rate mult: 0.850
      tx rate max: 1200 B/s
      rx: 62.7 B/s
      rx loss: 0.0%
    Received Messages:
      sysid: 51, compid: 68, Total: 215 (lost: 0)
      sysid:255, compid:190, Total: 226 (lost: 0)
    FTP enabled: YES, TX enabled: YES
    mode: Normal
    MAVLink version: 2
    transport protocol: serial (/dev/ttyS6 @57600)
    ping statistics:
      last: 174.94 ms
      mean: 170.62 ms
      max: 872.56 ms
      min: 113.10 ms
      dropped packets: 0

instance #1:
    mavlink chan: #1
    type:        GENERIC LINK OR RADIO
    flow control: OFF
    rates:
      tx: 0.0 B/s
      txerr: 80.9 B/s
      tx rate mult: 0.050
      tx rate max: 100000 B/s
      rx: 0.0 B/s
      rx loss: 0.0%
    FTP enabled: YES, TX enabled: YES
    mode: Normal
    MAVLink version: 1
    transport protocol: UDP (14550, remote port: 14550)
    Broadcast enabled: YES
nsh> 
```

FriedrichTuttas, 2023-01-17:
My output from `mavlink status` looks quite different (even though I made sure to issue `param reset_all` and restart the Pixhawk beforehand):

<details>
  <summary>Output</summary>
  
  ```
  instance #0:
      mavlink chan: #0
      type:        GENERIC LINK OR RADIO
      flow control: ON
      rates:
        tx: 0.0 B/s
        txerr: 0.0 B/s
        tx rate mult: 0.850
        tx rate max: 1200 B/s
        rx: 0.0 B/s
        rx loss: 0.0%
      FTP enabled: YES, TX enabled: YES
      mode: Normal
      MAVLink version: 1
      transport protocol: serial (/dev/ttyS6 @57600)
  
  instance #1:
      mavlink chan: #1
      type:        GENERIC LINK OR RADIO
      flow control: OFF
      rates:
        tx: 0.0 B/s
        txerr: 17.0 B/s
        tx rate mult: 0.050
        tx rate max: 100000 B/s
        rx: 0.0 B/s
        rx loss: 0.0%
      FTP enabled: YES, TX enabled: YES
      mode: Normal
      MAVLink version: 1
      transport protocol: UDP (14550, remote port: 14550)
      Broadcast enabled: YES
  
  instance #2:
      GCS heartbeat valid
      mavlink chan: #2
      type:        USB CDC
      flow control: ON
      rates:
        tx: 19716.4 B/s
        txerr: 0.0 B/s
        tx rate mult: 1.000
        tx rate max: 100000 B/s
        rx: 46.9 B/s
        rx loss: 0.0%
      Received Messages:
        sysid:255, compid:190, Total: 47 (lost: 0)
      FTP enabled: YES, TX enabled: YES
      mode: Config
      MAVLink version: 2
      transport protocol: serial (/dev/ttyACM0 @2000000)
      ping statistics:
        last: 1.86 ms
        mean: 4.03 ms
        max: 75.57 ms
        min: 1.21 ms
        dropped packets: 0
  ```
  
</details>

<details>
  <summary>Git Diff between my and your output</summary>
  
  ```diff
  $ git diff me.txt oes.txt
  diff --git a/me.txt b/oes.txt
  index 6954d24..56bfbeb 100644
  --- a/me.txt
  +++ b/oes.txt
  @@ -1,18 +1,35 @@
   instance #0:
  +    GCS heartbeat valid
       mavlink chan: #0
  -    type:        GENERIC LINK OR RADIO
  +    type:        RADIO Link
  +      rssi:        210
  +      remote rssi:    209
  +      txbuf:    100
  +      noise:    46
  +      remote noise:    48
  +      rx errors:    0
  +      fixed:    0
       flow control: ON
       rates:
  -      tx: 0.0 B/s
  +      tx: 759.9 B/s
         txerr: 0.0 B/s
         tx rate mult: 0.850
  :...skipping...
  diff --git a/me.txt b/oes.txt
  index 6954d24..56bfbeb 100644
  --- a/me.txt
  +++ b/oes.txt
  @@ -1,18 +1,35 @@
   instance #0:
  +    GCS heartbeat valid
       mavlink chan: #0
  -    type:        GENERIC LINK OR RADIO
  +    type:        RADIO Link
  +      rssi:        210
  +      remote rssi:    209
  +      txbuf:    100
  +      noise:    46
  +      remote noise:    48
  +      rx errors:    0
  +      fixed:    0
       flow control: ON
       rates:
  -      tx: 0.0 B/s
  +      tx: 759.9 B/s
         txerr: 0.0 B/s
         tx rate mult: 0.850
         tx rate max: 1200 B/s
  -      rx: 0.0 B/s
  +      rx: 62.7 B/s
         rx loss: 0.0%
  +    Received Messages:
  +      sysid: 51, compid: 68, Total: 215 (lost: 0)
  +      sysid:255, compid:190, Total: 226 (lost: 0)
       FTP enabled: YES, TX enabled: YES
       mode: Normal
  -    MAVLink version: 1
  +    MAVLink version: 2
       transport protocol: serial (/dev/ttyS6 @57600)
  +    ping statistics:
  +      last: 174.94 ms
  +      mean: 170.62 ms
  +      max: 872.56 ms
  +      min: 113.10 ms
  +      dropped packets: 0
  
   instance #1:
       mavlink chan: #1
  @@ -20,7 +37,7 @@ instance #1:
       flow control: OFF
       rates:
         tx: 0.0 B/s
  -      txerr: 17.0 B/s
  +      txerr: 80.9 B/s
         tx rate mult: 0.050
         tx rate max: 100000 B/s
         rx: 0.0 B/s
  @@ -29,29 +46,4 @@ instance #1:
       mode: Normal
       MAVLink version: 1
       transport protocol: UDP (14550, remote port: 14550)
  -    Broadcast enabled: YES
  -
  -instance #2:
  -    GCS heartbeat valid
  -    mavlink chan: #2
  -    type:        USB CDC
  -    flow control: ON
  -    rates:
  -      tx: 19716.4 B/s
  -      txerr: 0.0 B/s
  -      tx rate mult: 1.000
  -      tx rate max: 100000 B/s
  -      rx: 46.9 B/s
  -      rx loss: 0.0%
  -    Received Messages:
  -      sysid:255, compid:190, Total: 47 (lost: 0)
  -    FTP enabled: YES, TX enabled: YES
  -    mode: Config
  -    MAVLink version: 2
  -    transport protocol: serial (/dev/ttyACM0 @2000000)
  -    ping statistics:
  -      last: 1.86 ms
  -      mean: 4.03 ms
  -      max: 75.57 ms
  -      min: 1.21 ms
  -      dropped packets: 0
  \ No newline at end of file
  +    Broadcast enabled: YES
  \ No newline at end of file
  ```

</details>

I am using [this telemetry radio](https://www.mybotshop.de/Drotek-Radio-Telemetry-433Mhz-3DR). The cable that came with it had all six wires but I cannot tell whether all of them are actually used on the board. However, even if nothing is connected to TELEM1, I cannot see data on the TX pin in my oscilloscope. This leads me to believe that this behavior is not the radio's fault.

julianoes, 2023-01-18:
Ok, the same output via USB looks like this for me:

```
nsh> 
nsh> mavlink status

instance #0:
	mavlink chan: #0
	type:		RADIO Link
	  rssi:		49
	  remote rssi:	0
	  txbuf:	97
	  noise:	54
	  remote noise:	0
	  rx errors:	0
	  fixed:	0
	flow control: ON
	rates:
	  tx: 823.0 B/s
	  txerr: 0.0 B/s
	  tx rate mult: 0.850
	  tx rate max: 1200 B/s
	  rx: 17.0 B/s
	  rx loss: 0.0%
	Received Messages:
	  sysid:  1, compid:  1, Total: 3 (lost: 0)
	  sysid: 51, compid: 68, Total: 20 (lost: 0)
	FTP enabled: YES, TX enabled: YES
	mode: Normal
	MAVLink version: 1
	transport protocol: serial (/dev/ttyS6 @57600)

instance #1:
	mavlink chan: #1
	type:		GENERIC LINK OR RADIO
	flow control: OFF
	rates:
	  tx: 0.0 B/s
	  txerr: 185.6 B/s
	  tx rate mult: 0.050
	  tx rate max: 100000 B/s
	  rx: 0.0 B/s
	  rx loss: 0.0%
	FTP enabled: YES, TX enabled: YES
	mode: Normal
	MAVLink version: 1
	transport protocol: UDP (14550, remote port: 14550)
	Broadcast enabled: YES

instance #2:
	mavlink chan: #2
	type:		USB CDC
	flow control: ON
	rates:
	  tx: 11286.0 B/s
	  txerr: 0.0 B/s
	  tx rate mult: 1.000
	  tx rate max: 100000 B/s
	  rx: 103.8 B/s
	  rx loss: 0.0%
	Received Messages:
	  sysid:255, compid:  0, Total: 10 (lost: 0)
	FTP enabled: YES, TX enabled: YES
	mode: Config
	MAVLink version: 1
	transport protocol: serial (/dev/ttyACM0 @2000000)
nsh> 
nsh> 
nsh> 
nsh> ver all
HW arch: PX4_FMU_V6X
HW type: V6X04
HW version: 0x00000000
HW revision: 0x00000004
FW git-hash: 46a12a09bf11c8cbafc5ad905996645b4fe1a9df
FW version: Release 1.13.2 (17629951)
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: 91bece51afbe7da9db12e3695cdbb4f4bba4bc83
Build datetime: Nov 22 2022 09:23:35
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 00060000000037343637333051190042003d
MCU: STM32H7[4|5]xxx, rev. V

```

julianoes, 2023-01-18:
I believe I can reproduce it. I connected an FTDI cable with the flow control pins not connected, and I see only incoming bytes right at startup, then it stops.

```
instance #0:
        mavlink chan: #0
        type:           GENERIC LINK OR RADIO
        flow control: ON
        rates:
          tx: 0.0 B/s
          txerr: 0.0 B/s
          tx rate mult: 0.850
          tx rate max: 1200 B/s
          rx: 0.0 B/s
          rx loss: 0.0%
        FTP enabled: YES, TX enabled: YES
        mode: Normal
        MAVLink version: 1
        transport protocol: serial (/dev/ttyS6 @57600)
```

It looks like flow control is ON even though the flow control pins are not connected, so it presumably stops transmitting once the buffers are full.

julianoes, 2023-01-18:
And I can confirm that the flow control detection seems to work fine on Telem 3 but not Telem 1. The problem seems to be the same in `main`.

julianoes, 2023-01-18:
I found that disabling DMA for Rx and Tx solves the the problem.

```
diff --git a/boards/px4/fmu-v6x/nuttx-config/nsh/defconfig b/boards/px4/fmu-v6x/nuttx-config/nsh/defconfig
index edb5e8a0d8..0af648004e 100644
--- a/boards/px4/fmu-v6x/nuttx-config/nsh/defconfig
+++ b/boards/px4/fmu-v6x/nuttx-config/nsh/defconfig
@@ -298,9 +298,7 @@ CONFIG_UART7_BAUD=57600
 CONFIG_UART7_IFLOWCONTROL=y
 CONFIG_UART7_OFLOWCONTROL=y
 CONFIG_UART7_RXBUFSIZE=600
-CONFIG_UART7_RXDMA=y
 CONFIG_UART7_TXBUFSIZE=3000
-CONFIG_UART7_TXDMA=y
 CONFIG_UART8_BAUD=57600
 CONFIG_UART8_RXBUFSIZE=600
 CONFIG_UART8_TXBUFSIZE=1500
```

FriedrichTuttas, 2023-01-18:
Nice! I can confirm that the TELEM1 port works if I set MAV_0_FLOW_CTRL to "force off" (=0) and/or disable DMA for UART7. Thanks a lot for your investigative work! 👏 

Since you could only reproduce my issues after disconnecting the flow control pins, am I correct to assume that the reason for us to experience a different behavior at first is that your telemetry radio actually uses the flow control pins and mine does not? If so, you should also not be able to see data on the TX pin if there is nothing attached to the TELEM1 port. Do you by chance have an oscilloscope at hand to confirm this? 😃 

Do you believe that DMA is the root cause of this problem? I noticed that, for fmu-v5x, DMA is also not enabled on USART2 (which is the TELEM1 port of the Pixhawk 4).

davids5, 2023-01-18:
@julianoes It could be that with DMA and CTS floating. CTS is HIGH (not clear to send) If we add a pull down on the CTS GPIO in the PIN init, that my resolve it.

julianoes, 2023-01-18:
> Since you could only reproduce my issues after disconnecting the flow control pins, am I correct to assume that the reason for us to experience a different behavior at first is that your telemetry radio actually uses the flow control pins and mine does not?

I think so. I'd suggest you open up yours and check it.

> If so, you should also not be able to see data on the TX pin if there is nothing attached to the TELEM1 port

Correct.

> Do you believe that DMA is the root cause of this problem?

DMA in itself should work but something must be misconfigured. I'll try what @davids5 suggested.

davids5, 2023-01-19:
> 

No.  See the note at the [bottom](https://github.com/PX4/NuttX/blob/px4_firmware_nuttx-10.3.0%2B/arch/arm/src/stm32h7/hardware/stm32h7x3xx_pinmap.h#L38-L58)

That is the SoC defines not the board usage. The assignment and modifications go in board.h

We should never effect other designs (board) It should be done in our the board.h for the boards that needs it.

FriedrichTuttas, 2023-01-19:
@julianoes @davids5 
It works like this:

```diff
diff --git a/boards/px4/fmu-v6x/nuttx-config/include/board.h b/boards/px4/fmu-v6x/nuttx-config/include/board.h
index 0c68913320..140e51ab19 100644
--- a/boards/px4/fmu-v6x/nuttx-config/include/board.h
+++ b/boards/px4/fmu-v6x/nuttx-config/include/board.h
@@ -388,8 +388,8 @@

 #define GPIO_UART7_RX    GPIO_UART7_RX_4    /* PF6  */
 #define GPIO_UART7_TX    GPIO_UART7_TX_3    /* PE8  */
-#define GPIO_UART7_RTS   GPIO_UART7_RTS_2   /* PF8  */
-#define GPIO_UART7_CTS   GPIO_UART7_CTS_1   /* PE10 */
+#define GPIO_UART7_RTS   (GPIO_UART7_RTS_2|GPIO_PULLDOWN)   /* PF8  */
+#define GPIO_UART7_CTS   (GPIO_UART7_CTS_1||GPIO_PULLDOWN)   /* PE10 */

 #define GPIO_UART8_RX    GPIO_UART8_RX_1    /* PE0 */
 #define GPIO_UART8_TX    GPIO_UART8_TX_1    /* PE1 */
```

davids5, 2023-01-19:
@FriedrichTuttas  With DMA enabled and the above is the only change?

julianoes, 2023-01-19:
Nice thanks for testing @FriedrichTuttas. Can you make a PR?

FriedrichTuttas, 2023-01-20:
Yes, I just did ([link](https://github.com/PX4/PX4-Autopilot/pull/20974)). This is my first PR. So, any helpful advice is very welcome. Is it OK to close this issue now or should I wait for the PR to be approved?

closed 2023-01-21 by dagar

julianoes, 2023-01-22:
Thanks @FriedrichTuttas!

PR #20974 (2023-01-21): added pulldowns to GPIO pins UART7 RTS and UART7 CTS

### Solved Problem

Fixes #20762

### Solution
As suggested in [this comment](https://github.com/PX4/PX4-Autopilot/issues/20762#issuecomment-1396628897), pulldowns were added to the GPIO pins of UART7 RTS and CTS.

### Alternatives
It was found that [disabling flow control](https://github.com/PX4/PX4-Autopilot/issues/20762#issuecomment-1386351136) or [disabling DMA for UART7](https://github.com/PX4/PX4-Autopilot/issues/20762#issuecomment-1386379337) also enabled MAVLink to work on TELEM1. However, the solution of this PR can leave both flow [... cut]