PX4/PX4-Autopilot #20775: Error while starting gazebo simulation px4_sitl

marceloroberti, 2022-12-14:
## Describe the bug
I'm starting to learn about px4 and was trying to start a Gazebo simulation (sitl), but I'm facing problems with that. I followed all the steps from https://docs.px4.io/main/en/dev_setup/building_px4.html and after solving some errors with help of the issues submitted here, I could start a jmavsim simulation as explained in the tutorial. Then I tried the gazebo simulation to change the airframe to a vtol, but got stuck in error.

I'm using Ubuntu 22.04.01 LTS on VirtualBox. I tried the following command in terminal:

> make px4_sitl gazebo_stantard_vtol

And got this message:

> [0/5] Performing configure step for 'sitl_gazebo'
> -- install-prefix: /usr/local
> -- cmake build type: RelWithDebInfo
> -- Using C++17 standard
> CMake Error at CMakeLists.txt:48 (find_package):
>   By not providing "Findgazebo.cmake" in CMAKE_MODULE_PATH this project has
>   asked CMake to find a package configuration file provided by "gazebo", but
>   CMake did not find one.
> 
>   
> 
> Could not find a package configuration file provided by "gazebo" with any
>   of the following names:
> 
>     gazeboConfig.cmake
>     gazebo-config.cmake
> 
>   Add the installation prefix of "gazebo" to CMAKE_PREFIX_PATH or set
>   "gazebo_DIR" to a directory containing one of the above files.  If "gazebo"
>   provides a separate development package or SDK, be sure it has been
>   installed.
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/marceloroberti/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
> FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-configure /home/marceloroberti/Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure 
> cd /home/marceloroberti/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_ODOMETRY_DATA=ON -DGENERATE_ROS_MODELS=ON -GNinja /home/marceloroberti/Firmware/Tools/simulation/gazebo/sitl_gazebo && /usr/bin/cmake -E touch /home/marceloroberti/Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
> ninja: build stopped: subcommand failed.
> make: *** [Makefile:232: px4_sitl] Error 1

I'm initiating using Ubuntu/Linux and PX4 so I appreciate if any answers be as simple and clear as possible, for me to understand. Thank you.

jimmy-loyola, 2022-12-15:
It seems that it wasn't installed by the script so try to force the gazebo installation:
sudo apt install gazebo libgazebo-dev

marceloroberti, 2022-12-16:
It worked for this error, but then I got a new one. I pasted here only the output part that shows the error, cause the output now is much bigger.

> -- Gazebo version: 11.10
-- Found GStreamer: adding gst_camera_plugin
-- Found GStreamer: adding gst_video_stream_widget
-- Configuring done
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GSTREAMER_APP_LIBRARIES (ADVANCED)
    linked by target "AdvancedLiftDragPlugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "ForceVisual" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_airship_dynamics_plugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_pose_sniffer_plugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "sensor_msgs" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_airspeed_plugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_groundtruth_plugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "physics_msgs" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "std_msgs" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_usv_dynamics_plugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_video_stream_widget" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_gps_plugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_opticalflow_mockup_plugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_irlock_plugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_parachute_plugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_lidar_plugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_multirotor_base_plugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_wind_plugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_opticalflow_plugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_uuv_plugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_drop_plugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "nav_msgs" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_sonar_plugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_vision_plugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_user_camera_plugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_gst_camera_plugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "mav_msgs" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_controller_interface" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_gimbal_controller_plugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_imu_plugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_mavlink_interface" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_motor_model" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_camera_manager_plugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_magnetometer_plugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_barometer_plugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "LiftDragPlugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
    linked by target "gazebo_catapult_plugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
GSTREAMER_LIBRARIES (ADVANCED)
    linked by target "AdvancedLiftDragPlugin" in directory /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
[... cut ...]
> cd /home/marceloroberti/PX4-Autopilot/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_ODOMETRY_DATA=ON -DGENERATE_ROS_MODELS=ON -GNinja /home/marceloroberti/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo && /usr/bin/cmake -E touch /home/marceloroberti/PX4-Autopilot/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
> ninja: build stopped: subcommand failed.
> make: *** [Makefile:232: px4_sitl] Error 1

Seems to me that has something with this "gazebo_[something]_plugin", but I don't know exactly what's going on.

jimmy-loyola, 2022-12-16:
On ubuntu 22.04 simulation build is disabled by default if you want it you should run

bash ./PX4-Autopilot/Tools/setup/ubuntu.sh  --sim_jammy

marceloroberti, 2022-12-19:
You're right. But now I got another error... seems like there is no ros package for this version of Ubuntu.
The output was:

> marceloroberti@UbuntuMarcelo:~$ bash ./PX4-Autopilot/Tools/setup/ubuntu.sh --sim_jammy
> Ubuntu 22.04, simulation build off by default.
> Use --sim_jammy to enable simulation build.
> 
> Installing PX4 general dependencies
> [sudo] password for marceloroberti: 
> Hit:1 http://br.archive.ubuntu.com/ubuntu jammy InRelease
> Hit:2 http://br.archive.ubuntu.com/ubuntu jammy-updates InRelease
> Hit:3 http://br.archive.ubuntu.com/ubuntu jammy-backports InRelease
> Ign:4 http://packages.ros.org/ros/ubuntu jammy InRelease
> Hit:5 http://security.ubuntu.com/ubuntu jammy-security InRelease
> Hit:6 http://packages.ros.org/ros2/ubuntu jammy InRelease
> Err:7 http://packages.ros.org/ros/ubuntu jammy Release
>   404  Not Found [IP: 64.50.236.52 80]
> Reading package lists...
> E: The repository 'http://packages.ros.org/ros/ubuntu jammy Release' does not have a Release file.

marceloroberti, 2022-12-20:
I was using a VM and now tried to do a dual boot to ensure that it was not the problem. Now I have Ubuntu 22.04.01 LTS without VM. I did the --sim_jammy command you've told, but still coming some errors. The command and the output are:

> marcelo@marcelo:~/PX4-Autopilot$ make px4_sitl gazebo
> [0/882] git submodule src/drivers/gps/devices
> [4/882] git submodule src/modules/mavlink/mavlink
> [10/882] git submodule src/modules/microdds_client/Micro-XRCE-DDS-Client
> [12/882] Generating Mavlink developmen...ssage_definitions/v1.0/development.xml
> Validating /home/marcelo/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
> Parsing /home/marcelo/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
> Validating /home/marcelo/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/common.xml
> Parsing /home/marcelo/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/common.xml
> Validating /home/marcelo/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/standard.xml
> Parsing /home/marcelo/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/standard.xml
> Validating /home/marcelo/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/minimal.xml
> Parsing /home/marcelo/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/minimal.xml
> Merged enum MAV_CMD
> Merged enum MAV_CMD
> Found 232 MAVLink message types in 4 XML files
> Generating C implementation in directory /home/marcelo/PX4-Autopilot/build/px4_sitl_default/mavlink/development
> Generating C implementation in directory /home/marcelo/PX4-Autopilot/build/px4_sitl_default/mavlink/common
> Generating C implementation in directory /home/marcelo/PX4-Autopilot/build/px4_sitl_default/mavlink/standard
> Generating C implementation in directory /home/marcelo/PX4-Autopilot/build/px4_sitl_default/mavlink/minimal
> Copying fixed headers for protocol 2.0 to /home/marcelo/PX4-Autopilot/build/px4_sitl_default/mavlink
> [17/882] Generating Mavlink uAvionix: .../message_definitions/v1.0/uAvionix.xml
> Validating /home/marcelo/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/uAvionix.xml
> Parsing /home/marcelo/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/uAvionix.xml
> Validating /home/marcelo/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/common.xml
> Parsing /home/marcelo/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/common.xml
> Validating /home/marcelo/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/standard.xml
> Parsing /home/marcelo/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/standard.xml
> Validating /home/marcelo/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/minimal.xml
> Parsing /home/marcelo/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/minimal.xml
> Found 224 MAVLink message types in 4 XML files
> Generating C implementation in directory /home/marcelo/PX4-Autopilot/build/px4_sitl_default/mavlink/uAvionix
> Generating C implementation in directory /home/marcelo/PX4-Autopilot/build/px4_sitl_default/mavlink/common
> Generating C implementation in directory /home/marcelo/PX4-Autopilot/build/px4_sitl_default/mavlink/standard
> Generating C implementation in directory /home/marcelo/PX4-Autopilot/build/px4_sitl_default/mavlink/minimal
> Copying fixed headers for protocol 2.0 to /home/marcelo/PX4-Autopilot/build/px4_sitl_default/mavlink
> [18/882] git submodule Tools/simulation/gazebo/sitl_gazebo
> [23/882] Performing configure step for 'sitl_gazebo'
> -- install-prefix: /usr/local
> -- cmake build type: RelWithDebInfo
> -- Using C++17 standard
> -- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.58") found components: system thread filesystem 
> CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
>   The package name passed to `find_package_handle_standard_args` (PkgConfig)
>   does not match the name of the calling package (gazebo).  This can lead to
>   problems in calling code that expects `find_package` result variables
>   (e.g., `_FOUND`) to follow a certain pattern.
> Call Stack (most recent call first):
>   /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)
>   /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:72 (include)
>   CMakeLists.txt:48 (find_package)
> This warning is for project developers.  Use -Wno-dev to suppress it.
> 
> -- Found DART: /usr/include (Required is at least version "6.6") found components: dart 
> -- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.40.0") found components: thread system filesystem program_options regex iostreams date_time 
> -- Looking for ignition-math6 -- found version 6.10.0
> -- Searching for dependencies of ignition-math6
> -- Looking for OGRE...
> -- Found Ogre Ghadamon (1.9.0)
> -- Looking for OGRE_Paging...
> -- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
> -- Looking for OGRE_Terrain...
> -- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
> -- Looking for OGRE_Property...
[... cut ...]
> [878/882] Linking CXX executable bin/px4
> FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build /home/marcelo/PX4-Autopilot/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-build 
> cd /home/marcelo/PX4-Autopilot/build/px4_sitl_default/build_gazebo && /usr/bin/cmake --build /home/marcelo/PX4-Autopilot/build/px4_sitl_default/build_gazebo -- -j 8
> ninja: build stopped: subcommand failed.
> make: *** [Makefile:232: px4_sitl] Error 1

I don't know if I should open another issue, cause every time the error is different.

jimmy-loyola, 2022-12-20:
Try to add these two lines:

#include <boost/bind/bind.hpp>
using namespace boost::placeholders;

in the file
../PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo/src/gazebo_pose_sniffer_plugin.cpp

Jaeyoung-Lim, 2022-12-20:
@frederictaillandier Any ideas on how we can solve this? Should we try to get the pose sniffer plugin to build optionally?

frederictaillandier, 2022-12-20:
I will be totally fine with any solution you offer, but I' m not sure to understand why this breaks, the bind is exactly the same as in other plugins

marceloroberti, 2022-12-20:
Jimmy, your answers worked. I could open the simulator perfectly after doing what you said. Thank you.

closed 2023-01-05 by marceloroberti