PX4/PX4-Autopilot #20783: Software-In-Hardware (SIH) broken on git head

wluker, 2022-12-15:
## Describe the bug
Sensors are now missing in the software-in-hardware (SIH) mode.
Probably since commits 8f25acd4285e3696855ce220c128518bd519c542 bfe0d71a21e95960f4a82e89890a5b54a75ae0e3 c5336abba269d06ffeffbc0810e3e24fae505d27 and 99a20646e2f1f393d6ed2ed075e8ac260bb8880a ?

## To Reproduce
Steps to reproduce the behavior:
1. In QGS set the Airframe to `SIH Quadcopter X`
2. After reboot QGC reports `barometer 0 missing` and `compass sensor 0 missing`.
3. Drone won't fly due to pre-flight checks.

## Expected behavior
Should be able to arm and fly.

dagar, 2022-12-15:
I've been using it in SITL (we need to rename SIH at this point). What board are you using?

wluker, 2022-12-15:
> What board are you using?

Right, I should have mentioned that. Pixhawk 4

wluker, 2022-12-15:
And QGC v4.2.3 if that matters.

TompsonTan, 2023-01-03:
@dagar its because sensor_gps_sim not started.  should add    sensor_gps_sim start   to rcs

bresch, 2023-01-03:
I had the same issue a couple of weeks ago, I just pushed the fix

closed 2023-01-04 by dagar