PX4/PX4-Autopilot #20793: mRo Control Zero H7 support broken by 2cb4ef0629

Ecuashungo, 2022-12-19:
## Describe the bug
Commit 2cb4ef0629 breaks support for the control zero h7 board by mrobotics. Its parent commit (45b390b0bf) still works fine.

You get the following error in QGC: 
[image]
and in the nutt-shell the `dmesg` command only returns `NuttShell (NSH) NuttX-11.0.0`.

## To Reproduce
1. Flash 2cb4ef0629 onto the control zero h7: `make mro_ctrl-zero-h7 upload`
2. Open QGC and try to do the setup. I was not able to select an airframe.

## Expected behavior
I expect everything to work normally.

## Drone (please complete the following information):
- MRobotics Control Zero H7 with the aforementioned PX4-Autopilot software versions.

dagar, 2022-12-19:
Thanks for reporting, I'll take a closer look.

FYI @davids5 @PetervdPerk-NXP

PetervdPerk-NXP, 2022-12-19:
[mro_fix_nx_10_4.patch](https://github.com/PX4/PX4-Autopilot/files/10260473/mro_fix_nx_10_4.patch)
Could you try following patch?

Ecuashungo, 2022-12-19:
@PetervdPerk-NXP Thanks for that patch. 
It worked for both 2cb4ef0 and the current main (9c66f1b14a).

By working I mean that I conducted a quick benchtest where I configured everything and spun the motors without receiving errors. I haven't done a proper flight test.

What was the issue?

PetervdPerk-NXP, 2022-12-19:
https://github.com/PX4/PX4-Autopilot/commit/2cb4ef0629d7890e64c632bf57b2086a237dc35c Upgrades NuttX kernel but also all configuration, most of them were updates using regexes but some targets might be missed by this.

I did a compare with the FMU-V6X target and took over the configs from there.

gorghino, 2022-12-27:
Glad you fixed it!
To be honest it was a duplicate of #20743

PR #20812 (2022-12-28): Fix MRO NuttX defconfig
Fixes #20793

By correcting NuttX defconfig
closed 2022-12-28 by dagar