PX4/PX4-Autopilot #20813: "MAV_CMD_COMPONENT_ARM_DISARM command temporarily rejected" while arming

Attahayat, 2022-12-21:
I am using Pixhawk 5 with PX4 installed on it. My operating system is Ubuntu 20.04. When I try to arm my rover using QGC, I get an error "MAV_CMD_COMPONENT_ARM_DISARM temporarily disabled". I disabled all the prearm checks that could have caused this but I am still getting it.

When I go to the safety tab I get an error that my firmware does not have the param CP_DIST which after looking up on here, I found out that it might be caused because the power is not set up correctly. I set up the batteries again (power module) and calibrated them but I still get the error. Also on the main screen, I am notified that there is some crash dump on SD, and the vehicle needs service which I do not understand the meaning of.

I worked before with micrortps_client and everything worked fine. Now with microdds_client installed on my Pixhawk 5, I am facing this problem while arming. SITL works fine as well.

I tried to find what might be the cause here but I cannot find anything. Any help would be appreciated as this has been very frustrating. Thanks.

AuroraRAS, 2023-02-02:
encountered the same problem. 

- "CP_DIST" problem in safety tab
- "Ready To Fly" display green on QGC
- "MAV_CMD_COMPONENT_ARM_DISARM command temporarily rejected" without any reason

my rover is working good with 1.13.2 [px4_fmu-v2_rover.px4](https://github.com/PX4/PX4-Autopilot/releases/download/v1.13.2/px4_fmu-v2_rover.px4), but not 1.14 beta

junwoo091400, 2023-06-29:
The arming rejected should show the reason for it through the [event interface](https://github.com/mavlink/qgroundcontrol/pull/10362) in QGC, like this:
[image]

@AuroraRAS @Attahayat Have you not noticed any reasons popping up?

---

Below are just some debugging steps I took:

command temporarily rejected is arising from this:
https://github.com/mavlink/qgroundcontrol/blob/7dc5315b26f4be4e90bf7b6a412c6b5e340362e7/src/Vehicle/Vehicle.cc#L3233-L3236

Which is in turn triggered due to this:
https://github.com/PX4/PX4-Autopilot/blob/a6d2c2cf5e21ecdad8d737c31de81fab128898c5/src/modules/commander/Commander.cpp#L885
-->

junwoo091400, 2023-06-29:
Also, the `CP_DIST` missing error is in fact due to the parameter missing, I have added that for v1.14 release TODOs, thanks!

PrakritTyagi, 2023-08-21:
I am also having the issue of "MAV_CMD_COMPONENT_ARM_DISARM command temporarily rejected". 

encountered problem.
"Ready To Fly" display green on QGC
"MAV_CMD_COMPONENT_ARM_DISARM command temporarily rejected" without any reason when trying to arm form QGC or through ros2 offboard control example.

PX4 Firmware: release/1.14
Board: fmuV5
Arming rejection shows : crash dump on SD.

mkarklins, 2023-11-15:
Has anyone solved this issue?. There are no error/warning messages, however arm'ing is not possible, also ROS2 reports that "pre_arm_checks: fail"

PrakritTyagi, 2023-11-16:
It worked for me after I deleted the logs from the SD card inside Pixhawk and flashed the board with a fresh firmware. I followed the PX4 setup and got no issue after that.

mkarklins, 2023-11-16:
Thanks for you reply!

Turns out that if you're trying to arm the drone in "position" mode indoors without GPS it will just silently fail pre-arm check. Other modes work fine.

darshankt, 2023-12-18:
Hi,
I'm getting the same issue with RTK GPS setup. 
Using px4 v1.14.0, 
All the pre-flights pass, QGC saying Ready to Fly(Green)
When I moved into position mode, getting "MAV_CMD_COMPONENT_ARM_DISARM command temporarily rejected". 

How can I resolve this issue, Is there any debugging approaches to detect the hidden check-list fails which causing this issue.

mkarklins, 2023-12-18:
> Hi, I'm getting the same issue with RTK GPS setup. Using px4 v1.14.0, All the pre-flights pass, QGC saying Ready to Fly(Green) When I moved into position mode, getting "MAV_CMD_COMPONENT_ARM_DISARM [...]

FWIW I wasn't able to find any debugguing approaches. Are you 100% sure GPS is recognized and enabled? Do other modes fail as well?

darshankt, 2023-12-18:
Yes, GPS is active with 25+ satellites and 3D RTK Lock(Fixed). This is use start coming when I disable EKF2_MAG_TYPE = None. Since i want get heading angle from Dual RTK GPS setup i disabled compass usage. Now this error keep on popping up when try arm.

darshankt, 2023-12-18:
I just updated the px4 v1.15.0aphla firmware, still this error keeps coming when arming drone. I tried with stabilized mode and position mode. Even I disabled GPS by setting COM_ARM_WO_GPS still getting this error in stabilize mode.

darshankt, 2023-12-19:
After calibrating compass, this error got resolved for me!