PX4/PX4-Autopilot #20820: Cannot successfully run the sensor_combine_listener example.

fbenti, 2022-12-22:
## Setup
Ubunutu 20.04
Ros2 Foxy

First I tried following the Documentation [PX4 User Guide](https://docs.px4.io/main/en/ros/ros2_comm.html)
## To Reproduce
Steps to reproduce the behavior:
1.  launch simulator
```
cd ~/PX4-Autopilot
make px4_sitl gazebo
```
Everything look  fine until here, as I receive the INFO [commander] Ready for takeoff
2. launch micrortps_agent
```
source ~/px4_ros_com_ros2/install/setup.bash
micrortps_agent -t UDP
```

I get this error:
```
micrortps_agent: command not found
```

So I installed the microROS aplication [](https://micro.ros.org/docs/tutorials/core/first_application_linux/), following the provided instruction. 

After step 1., I run the following commands:
```
source ~/microros_ws/install/local_setup.bash
ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888 -v
```
Again everything looks fine, as the terminal with the simulator running shows that the topics have been successfully created.
```
INFO  [microdds_client] synchronized with time offset 1671720811191683us
INFO  [microdds_client] successfully created rt/fmu/out/failsafe_flags data writer, topic id: 79
INFO  [microdds_client] successfully created rt/fmu/out/sensor_combined data writer, topic id: 164
INFO  [microdds_client] successfully created rt/fmu/out/timesync_status data writer, topic id: 184
INFO  [microdds_client] successfully created rt/fmu/out/vehicle_attitude data writer, topic id: 201
INFO  [microdds_client] successfully created rt/fmu/out/vehicle_control_mode data writer, topic id: 207
INFO  [microdds_client] successfully created rt/fmu/out/vehicle_global_position data writer, topic id: 208
INFO  [microdds_client] successfully created rt/fmu/out/vehicle_gps_position data writer, topic id: 210
INFO  [microdds_client] successfully created rt/fmu/out/vehicle_local_position data writer, topic id: 214
INFO  [microdds_client] successfully created rt/fmu/out/vehicle_odometry data writer, topic id: 219
INFO  [microdds_client] successfully created rt/fmu/out/vehicle_status data writer, topic id: 224
```

Now after having removed the ExecuteProcess from sensor_combined_listener.launch.py, as below:
```
def generate_launch_description():

    sensor_combined_listener_node = Node(
        package='px4_ros_com',
        executable='sensor_combined_listener',
        shell=True
    )

    return LaunchDescription([
        sensor_combined_listener_node
    ])

```
3. run launch file:
```
ros2 launch px4_ros_com sensor_combined_listener.launch.py

[INFO] [launch]: All log files can be found below /home/aut/.ros/log/2022-12-22-15-58-24-512841-aut-hp-441179
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sensor_combined_listener-1]: process started with pid [441181]

```
But then nothing gets printed. 

However, if I manually subscribe to the topic, I get data:
```
ros2 topic echo /fmu/out/sensor_combined 

timestamp: 1671721346613820
gyro_rad:
- -0.0010652621276676655
- -0.007190529722720385
- -0.0031958085019141436
gyro_integral_dt: 4000
accelerometer_timestamp_relative: 0
accelerometer_m_s2:
- 0.023942016065120697
- -0.07062878459692001
- -9.855731964111328
accelerometer_integral_dt: 4000
accelerometer_clipping: 0
gyro_clipping: 0
accel_calibration_count: 0
gyro_calibration_count: 1
---
```

beniaminopozzan, 2022-12-24:
The [QoS](https://docs.ros.org/en/rolling/Concepts/About-Quality-of-Service-Settings.html) of the subscriber does not match the one of the publisher. You can see that from `ros2 topic info /fmu/out/sensor_combined -v`:

[image]

fbenti, 2022-12-24:
Thanks for the answer, but I can make the two qos to match? I'm trying to modify the sensor_combined_listener.cpp without success.

beniaminopozzan, 2022-12-24:
They don't have to match, you just need to make them compatible.
Try changing https://github.com/PX4/px4_ros_com/blob/af846cbeecdb147d4062f7c928ea3fc19ae60263/src/examples/listeners/sensor_combined_listener.cpp#L53-L54 to

```c++
subscription_ = this->create_subscription<px4_msgs::msg::SensorCombined>("/fmu/out/sensor_combined",
rclcpp::QoS(rclcpp::KeepLast(1)).best_effort().durability_volatile(),
[this](const px4_msgs::msg::SensorCombined::UniquePtr msg) {
```

fbenti, 2022-12-25:
Solved, thanks. Also, the launch file was missing the `output='screen'` parameter.

closed 2022-12-25 by fbenti