PX4/PX4-Autopilot #20826: Cannot take off with x500 in ignition gazebo fortress

Jaeyoung-Lim, 2022-12-23:
## Describe the bug
I tried to run SITL with ignition gazebo fortress on the current main, but the vehicle does not take off

## To Reproduce
Steps to reproduce the behavior:
1. Run the following command as written in https://docs.px4.io/main/en/simulation/ignition_gazebo.html#running-the-simulation
```
make px4_sitl gz_x500
```
2. In QGC, command a takeoff
3. Vehicle does not take off from the ground

## Expected behavior
The vehicle takes off and goes into hold mode

## Log Files and Screenshots
[image]

## Additional context
@benjinne @dagar am I missing some context here? or is it expected to be broken?

bperseghetti, 2022-12-23:
This might need to be brought back in: https://github.com/PX4/PX4-Autopilot/pull/20342

What does your terminal say, should have a debug about motor topic subscriptions.

Jaeyoung-Lim, 2022-12-23:
@bperseghetti Here is the full log:
```
$ make px4_sitl gz_x500
[0/2] Re-checking globbed directories...
[0/1] cd /home/jaeyoung/src/PX4-Autopi...topilot/build/px4_sitl_default/bin/px4

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=4001
INFO  [param] selected parameter default file parameters.bson
INFO  [param] importing from 'parameters.bson'
INFO  [parameters] BSON document size 274 bytes, decoded 274 bytes (INT32:11, FLOAT:3)
INFO  [param] selected parameter backup file parameters_backup.bson
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
INFO  [init] starting gazebo with world: /home/jaeyoung/src/PX4-Autopilot/Tools/simulation/gz/worlds/default.sdf
WARN  [init] PX4_GZ_MODEL_NAME or PX4_GZ_MODEL not set using PX4_SIM_MODEL.
INFO  [gz_bridge] world: default, model name: x500_0, simulation model: x500
INFO  [gz_bridge] subscribed: /x500_0/command/motor_speed
INFO  [gz_bridge] subscribed: /world/default/model/x500_0/link/base_link/sensor/imu_sensor/imu
INFO  [gz_bridge] subscribed: /world/default/pose/info
INFO  [gz_bridge] subscribed: /world/default/clock
../src/intel/isl/isl.c:2105: FINISHME: ../src/intel/isl/isl.c:isl_surf_supports_ccs: CCS for 3D textures is disabled, but a workaround is available.
[Err] [MulticopterMotorModel.cc:253] Please specify a robotNamespace.
[Err] [MulticopterMotorModel.cc:253] Please specify a robotNamespace.
[Err] [MulticopterMotorModel.cc:253] Please specify a robotNamespace.
[Err] [MulticopterMotorModel.cc:253] Please specify a robotNamespace.
INFO  [lockstep_scheduler] setting initial absolute time to 8000 us
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2022-12-23/13_13_48.ulg	
INFO  [logger] Opened full log file: ./log/2022-12-23/13_13_48.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [mavlink] partner IP: 127.0.0.1
INFO  [tone_alarm] home set
INFO  [commander] Ready for takeoff!
INFO  [commander] Armed by external command	
INFO  [tone_alarm] arming warning
INFO  [commander] Takeoff detected	
```

I am on Ubuntu 20.04, and my ignition gazebo version is
```
$ ign gazebo --versions
6.12.0
```

bperseghetti, 2022-12-23:
Did you build fortress from source and/or is your system fully updated with latest binary release?

[Err] [MulticopterMotorModel.cc:253] Please specify a robotNamespace. Shouldn't exist in updated plugin.

bperseghetti, 2022-12-23:
@Jaeyoung-Lim current version with updates should be 6.13.0: 
[image]

Wondering why the old script to check for that got removed, as that would have eliminated this issue to begin with, @dagar should we re-add it? https://github.com/PX4/PX4-Autopilot/pull/20342

Jaeyoung-Lim, 2022-12-24:
@bperseghetti 

Reinstalling the package `ignition-fortress` didn't help, but wondering if it is in one of the dependency trees that are not getting reinstalled properly.

> Wondering why the old script to check for that got removed, as that would have eliminated this issue to begin with, @dagar should we re-add it? https://github.com/PX4/PX4-Autopilot/pull/20342

Probably we now need mission tests running in CI to prevent this from silently breaking

Jaeyoung-Lim, 2022-12-24:
Thanks @bperseghetti  for the support!

Updating manually to libignition-6-dev to point at 6.13.0 solved the problem

closed 2022-12-24 by Jaeyoung-Lim

MichaelBosello, 2023-03-28:
I have the same issue on the current main with Gazebo Garden and Ubuntu 20.04. Any suggestion?

Jaeyoung-Lim, 2023-03-28:
@MichaelBosello Could you open a new issue and describe your problem exactly?
The issue here is resolved, so likely you have a different issue with similar behavior

MichaelBosello, 2023-03-28:
@Jaeyoung-Lim  I get the same error in this issue
`[Err] [MulticopterMotorModel.cc:255] Please specify a robotNamespace.`
And the vehicle doesn't move. I make it work by manually defining the `robotNamespace` of the motors plugin in the `model.sdf`:
`<robotNamespace>x500_0</robotNamespace>`
But it isn't a good way to handle the problem

I'll open a new issue

Jaeyoung-Lim, 2023-03-28:
@MichaelBosello Please post a new issue