PX4/PX4-Autopilot #20834: Ubuntu 22.04: Gazebo (not Ignition) Simulation doesn't work with `ubuntu.sh` script dependencies

junwoo091400, 2022-12-28:
## Describe the bug
As described in the doc: https://docs.px4.io/main/en/dev_setup/dev_env_linux_ubuntu.html#gazebo-jmavsim-and-nuttx-pixhawk-targets, the `ubuntu.sh` script should suffice to install simulation environment using Gazebo (not Ignition-gazebo, which is documented separately: https://docs.px4.io/main/en/simulation/ignition_gazebo.html)

However, currently the `ubuntu.sh` script doesn't even install Gazebo, as changed by this PR: https://github.com/PX4/PX4-Autopilot/pull/20833. And hence, the SITL for using legacy Gazebo doesn't work.

## To Reproduce
```bash
./Tools/setup/ubuntu.sh --sim_jammy
# Restart the computer
make px4_sitl gazebo
```

[image]

## Expected behavior
`ubuntu.sh` script should install gazebo as part of it's dependencies and enable SITL

junwoo091400, 2022-12-28:
@bperseghetti could you check on this?

junwoo091400, 2022-12-28:
Issue was resolved after installing `gazebo` and `libgazebo-dev` packages via `apt-get`.

Shall we get these back into the script then? @bperseghetti

bperseghetti, 2022-12-28:
If you want people to not have to worry about dependency issues and spending hours explaining why peoples simulations are broken, you have to choose between dead gazebo (aka classic gazebo) and the only gazebo going forward (we call gz in PX4). 

PX4-SITL_gazebo is on dead/classic gazebo. We need a clean break from stuff that loses support/updates in a year and will no longer even have binary installs in 2 years. 

That break point is 22.04, as gazebo11 is no longer bundled in the ros-humble-desktop image (fortress is). Installing gazebo11 on 22.04 (with gz-fortress) is playing Russian roulette with your system dependencies. Classic gazebo (gazebo11) technically is only supported on bionic and focal https://classic.gazebosim.org/ . If people want to install it, they can choose to after reading all the docs and figuring how to themselves, it is not something a script should ever do.

junwoo091400, 2022-12-28:
> That break point is 22.04, as gazebo11 is no longer bundled in the ros-humble-desktop image (fortress is). Installing gazebo11 on 22.04 (with gz-fortress) is playing Russian roulette with your system dependencies.

So would that mean that `sitl_gazebo` submodule isn't usable from now point on? (Since it includes all the plugins to work with classic Gazebo), or do we just need to port it to use it?

Also, I think this should be explained more clearly in the Docs then, since I was also wondering why the old scripts (basically the whole doc: https://docs.px4.io/main/en/simulation/gazebo.html) wouldn't work, with ignition :thinking:

bperseghetti, 2022-12-28:
Yes, so the docs need a renaming, the sub-module needs a renaming, and we need to let people know that while gazebo11 on 20.04 will work till Jan 2025 (another 2 years) there is no official support of it on 22.04 (and it can get yanked at anytime as it has name collisions with garden). Note that while the libraries currently do say ignition, the simulator has been full re-branded into being called Gazebo. Gazebo Garden uses gz launch command and I'm trying to see if we can have them full backport that to fortress (there already is a gz-fortress package). I did some initial rework on the docs here, I just simply don't have the time to go through everything and fix/rename it. https://github.com/PX4/PX4-user_guide/tree/pr-ignition_is_gazebo-gazebo_is_classic

I think this needs to be done ASAP before all of this gets much more confusing for people.

bperseghetti, 2022-12-28:
Also the install target should be renamed to something like sitl_classic_gazebo or even drop the sitl as that's sorta obvious and pointeless given the context of gazebo, should really just be classic_gazebo as the target. and we should rename the repo to PX4-classic_gazebo @dagar thoughts?

beniaminopozzan, 2022-12-28:
@bperseghetti , in #20833 you basically switched off the installation of the dependencies for the simulators in ubuntu 22.04, leaving `--sim_jammy` to take care to install only `ignition-fortress`. It is clear to me that we need to break between gazebo classic and gazebo, but what about the other simulators? Should ignition-fortress be the only supported simulator for ubuntu 22.04? Starting from this, we can make the docs clearer.

zeroos, 2023-01-13:
It seems like the initially described issue has already been resolved in https://github.com/PX4/PX4-Autopilot/commit/19cee04f3af66f0be1aa45995e6a01772c670ded. However, the script still did not run flawlessly for me, I've described the problem I've encountered in a separate issue: https://github.com/PX4/PX4-Autopilot/issues/20923 .

closed 2023-02-27 by Jaeyoung-Lim