PX4/PX4-Autopilot #20836: Gazebo VTOL and fixed wing plane support

ahcorde, 2022-12-29:
Hello,

I'm trying to support VTOL and fixed wing planes models in Gazebo. But I'm having some issues that I will try to sumarize and I wil give some context too.

## Airspeed sensor in Gazebo

The airspeed sensor doesn't exist in Gazebo so I added this sensor, These are the PRs

 - https://github.com/gazebosim/sdformat/pull/1215
 - https://github.com/gazebosim/gz-msgs/pull/321
 - https://github.com/gazebosim/gz-sensors/pull/305
 - https://github.com/gazebosim/gz-sim/pull/1847

## `standard_vtol` and `plane` models

Then I adapted the `standard_vtol` and `plane` models to Gazebo (mainly the plugins and materials). You can find them [here](https://github.com/osrf/vehicle_gateway/tree/ahcorde/px4/patch_bridge/px4_sim/models) 

## Modify airframes file

I added some new params to the airframes config file which are [here](https://github.com/osrf/vehicle_gateway/tree/ahcorde/px4/patch_bridge/px4_sim/airframes) , such us: 

```bash
PX4_SIMULATOR=${PX4_SIMULATOR:=gz}
PX4_GZ_WORLD=${PX4_GZ_WORLD:=default}
PX4_SIM_MODEL=${PX4_SIM_MODEL:=plane}

...
param set-default SIM_GZ_FUNCX 
...

param set-default ASPD_PRIMARY 1
param set-default FW_ARSP_MODE 0
param set-default CBRK_AIRSPD_CHK 162128

```

For some reason the airspeed sensor it's not working or at least the check that's why I added `CBRK_AIRSPD_CHK`. I included in the `GzBridge` a new uorb topic called `differential_pressure`. You can see the patches [here](https://github.com/osrf/vehicle_gateway/tree/ahcorde/px4/patch_bridge/px4_sim/patches).

# Issues

## VTOL 

The VTOL is able to fly in quadcopter mode even it's able to move to a specific position using autonomous modes, for example POSITION mode, but when the transition to VTOL is activated the drone start the transition but when the transition finished I got an ESC failure crashing the drone.

I can share a log or add more details

## Plane

The plane is able to take off but is always turning the left making circles, maybe one of the channels are not properly configured?

# How to reproduce this setup

I'm testing this with ROS 2 humble and Gazebo Garden

```bash
mkdir ~/px4_ws/src
cd ~/px4_ws/src
git clone https://github.com/osrf/vehicle_gateway -b ahcorde/px4/patch_bridge
cd ~/px4_src/
vcs import src < src/vehicle_gateway/dependencies.repos
source /opt/ros/humble/setup.bash
colcon build --merge-install --event-handlers console_direct+
```

Please let me know if you need more details or info I would like to include these changes upstream.

Jaeyoung-Lim, 2022-12-30:
@ahcorde This is awesome, and great to see it also made it into the sdf format

The description seems to hint that it might be related to some channel mapping or the direction of it.

> I can share a log or add more details

If you can provide a log, this would be great to see what is actually going wrong.

> For some reason the airspeed sensor it's not working or at least the check that's why I added CBRK_AIRSPD_CHK. I included in the GzBridge a new uorb topic called differential_pressure

This means that the airspeed sensor data is not correct for the vehicle. Would definitely benefit from looking at the log.

I can also test on myside, but I only have access to Ubuntu 20.04, which I think might be a problem for running Gazebo Garden

ahcorde, 2023-01-03:
- VTOL transition crash [VTOL_Transition_crash.zip](https://github.com/PX4/PX4-Autopilot/files/10336535/VTOL_Transition_crash.zip)

ahcorde, 2023-01-03:
- Plane circles [plane.zip](https://github.com/PX4/PX4-Autopilot/files/10336571/plane.zip)

beniaminopozzan, 2023-01-07:
@ahcorde thank you!
I tried to reproduce a bare minimum version of your work (I based everything from https://github.com/osrf/vehicle_gateway):

- `gazebo garden` source installed using your PRs to have the airspeed sensor.
- `gz_bridge` patched.
- `plane` airframe and model and `empty_px4_world` world added to PX4.

The control surfaces are not actuated, there is nothing that maps `command/motor_speed` to the joint positions. As a result in your case you kept turning left, while in my case I kept making loops.

On the other hand, the airspeed sensor worked, or at least I was able to set `CBRK_AIRSPD_CHK=0` without getting any preflight check errors and QGC showed reasonable values of _airspeed_.

A side note: your plugin directly publishes the differential pressure in `Pa`, right? here https://github.com/osrf/vehicle_gateway/blob/3aa0a3391effadc8f1601866570562bafa1aaf65/px4_sim/patches/GzBridge.cpp.patch#L37, can the comment be removed?

Jaeyoung-Lim, 2023-01-14:
> Plane circles [plane.zip](https://github.com/PX4/PX4-Autopilot/files/10336571/plane.zip)

@ahcorde I tried to reproduce your problems using the vehicle_gateway repository, but could not get the vehicle to take off.

I have written down some questions in https://github.com/osrf/vehicle_gateway/issues/17

Jaeyoung-Lim, 2023-01-15:
@bperseghetti I had a look at the `pr-gz-bridge-airspeed` branch, and there doesn't seem to be anything related to joint control. Are the control surfaces supposed to be actuated in this branch?

closed 2023-01-24 by dagar